coolsnowwolf/lede #2062: softetherVPN报错

mg6630, 2019-11-12:
选择luci-app-softethervpn 插件 编译就会报错，不选择的话可以编译成功。4.9  4.19   ubuntu14.04. 16.04  18.04 都试过，不选择这个插件都能成功，选了就报错。
/home/mg/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/7.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: tmp/objs/Mayaqua/Encrypt.o: in function `X509ToX':
Encrypt.c:(.text+0x4b94): undefined reference to `ASN1_STRING_get0_data'
/home/mg/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/7.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: tmp/objs/Mayaqua/Network.o: in function `StartSSLEx':
Network.c:(.text+0xbaf0): undefined reference to `SSLv3_method'
collect2: error: ld returned 1 exit status
src/makefiles/linux_64bit.mak:355: recipe for target 'tmp/hamcorebuilder' failed
make[4]: *** [tmp/hamcorebuilder] Error 1
make[4]: Leaving directory '/home/mg/lede/build_dir/target-x86_64_musl/softethervpn/v4.29-9680'
Makefile:194: recipe for target '/home/mg/lede/build_dir/target-x86_64_musl/softethervpn/v4.29-9680/.built' failed
make[3]: *** [/home/mg/lede/build_dir/target-x86_64_musl/softethervpn/v4.29-9680/.built] Error 2
make[3]: Leaving directory '/home/mg/lede/feeds/packages/net/softethervpn'
time: package/feeds/packages/softethervpn/compile#0.26#0.07#0.33
package/Makefile:107: recipe for target 'package/feeds/packages/softethervpn/compile' failed
make[2]: *** [package/feeds/packages/softethervpn/compile] Error 2
make[2]: Leaving directory '/home/mg/lede'
package/Makefile:103: recipe for target '/home/mg/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/mg/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/mg/lede'
/home/mg/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

liuluo1979, 2019-11-12:
我也发现了，编译softethervpn报错。

liuluo1979, 2019-11-12:
还有ssrplus，ss能正常使用，v2ray不能用。上个月底编译的就有这个问题了。

liuluo1979, 2019-11-12:
新功能越加越多，老功能各种错……

mg6630, 2019-11-12:
> 我也发现了，编译softethervpn报错。
> 上个月编译正常的

还有重启，wan口重启， 但是这个还没确定是硬件问题还是软件问题，因为编译不出来新的固件。

liuluo1979, 2019-11-12:
/home/shaun/projects/ctcc/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/include/sys/poll.h:1:2: warning: #warning redirecting incorrect #include <sys/poll.h> to <poll.h> [-Wcpp]
 #warning redirecting incorrect #include <sys/poll.h> to <poll.h>
  ^~~~~~~
rm -f tmp/as/vpnserver.a
ar r tmp/as/vpnserver.a tmp/objs/Mayaqua/Cfg.o tmp/objs/Mayaqua/Encrypt.o tmp/objs/Mayaqua/FileIO.o tmp/objs/Mayaqua/Internat.o tmp/objs/Mayaqua/Kernel.o tmp/objs/Mayaqua/Mayaqua.o tmp/objs/Mayaqua/Memory.o tmp/objs/Mayaqua/Microsoft.o tmp/objs/Mayaqua/Network.o tmp/objs/Mayaqua/Object.o tmp/objs/Mayaqua/OS.o tmp/objs/Mayaqua/Pack.o tmp/objs/Mayaqua/Secure.o tmp/objs/Mayaqua/Str.o tmp/objs/Mayaqua/Table.o tmp/objs/Mayaqua/TcpIp.o tmp/objs/Mayaqua/Tick64.o tmp/objs/Mayaqua/Tracking.o tmp/objs/Mayaqua/Unix.o tmp/objs/Mayaqua/Win32.o tmp/objs/Cedar/Account.o tmp/objs/Cedar/Admin.o tmp/objs/Cedar/AzureClient.o tmp/objs/Cedar/AzureServer.o tmp/objs/Cedar/Bridge.o tmp/objs/Cedar/BridgeUnix.o tmp/objs/Cedar/BridgeWin32.o tmp/objs/Cedar/Cedar.o tmp/objs/Cedar/CedarPch.o tmp/objs/Cedar/Client.o tmp/objs/Cedar/CM.o tmp/objs/Cedar/Command.o tmp/objs/Cedar/Connection.o tmp/objs/Cedar/Console.o tmp/objs/Cedar/Database.o tmp/objs/Cedar/DDNS.o tmp/objs/Cedar/EM.o tmp/objs/Cedar/EtherLog.o tmp/objs/Cedar/Hub.o tmp/objs/Cedar/Interop_OpenVPN.o tmp/objs/Cedar/Interop_SSTP.o tmp/objs/Cedar/IPsec.o tmp/objs/Cedar/IPsec_EtherIP.o tmp/objs/Cedar/IPsec_IKE.o tmp/objs/Cedar/IPsec_IkePacket.o tmp/objs/Cedar/IPsec_IPC.o tmp/objs/Cedar/IPsec_L2TP.o tmp/objs/Cedar/IPsec_PPP.o tmp/objs/Cedar/IPsec_Win7.o tmp/objs/Cedar/Layer3.o tmp/objs/Cedar/Link.o tmp/objs/Cedar/Listener.o tmp/objs/Cedar/Logging.o tmp/objs/Cedar/Nat.o tmp/objs/Cedar/NativeStack.o tmp/objs/Cedar/NM.o tmp/objs/Cedar/NullLan.o tmp/objs/Cedar/Protocol.o tmp/objs/Cedar/Radius.o tmp/objs/Cedar/Remote.o tmp/objs/Cedar/Sam.o tmp/objs/Cedar/SecureInfo.o tmp/objs/Cedar/SecureNAT.o tmp/objs/Cedar/SeLowUser.o tmp/objs/Cedar/Server.o tmp/objs/Cedar/Session.o tmp/objs/Cedar/SM.o tmp/objs/Cedar/SW.o tmp/objs/Cedar/UdpAccel.o tmp/objs/Cedar/UT.o tmp/objs/Cedar/VG.o tmp/objs/Cedar/Virtual.o tmp/objs/Cedar/VLan.o tmp/objs/Cedar/VLanUnix.o tmp/objs/Cedar/VLanWin32.o tmp/objs/Cedar/WaterMark.o tmp/objs/Cedar/WebUI.o tmp/objs/Cedar/WinUi.o tmp/objs/Cedar/Wpc.o tmp/objs/vpnserver.o
ar: creating tmp/as/vpnserver.a
ranlib tmp/as/vpnserver.a
make[4]: Warning: File 'src/bin/BuiltHamcoreFiles/unix/hamcore.se2' has modification time 86351 s in the future
x86_64-openwrt-linux-musl-gcc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -I/home/shaun/projects/ctcc/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/include -I/home/shaun/projects/ctcc/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/include -iremap/home/shaun/projects/ctcc/build_dir/target-x86_64_musl/softethervpn/v4.29-9680:v4.29-9680 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DSSL_OP_NO_SSLv3  -I/home/shaun/projects/ctcc/staging_dir/target-x86_64_musl/usr/include -I/home/shaun/projects/ctcc/staging_dir/target-x86_64_musl/include -I/home/shaun/projects/ctcc/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/usr/include -I/home/shaun/projects/ctcc/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/include/fortify -I/home/shaun/projects/ctcc/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/include -I/home/shaun/projects/ctcc/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/include -I/home/shaun/projects/ctcc/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/include -L/home/shaun/projects/ctcc/staging_dir/target-x86_64_musl/usr/lib -L/home/shaun/projects/ctcc/staging_dir/target-x86_64_musl/lib -L/home/shaun/projects/ctcc/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/usr/lib -L/home/shaun/projects/ctcc/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/lib -L/home/shaun/projects/ctcc/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/lib -Wl,-rpath-link=/home/shaun/projects/ctcc/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/lib -L/home/shaun/projects/ctcc/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/lib -Wl,-rpath-link=/home/shaun/projects/ctcc/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/lib -znow -zrelro -DNDEBUG -DVPN_SPEED -DUNIX -DUNIX_LINUX -DCPU_64 -D_REENTRANT -DREENTRANT -D_THREAD_SAFE -D_THREADSAFE -DTHREAD_SAFE -DTHREADSAFE -D_FILE_OFFSET_BITS=64 -I./src/ -I./src/Cedar/ -I./src/Mayaqua/ -O2 -fsigned-char tmp/objs/Mayaqua/Cfg.o tmp/objs/Mayaqua/Encrypt.o tmp/objs/Mayaqua/FileIO.o tmp/objs/Mayaqua/Internat.o tmp/objs/Mayaqua/Kernel.o tmp/objs/Mayaqua/Mayaqua.o tmp/objs/Mayaqua/Memory.o tmp/objs/Mayaqua/Microsoft.o tmp/objs/Mayaqua/Network.o tmp/objs/Mayaqua/Object.o tmp/objs/Mayaqua/OS.o tmp/objs/Mayaqua/Pack.o tmp/objs/Mayaqua/Secure.o tmp/objs/Mayaqua/Str.o tmp/objs/Mayaqua/Table.o tmp/objs/Mayaqua/TcpIp.o tmp/objs/Mayaqua/Tick64.o tmp/objs/Mayaqua/Tracking.o tmp/objs/Mayaqua/Unix.o tmp/objs/Mayaqua/Win32.o tmp/objs/Cedar/Account.o tmp/objs/Cedar/Admin.o tmp/objs/Cedar/AzureClient.o tmp/objs/Cedar/AzureServer.o tmp/objs/Cedar/Bridge.o tmp/objs/Cedar/BridgeUnix.o tmp/objs/Cedar/BridgeWin32.o tmp/objs/Cedar/Cedar.o tmp/objs/Cedar/CedarPch.o tmp/objs/Cedar/Client.o tmp/objs/Cedar/CM.o tmp/objs/Cedar/Command.o tmp/objs/Cedar/Connection.o tmp/objs/Cedar/Console.o tmp/objs/Cedar/Database.o tmp/objs/Cedar/DDNS.o tmp/objs/Cedar/EM.o tmp/objs/Cedar/EtherLog.o tmp/objs/Cedar/Hub.o tmp/objs/Cedar/Interop_OpenVPN.o tmp/objs/Cedar/Interop [... cut ...]
make[2]: *** [package/feeds/packages/softethervpn/compile] Error 2
make[2]: Leaving directory '/home/shaun/projects/ctcc'
package/Makefile:103: recipe for target '/home/shaun/projects/ctcc/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/shaun/projects/ctcc/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/shaun/projects/ctcc'
/home/shaun/projects/ctcc/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

xiangfeigao, 2019-11-12:
估计是 www.softether-download.com官网无法打开的原因

mg6630, 2019-11-13:
> 估计是 [www.softether-download.com官网无法打开的原因](http://www.softether-download.com%E5%AE%98%E7%BD%91%E6%97%A0%E6%B3%95%E6%89%93%E5%BC%80%E7%9A%84%E5%8E%9F%E5%9B%A0)

应该不是，我从别的源码能编译出这个插件

P0lari5, 2019-11-13:
> 选择luci-app-softethervpn 插件 编译就会报错，不选择的话可以编译成功。4.9 4.19 ubuntu14.04. 16.04 18.04 都试过，不选择这个插件都能成功，选了就报错。 [...]

修改/lede/feeds/packages/net/softethervpn/Makefile
PKG_NAME:=softethervpn
PKG_VERSION:=4.29-9680
PKG_VERREL:=rtm
PKG_VERDATE:=2019.02.28
PKG_RELEASE:=1

PKG_SOURCE:=softether-src-v$(PKG_VERSION)-$(PKG_VERREL).tar.gz
PKG_SOURCE_URL:=http://www.softether-download.com/files/softether/v$(PKG_VERSION)-$(PKG_VERREL)-$(PKG_VERDATE)-tree/Source_Code/
PKG_HASH:=e6035fa7d9aaf59bdb342cd7ab5ecfdff89811a875f62a3230208cdc8a4e26e4

shenyz, 2019-11-13:
不选 luci-app-softethervpn  到network--》vpn--》选择softethervpn5 ，就可以了，内核用4.9的

shenyz, 2019-11-13:
> > 我也发现了，编译softethervpn报错。
> > 上个月编译正常的
> 
> 还有重启，wan口重启， 但是这个还没确定是硬件问题还是软件问题，因为编译不出来新的固件。

出现过一次夜里wan口重启，2-4天会没响应，正在排查，怀疑是docker的锅，以前没有加入docker的时候一切正常，正在观察 好几个礼拜了

mg6630, 2019-11-13:
> 不选 luci-app-softethervpn 到network--》vpn--》选择softethervpn5 ，就可以了，内核用4.9的
> 或者修改 luci-app-softethervpn 的makefile 改依赖包为softethervpn5

好的，谢谢，我再试一试，之前编译5好像也会出错。顺便问一下，有5没5使用上有什么区别？

mg6630, 2019-11-13:
> 不选 luci-app-softethervpn 到network--》vpn--》选择softethervpn5 ，就可以了，内核用4.9的
> 或者修改 luci-app-softethervpn 的makefile 改依赖包为softethervpn5

是否可以只编译server，不选择brigde和client.因为这两个在路由上也用不到。

shenyz, 2019-11-13:
> > 不选 luci-app-softethervpn 到network--》vpn--》选择softethervpn5 ，就可以了，内核用4.9的
> > 或者修改 luci-app-softethervpn 的makefile 改依赖包为softethervpn5
> 
> 是否可以只编译server，不选择brigde和client.因为这两个在路由上也用不到。

是的 可以的额 我就只编译的server

TerryWang159, 2019-11-13:
我也是遇到同样的问题 ，编译了两次，都是softethervpn 模块报错， 最后取消选择softethervpn 才编译通过。期待大神更新一下库的代码。辛苦了。

lzx2009vb, 2019-11-13:
> 不选 luci-app-softethervpn 到network--》vpn--》选择softethervpn5 ，就可以了，内核用4.9的
> 或者修改 luci-app-softethervpn 的makefile 改依赖包为softethervpn5

我也改了，还是不行，编译5好像也会报错。

L-vivi, 2019-11-14:
搞乐几天了，我这也才发现取消luci-app-softethervpn编译就通过了

xinxijishuwyq, 2019-11-14:
用openwrt官方的最新版softethervpn5
https://github.com/openwrt/packages/tree/master/net/softethervpn5
记得把4的选项去干净

mg6630, 2019-11-18:
已解决

closed 2019-11-18 by mg6630