coolsnowwolf/lede #2064: R619AC 最新源码，编译失败

joybin08, 2019-11-13:
4.14,4.19内核都试过，完全默认设置，编译也会失败，系统Ubuntu18.04 x64
失败部分日志如下：
make[5]: Leaving directory '/home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.19.82'
cmp -s /home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.19.82/vmlinux /home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/vmlinux-initramfs.debug || { arm-openwrt-linux-muslgnueabi-objcopy -O binary -R .reginfo -R .notes -R .note -R .comment -R .mdebug -R .note.gnu.build-id -S /home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.19.82/vmlinux /home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/vmlinux-initramfs; arm-openwrt-linux-muslgnueabi-objcopy -R .reginfo -R .notes -R .note -R .comment -R .mdebug -R .note.gnu.build-id -S /home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.19.82/vmlinux /home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/vmlinux-initramfs.elf; cp -fpR /home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.19.82/vmlinux /home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/vmlinux-initramfs.debug;  cp -fpR /home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.19.82/arch/arm/boot//zImage /home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/zImage-initramfs;   cp -fpR /home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.19.82/arch/arm/boot//Image /home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/Image-initramfs;  }
touch /home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.19.82/.image
make -C image compile install TARGET_BUILD=
make[5]: Entering directory '/home/user/lede/target/linux/ipq40xx/image'
make[5]: Nothing to be done for 'compile'.
arm-openwrt-linux-muslgnueabi-cpp -nostdinc -x assembler-with-cpp -I/home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.19.82/arch/arm/boot/dts -I/home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.19.82/arch/arm/boot/dts/include -I/home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.19.82/include/ -undef -D__DTS__  -o /home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/image-qcom-ipq4018-jalapeno.dtb.tmp /home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.19.82/arch/arm/boot/dts/qcom-ipq4018-jalapeno.dts
/home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.19.82/scripts/dtc/dtc -O dtb -i/home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.19.82/arch/arm/boot/dts/ -Wno-unit_address_vs_reg -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address  -o /home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/image-qcom-ipq4018-jalapeno.dtb /home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/image-qcom-ipq4018-jalapeno.dtb.tmp
rm -f /home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/image-qcom-ipq4018-jalapeno.dtb.tmp
arm-openwrt-linux-muslgnueabi-cpp -nostdinc -x assembler-with-cpp -I/home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.19.82/arch/arm/boot/dts -I/home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.19.82/arch/arm/boot/dts/include -I/home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.19.82/include/ -undef -D__DTS__  -o /home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/image-qcom-ipq4019-map-ac2200.dtb.tmp /home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.19.82/arch/arm/boot/dts/qcom-ipq4019-map-ac2200.dts
/home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.19.82/scripts/dtc/dtc -O dtb -i/home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.19.82/arch/arm/boot/dts/ -Wno-unit_address_vs_reg -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address  -o /home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/image-qcom-ipq4019-map-ac2200.dtb /home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/image-qcom-ipq4019-map-ac2200.dtb.tmp
rm -f /home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/image-qcom-ipq4019-map-ac2200.dtb.tmp
arm-openwrt-linux-muslgnueabi-cpp -nostdinc -x assembler-with-cpp -I/home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.19.82/arch/arm/boot/dts -I/home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.19.82/arch/arm/boot/dts/include -I/home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.19.82/include/ -undef -D__DTS__  -o /home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/image-qcom-ipq4018-rt-ac58u.dtb.tmp /home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.19.82/arch/arm/boot/dts/qcom-ipq4018-rt-ac58u.dts
/home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.19.82/scripts/dtc/dtc -O dtb -i/home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.19.82/arch/arm/boot/dts/ -Wno-unit_address_vs_reg -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address  -o /home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/image-qcom-ipq4018-rt-ac58u.dtb /home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/image-qcom-ipq4018-rt-ac58u.dtb.tmp
rm -f /home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/image-qcom-ipq4018-rt-ac58u.dtb.tmp
arm-openwrt-linux-muslgnueabi-cpp -nostdinc -x assembler-with-cpp -I/home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.19.82/arch/arm/boot/dts -I/home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.19.82/arch/arm/boot/dts/include -I/home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.19.82/include/ -undef -D__DTS__  -o /home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/image-qcom-ipq4019-rt-acrh17.dtb.tmp /home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.19.82/arch/arm/boot/dts/qcom-ipq4019-rt-acrh17.dts
arm-openwrt-linux-muslgnueabi-cpp: error: /home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.19.82/arch/arm/boot/dts/qcom-ipq4019-rt-acrh17.dts: No such file or directory
arm-openwrt-linux-muslgnueabi-cpp: warning: '-x assembler-with-cpp' after last input file has no effect
arm-openwrt-linux-muslgnueabi-cpp: fatal error: no input files
compilation terminated.
Makefile:349: recipe for target '/home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/image-qcom-ipq4019-rt-acrh17.dtb' failed
make[5]: *** [/home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/image-qcom-ipq4019-rt-acrh17.dtb] Error 1
make[5]: Leaving directory '/home/user/lede/target/linux/ipq40xx/image'
Makefile:25: recipe for target 'install' failed
make[4]: *** [install] Error 2
make[4]: Leaving directory '/home/user/lede/target/linux/ipq40xx'
Makefile:13: recipe for target 'install' failed
make[3]: *** [install] Error 2
make[3]: Leaving directory '/home/user/lede/target/linux'
time: target/linux/install#46.03#98.32#161.79
[... cut ...] r target 'target/linux/install' failed
make[2]: *** [target/linux/install] Error 2
make[2]: Leaving directory '/home/user/lede'
target/Makefile:19: recipe for target '/home/user/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.target_install' failed
make[1]: *** [/home/user/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/user/lede'
/home/user/lede/include/toplevel.mk:216: recipe for target 'world' failed

Her0R0cket, 2019-11-13:
1,你是不是没选具体的设备,选的是multiple devices(看到他处理了好多的dts)
2,看了下文件结构,r619ac和acrh17的dts只存在于4.14内核目录下,我不知道这些dts具体是怎么处理的,但4.14,应该是能成功的
如果4.14失败,最好是把记录贴出来给大家看下

vincentchen, 2019-11-14:
[image]
[image]

hellodong, 2019-11-16:
我也遇到编译不过去情况，跟楼上情况类似：
Exportable Squashfs 4.0 filesystem, xz compressed, data block size 262144
        compressed data, compressed metadata, compressed fragments, no xattrs
        duplicates are removed
Filesystem size 16342.07 Kbytes (15.96 Mbytes)
        35.93% of uncompressed filesystem size (45484.25 Kbytes)
Inode table size 25062 bytes (24.47 Kbytes)
        21.02% of uncompressed inode table size (119221 bytes)
Directory table size 34294 bytes (33.49 Kbytes)
        40.74% of uncompressed directory table size (84180 bytes)
Number of duplicate files found 619
Number of inodes 3547
Number of files 2897
Number of fragments 70
Number of symbolic links  407
Number of device nodes 1
Number of fifo nodes 0
Number of socket nodes 0
Number of directories 242
Number of ids (unique uids + gids) 1
Number of uids 1
        root (0)
Number of gids 1
        root (0)
rm -f /home/dong/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-initramfs-fit-zImage.itb
cp /home/dong/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/zImage-initramfs /home/dong/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-initramfs-fit-zImage.itb
/home/dong/lede/scripts/mkits.sh -D p2w_r619ac -o /home/dong/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-initramfs-fit-zImage.itb.its -k /home/dong/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-initramfs-fit-zImage.itb -d /home/dong/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.14.152/arch/arm/boot/dts/qcom-ipq4019-r619ac.dtb -C none -a 0x80208000 -e 0x80208000 -c config@10 -A arm -v 4.14.152
PATH=/home/dong/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.14.152/scripts/dtc:/home/dong/lede/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-7.4.0_musl_eabi/bin:/home/dong/lede/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-7.4.0_musl_eabi/bin:/home/dong/lede/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-7.4.0_musl_eabi/bin:/home/dong/lede/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-7.4.0_musl_eabi/bin:/home/dong/lede/staging_dir/host/bin:/home/dong/lede/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-7.4.0_musl_eabi/bin:/home/dong/lede/staging_dir/host/bin:/home/dong/lede/staging_dir/host/bin:/usr/local/bin:/usr/bin:/usr/local/sbin:/usr/sbin:/home/dong/x-tools/arm-2440-linux-gnueabi/bin:/home/dong/.local/bin:/home/dong/bin mkimage -f /home/dong/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-initramfs-fit-zImage.itb.its /home/dong/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-initramfs-fit-zImage.itb.new
/home/dong/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-initramfs-fit-zImage.itb.new.tmp: Warning (unit_address_vs_reg): Node /images/kernel@1 has a unit name, but no reg property
/home/dong/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-initramfs-fit-zImage.itb.new.tmp: Warning (unit_address_vs_reg): Node /images/kernel@1/hash@1 has a unit name, but no reg property
/home/dong/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-initramfs-fit-zImage.itb.new.tmp: Warning (unit_address_vs_reg): Node /images/kernel@1/hash@2 has a unit name, but no reg property
/home/dong/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-initramfs-fit-zImage.itb.new.tmp: Warning (unit_address_vs_reg): Node /images/fdt@1 has a unit name, but no reg property
/home/dong/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-initramfs-fit-zImage.itb.new.tmp: Warning (unit_address_vs_reg): Node /images/fdt@1/hash@1 has a unit name, but no reg property
/home/dong/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-initramfs-fit-zImage.itb.new.tmp: Warning (unit_address_vs_reg): Node /images/fdt@1/hash@2 has a unit name, but no reg property
/home/dong/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-initramfs-fit-zImage.itb.new.tmp: Warning (unit_address_vs_reg): Node /configurations/config@10 has a unit name, but no reg property
FIT description: ARM OpenWrt FIT (Flattened Image Tree)
Created:         Tue Nov 12 14:51:28 2019
 Image 0 (kernel@1)
  Description:  ARM OpenWrt Linux-4.14.152
  Created:      Tue Nov 12 14:51:28 2019
  Type:         Kernel Image
  Compression:  uncompressed
  Data Size:    18542704 Bytes = 18108.11 KiB = 17.68 MiB
  Architecture: ARM
  OS:           Linux
  Load Address: 0x80208000
  Entry Point:  0x80208000
  Hash algo:    crc32
  Hash value:   f0967f63
  Hash algo:    sha1
  Hash value:   7d3019328d1c5a58c151c9f3afeded5514cae219
 Image 1 (fdt@1)
  Description:  ARM OpenWrt p2w_r619ac device tree blob
  Created:      Tue Nov 12 14:51:28 2019
  Type:         Flat Device Tree
  Compression:  uncompressed
  Data Size:    16818 Bytes = 16.42 KiB = 0.02 MiB
  Architecture: ARM
  Hash algo:    crc32
  Hash value:   2818eabf
  Hash algo:    sha1
  Hash value:   8d65353853c09fd6520e32038dd0fa97d3d249ec
[... cut ...]
make[4]: *** [install] Error 2
make[4]: Leaving directory `/home/dong/lede/target/linux/ipq40xx'
make[3]: *** [install] Error 2
make[3]: Leaving directory `/home/dong/lede/target/linux'
time: target/linux/install#494.05#73.83#592.31
make[2]: *** [target/linux/install] Error 2
make[2]: Leaving directory `/home/dong/lede'
make[1]: *** [/home/dong/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.target_install] Error 2
make[1]: Leaving directory `/home/dong/lede'
make: *** [world] Error 2

AmadeusGhost, 2019-11-16:
> 我也遇到编译不过去情况，跟楼上情况类似：
> Exportable Squashfs 4.0 filesystem, xz compressed, data block size 262144
> compressed data, compressed metadata, compressed fragments, no xattrs
> duplicates are removed
> Filesystem [...]

你依赖没装全：```sudo apt-get install device-tree-compiler```

closed 2019-11-28 by coolsnowwolf