coolsnowwolf/lede #2065: v2ray版本已经更新到4.21.3，麻烦大雕更新一下

letmekillthemall, 2019-11-13:
v2ray版本已经更新到4.21.3，麻烦更新一下，幸苦了

outemancut, 2019-11-13:
越更新越大！有什么用？除了占地方 我就差点装不下了

liuluo1979, 2019-11-13:
v2ray能用？

Kinodi, 2019-11-13:
[v2ray_v4.21.3_x86_64.zip](https://github.com/coolsnowwolf/lede/files/3840786/v2ray_v4.21.3_x86_64.zip)

iamggs, 2019-11-13:
> [v2ray_v4.21.3_x86_64.zip](https://github.com/coolsnowwolf/lede/files/3840786/v2ray_v4.21.3_x86_64.zip)

感谢感谢，体积几乎翻了一番，已顺手更新。

Kinodi, 2019-11-13:
> > [v2ray_v4.21.3_x86_64.zip](https://github.com/coolsnowwolf/lede/files/3840786/v2ray_v4.21.3_x86_64.zip)
> 
> 感谢感谢，体积几乎翻了一番，已顺手更新。

因为没有压缩

Zack0617, 2019-11-13:
> > [v2ray_v4.21.3_x86_64.zip](https://github.com/coolsnowwolf/lede/files/3840786/v2ray_v4.21.3_x86_64.zip)
> 
> 感谢感谢，体积几乎翻了一番，已顺手更新。
请教一下，如何自己更新最新版本的v2ray到L大的源码中编译啊？我用的软路由，体积倒不是问题，不知道怎么更新，谢谢！

iamggs, 2019-11-14:
> > > [v2ray_v4.21.3_x86_64.zip](https://github.com/coolsnowwolf/lede/files/3840786/v2ray_v4.21.3_x86_64.zip)
> > 
> > 
> > 感谢感谢，体积几乎翻了一番，已顺手更新。
> 
> 因为没有压缩

这是你自己编译的么？还是有其他现成的渠道？

Zack0617, 2019-11-14:
@Kinodi 请教一下我如何将最新版本更新到l大的源码中编译，谢谢!

KFERMercer, 2019-11-14:
> @Kinodi 请教一下我如何将最新版本更新到l大的源码中编译，谢谢!

https://github.com/KFERMercer/openwrt-v2ray

lbbboy, 2019-11-14:
@Kinodi 
V2Ray 4.21.3 (V2Fly, a community-driven edition of V2Ray.) OpenWrt - Release 1 Started!
运行不起来！！

xinxijishuwyq, 2019-11-16:
> @Kinodi
> V2Ray 4.21.3 (V2Fly, a community-driven edition of V2Ray.) OpenWrt - Release 1 Started!
> 运行不起来！！

那个项目的路径和lean的不大一样，要改一下

Kinodi, 2019-11-17:
> @Kinodi
> V2Ray 4.21.3 (V2Fly, a community-driven edition of V2Ray.) OpenWrt - Release 1 Started!
> 运行不起来！！

[image]
[image]
没问题呀！  ？

lbbboy, 2019-11-18:
> > @Kinodi
> > V2Ray 4.21.3 (V2Fly, a community-driven edition of V2Ray.) OpenWrt - Release 1 Started!
> > 运行不起来！！
> 
> [image]
> [image]
> 没问题呀！ ？

不用自己编的，直接用官方的替换就行了。普通路由可以用
符号连接解决。更新方便。

lbbboy, 2019-11-18:
> > @Kinodi 请教一下我如何将最新版本更新到l大的源码中编译，谢谢!
> 
> https://github.com/KFERMercer/openwrt-v2ray

V2Ray 4.21.3 (V2Fly, a community-driven edition of V2Ray.) OpenWrt - Release 1 Started!

只有这样的启动提示，mtk7621路由。没有进程。

closed 2019-12-09 by letmekillthemall