coolsnowwolf/lede #2071: SSR 插件想要设置超过8线程并发怎么改？

alechsu, 2019-11-14:
界面上只允许最多8线程并发，想要设置16线程。
[image]

lifetyper, 2019-11-14:
你什么样的CPU需要16线程？

hcym, 2019-11-14:
好像区别不大

alechsu, 2019-11-15:
> 你什么样的CPU需要16线程？

公司300台设备环境下，并发量巨大。

alechsu, 2019-11-15:
> 好像区别不大

我是公司环境并发量比较大，做过测试从默认2线程到8线程，开网页效果明显。

lifetyper, 2019-11-15:
> > 你什么样的CPU需要16线程？
> 
> 公司300台设备环境下，并发量巨大。

两个问题：
1.你SSR服务器端的CPU跑满了吗？Python版只能跑单核，瓶颈到底在哪边还不一定。
2.你路由器的CPU现在8核8线程已经跑满了吗？

alechsu, 2019-11-15:
> > > 你什么样的CPU需要16线程？
> > 
> > 
> > 公司300台设备环境下，并发量巨大。
> 
> 两个问题：
> 1.你服务器端的CPU跑满了吗？
> 2.你路由器的CPU现在8核8线程已经跑满了吗？
> 
> 如果路由器CPU没这么多核的话，我有点怀疑强开多线程是否有效。

cpu没跑满，但是我觉得墙针对每个连接也有带宽限制，事实是我从默认2线程改到8线程，开网页速度是有明显提高的。

lifetyper, 2019-11-15:
> > > > 你什么样的CPU需要16线程？
> > > 
> > > 
> > > 公司300台设备环境下，并发量巨大。
> > 
> > 
> > 两个问题：
> > 1.你服务器端的CPU跑满了吗？
> > 2.你路由器的CPU现在8核8线程已经跑满了吗？
> > 如果路由器CPU没这么多核的话，我有点怀疑强开多线程是否有效。
> 
> cpu没跑满，但是我觉得墙针对每个连接也有带宽限制，事实是我从默认2线程改到8线程，开网页速度是有明显提高的。

我觉得你的问题需要参考这个issue：
https://github.com/coolsnowwolf/lede/issues/1607

可能是客户端的打开文件数被限制了

alechsu, 2019-11-15:
> > > > > 你什么样的CPU需要16线程？
> > > > 
> > > > 
> > > > 公司300台设备环境下，并发量巨大。
> > > 
> > > 
> > > 两个问题：
> > > 1.你服务器端的CPU跑满了吗？
> > > 2.你路由器的CPU现在8核8线程已经跑满了吗？
> > > 如果路由器CPU没这么多核的话，我有点怀疑强开多线程是否有效。
> > 
> > 
> > cpu没跑满，但是我觉得墙针对每个连接也有带宽限制，事实是我从默认2线程改到8线程，开网页速度是有明显提高的。
> > 不是之前那个打开文件数的坑吗？

之前文件数的坑是直接让插件不可用了，客户端怎么点都打不开网页，现在并发数低的话发现在几个网页之间跳转时候经常会卡几秒。

z826540272, 2019-11-29:
那个是CPU线程数,不是TCP连接并发数,不是一回事

closed 2019-12-30 by coolsnowwolf