coolsnowwolf/lede #2078: 默认配置x64编译失败

jzhou404, 2019-11-15:
X86， MT7621， MT7688都可以编译顺利通过

就是选择X64失败，而且是删掉.config, 用make defconfig生成默认后， 很奇怪。

jzhou404, 2019-11-15:
error message as following:

mkdir -p /home/jzhou/lede/staging_dir/target-x86_64_musl/root-x86/stamp
echo '1.8.2' | cmp -s - /home/jzhou/lede/staging_dir/target-x86_64_musl/pkginfo/libip6tc.version || echo '1.8.2' > /home/jzhou/lede/staging_dir/target-x86_64_musl/pkginfo/libip6tc.version
SHELL= flock /home/jzhou/lede/tmp/.root-copy.flock -c 'cp -fpR /home/jzhou/lede/build_dir/target-x86_64_musl/linux-x86_64/iptables-1.8.2/.pkgdir/libip6tc/. /home/jzhou/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/jzhou/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.libip6tc_installed
echo "libip6tc" >> /home/jzhou/lede/staging_dir/target-x86_64_musl/pkginfo/iptables.default.install
mkdir -p /home/jzhou/lede/staging_dir/target-x86_64_musl/root-x86/stamp
echo '1.8.2' | cmp -s - /home/jzhou/lede/staging_dir/target-x86_64_musl/pkginfo/libxtables.version || echo '1.8.2' > /home/jzhou/lede/staging_dir/target-x86_64_musl/pkginfo/libxtables.version
SHELL= flock /home/jzhou/lede/tmp/.root-copy.flock -c 'cp -fpR /home/jzhou/lede/build_dir/target-x86_64_musl/linux-x86_64/iptables-1.8.2/.pkgdir/libxtables/. /home/jzhou/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/jzhou/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.libxtables_installed
echo "libxtables" >> /home/jzhou/lede/staging_dir/target-x86_64_musl/pkginfo/iptables.default.install
make[3]: Leaving directory '/home/jzhou/lede/package/network/utils/iptables'
time: package/network/utils/iptables/compile#25.68#3.30#32.06
make[2]: Leaving directory '/home/jzhou/lede'
package/Makefile:103: recipe for target '/home/jzhou/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/jzhou/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/jzhou/lede'
Build failed - please re-run with -j1 to see the real error message
/home/jzhou/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 1
jzhou@openwrt:~/lede$

EnnawYang, 2019-11-15:
我也是 80211过不去

xinxijishuwyq, 2019-11-16:
试试用openwrt官方最新的替换过来(

czjesda, 2019-11-18:
> 
> 
> 试试用openwrt官方最新的替换过来(

openwrt的mac80211文件夹拷贝过后还是不行么？应该是如何替换呢？

foyo23, 2019-11-21:
我也是 mac80211编译通不过

czjesda, 2019-11-21:
> 
> 
> 我也是 mac80211编译通不过

试试这个方法，重新git clone源码，然后make menuconfig后什么都不动，然后make download V=s下载DL包，如果有就忽略，最后运行make -j1 V=s看能不能成功。如果成功你在选择你要按照的插件来再次编译

EnnawYang, 2019-11-21:
> > 我也是 mac80211编译通不过
> 
> 试试这个方法，重新git clone源码，然后make menuconfig后什么都不动，然后make download V=s下载DL包，如果有就忽略，最后运行make -j1 V=s看能不能成功。如果成功你在选择你要按照的插件来再次编译

一样的

czjesda, 2019-11-21:
> 
> 
> > > 我也是 mac80211编译通不过
> > 
> > 
> > 试试这个方法，重新git clone源码，然后make menuconfig后什么都不动，然后make download V=s下载DL包，如果有就忽略，最后运行make -j1 V=s看能不能成功。如果成功你在选择你要按照的插件来再次编译
> 
> 一样的 没用

那就只能使用终极大法了，把mac80211的组件取消编译。这方法只可以使用在软路由，如果要用到WIFI那就不知道了

EnnawYang, 2019-11-21:
> > > > 我也是 mac80211编译通不过
> > > 
> > > 
> > > 试试这个方法，重新git clone源码，然后make menuconfig后什么都不动，然后make download V=s下载DL包，如果有就忽略，最后运行make -j1 V=s看能不能成功。如果成功你在选择你要按照的插件来再次编译
> > 
> > 
> > 一样的 没用
> 
> 那就只能使用终极大法了，把mac80211的组件取消编译。这方法只可以使用在软路由，如果要用到WIFI那就不知道了

我直接把内核改成4.9了 可以成功编译

jzhou404, 2019-11-28:
发现问题是 /lede/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-4.19.82， 但编译代码需要linux-4.19.85，是源代码的问题， 改名就可以通过 @coolsnowwolf

closed 2019-11-28 by coolsnowwolf

jzhou404, 2019-11-29:
编译切换内核：
 
/lede/target/linux/x86/Makefile

编辑
KERNEL_PATCHVER:=4.19改成KERNEL_PATCHVER:=4.9

完美解决