coolsnowwolf/lede #2081: 请问从的lede和openwrt有什么区别？

viagram, 2019-11-16:
一直用你的lede源码编译固件，刚看到你也在维护一个openwrt，请问它们有什么区别？谢谢！

outemancut, 2019-11-16:
LEDE是17.01后面合并了而已 现在都叫OPENWRT  没有什么区别！

viagram, 2019-11-16:
> LEDE是17.01后面合并了而已 现在都叫OPENWRT 没有什么区别！
这个我知道，我是问@coolsnowwolf 他这两份源码的区别。大概了解了一下，这样形容：lede是开发版，openwrt是稳定版。不知道这样理解对不对。

yinzhefuyun, 2019-11-16:
> > LEDE是17.01后面合并了而已 现在都叫OPENWRT 没有什么区别！
> > 这个我知道，我是问@coolsnowwolf 他这两份源码的区别。大概了解了一下，这样形容：lede是开发版，openwrt是稳定版。不知道这样理解对不对。

差不多是这个意思lede和openwrt内容都是openwrt的源码，只不过lede每天都会更新，openwrt一年更新一次

Zack0617, 2019-11-17:
> > > LEDE是17.01后面合并了而已 现在都叫OPENWRT 没有什么区别！
> > > 这个我知道，我是问@coolsnowwolf 他这两份源码的区别。大概了解了一下，这样形容：lede是开发版，openwrt是稳定版。不知道这样理解对不对。
> 
> 差不多是这个意思lede和openwrt内容都是openwrt的源码，只不过lede每天都会更新，openwrt一年更新一次

这个解释…………

viagram, 2019-11-17:
@coolsnowwolf 他这几天好像这些忙，lede很少更新了，前段时间一天提交好多。
希望有时间来介绍一下。谢谢。

smallprogram, 2019-11-18:
lean是基于Openwrt的分支版本，加入一些自己的东西比如SSR plus等，OpenWrt的源代码也是每天都在更新，维护人数也相当多。目前原版OpenWrt稳定版本是18.06，想要稳直接上原版，自己写插件。

viagram, 2019-11-18:
> lean是基于Openwrt的分支版本，加入一些自己的东西比如SSR plus等，OpenWrt的源代码也是每天都在更新，维护人数也相当多。目前原版OpenWrt稳定版本是18.06，想要稳直接上原版，自己写插件。

我说的lede和openwrt指的是@coolsnowwolf 下的lede和openwrt。

smallprogram, 2019-11-18:
> > lean是基于Openwrt的分支版本，加入一些自己的东西比如SSR plus等，OpenWrt的源代码也是每天都在更新，维护人数也相当多。目前原版OpenWrt稳定版本是18.06，想要稳直接上原版，自己写插件。
> 
> 我说的lede和openwrt指的是@coolsnowwolf 下的lede和openwrt。

lede是尝鲜版，openwrt是稳定版

viagram, 2019-11-20:
> > > lean是基于Openwrt的分支版本，加入一些自己的东西比如SSR plus等，OpenWrt的源代码也是每天都在更新，维护人数也相当多。目前原版OpenWrt稳定版本是18.06，想要稳直接上原版，自己写插件。
> > 
> > 
> > 我说的lede和openwrt指的是@coolsnowwolf 下的lede和openwrt。
> 
> lede是尝鲜版，openwrt是稳定版

谢谢解答

closed 2019-11-20 by viagram

waynon, 2021-09-03:
我发现coolsnowwolf的lede仓库没有fork之前的commit历史，那怎么拉取上游的官方openwrt中的commit？

ForgotFun, 2022-06-09:
大家如果对OpenWrt开发感兴趣的话，可以关注下佐大的OpenWrt培训班。
https://forgotfun.org/2018/04/openwrt-training-2018.html