coolsnowwolf/lede #2082: x86编译失败，最近更新的都编译不成功

rhww, 2019-11-16:
remote: Enumerating objects: 22, done.
remote: Counting objects: 100% (22/22), done.
remote: Compressing objects: 100% (19/19), done.
error: RPC failed; curl 56 GnuTLS recv error (-9): A TLS packet with unexpected length was received.
fatal: The remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
Makefile:71: recipe for target '/home/rhw/lede/dl/rblibtorrent-1.1.13.tar.gz' failed
make[3]: *** [/home/rhw/lede/dl/rblibtorrent-1.1.13.tar.gz] Error 128
make[3]: Leaving directory '/home/rhw/lede/package/lean/rblibtorrent'
time: package/lean/rblibtorrent/compile#0.21#0.16#883.24
package/Makefile:107: recipe for target 'package/lean/rblibtorrent/compile' failed
make[2]: *** [package/lean/rblibtorrent/compile] Error 2
make[2]: Leaving directory '/home/rhw/lede'
package/Makefile:103: recipe for target '/home/rhw/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/rhw/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/rhw/lede'
/home/rhw/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

gsblay, 2019-11-16:
最近我的编译都是成功的，建议编译前执行 make clean ；
不行的话执行 make dirclean （尤其是同时编译不同构架的固件后）；
还是不行，就干脆执行 make dictclean 后再重新编译试试看 ！

gsblay, 2019-11-16:
仍然不行，那就把 lean 文件夹删除，重新 git 后重新编译！

rhww, 2019-11-16:
我昨晚已经把整个文件夹删除了，编译了一晚上还是失败，不知道为啥，心累。

gsblay, 2019-11-16:
可能和你选择的组件有关，试试默认的能否编译过去 ？！

rhww, 2019-11-16:
很有可能，我默认试试能不能行吧，多谢。

inautri, 2019-11-16:
有编译成功的朋友分享一个

king9039, 2019-11-16:
我遇到过，我是执行了以下成功的，最后编译时用make -j1 V=s
./scripts/feeds clean
sudo make clean
chmod -R 777 *
chown -R 自己的用户:自己的用户 *

qiangwushuang, 2019-11-16:
斐讯K3编译遇到同样的问题，重新clone使用原先的config还是失败。

L-vivi, 2019-11-16:
好像是qB的问题

qiangwushuang, 2019-11-17:
> 
> 
> 我遇到过，我是执行了以下成功的，最后编译时用make -j1 V=s
> ./scripts/feeds clean
> sudo make clean
> rm -rf tmp/ build_dir/ staging_dir/
> chmod -R 777 *
> chown -R 自己的用户:自己的用户 *

按照你的方法重试了，还是不行。

outemancut, 2019-11-17:
这不是下载出错么？你自己去https://github.com/arvidn/libtorrent/releases下载对应的版本源码丢在dl文件夹里面再编译

rhww, 2019-11-17:
我下载了dl，然后重新clone默认配置编译成功了

czjesda, 2019-11-20:
> 
> 
> 我下载了dl，然后重新clone默认配置编译成功了

如何下载DL，是用命令来下载的么？

qiangwushuang, 2019-11-20:
重新运行
```./scripts/feeds update -a 
   ./scripts/feeds install -a
```
然后全程挂代理成功了。

gsblay, 2019-11-20:
是用命令 make download V=s 来进行下载，我一般在 make -j1 V=s 之前执行。

f3404923, 2019-11-28:
同感，我的之前都没有问题.target为x86，好像从11.26日左右开始，老是报MAC80211 MISSING错误。用openwrt官方的mac80211目录替换了也还是一样错误

closed 2019-11-28 by coolsnowwolf