coolsnowwolf/lede #2083: 希望恢复V2Ray的编译

KFERMercer, 2019-11-17:
在直接提供v2ray二进制的情况下, 每次更新都需要开发者根据平台自行编译出二进制文件, 这需要事先编译出各个平台的工具链, 徒增了维护负担. 

由于git无法对二进制文件进行有效的压缩, 为了储存版本信息, 会导致每次对v2ray进行版本更新都会使整个库大小增加至少20MB.

只提供二进制文件, 也不利于整个项目的透明度. 

希望重新释出v2ray的编译代码, 或使用第三方服务器提供二进制文件.

KFERMercer, 2019-11-17:
如果有人想要使用新版本的v2ray, 可以暂时使用以下项目代替:
[KFERMercer/openwrt-v2ray](https://github.com/KFERMercer/openwrt-v2ray)

outemancut, 2019-11-17:
算了吧 直接编译的体积大几倍 比我整个固件都大！

KFERMercer, 2019-11-17:
> 算了吧 直接编译的体积大几倍 比我整个固件都大！

你现在用的二进制是lean用手按01敲出来的吗

Zack0617, 2019-11-17:
> 如果有人想要使用新版本的v2ray, 可以暂时使用以下项目代替:
> [KFERMercer/openwrt-v2ray](https://github.com/KFERMercer/openwrt-v2ray)

官方openwrt，按照个尝试了一下，添加ssr+编译不成功，

xinxijishuwyq, 2019-11-18:
顶你

coolsnowwolf, 2019-11-19:
这样编译的结果是 K3 之类不带 FPU 的 ARM 是不能运行 V2Ray 的，因为 OpenWrt 定义了 K3 是 ARMv7，实际上根据 V2Ray 来说应该算是 ARMv5/v6

chenjau, 2019-11-21:
翻 issues 才了解原来包含预编译二进制文件, 建议 README 里面说明下.

KFERMercer, 2019-11-25:
lean可以另外开一个v2ray的存储库来放v2ray的代码, 并用release的方式释出二进制文件, lede库中的v2ray则可以获取该库release出的文件.

lifetyper, 2019-12-09:
> 
> 
> lean可以另外开一个v2ray的存储库来放v2ray的代码, 并用release的方式释出二进制文件, lede库中的v2ray则可以获取该库release出的文件.

支持这种做法，有安全顾虑就自己去编译，否则就直接用lean给出的2进制文件。

felix-fly, 2019-12-16:
编译出来的二进制文件可以用upx压缩下就小多了，只是不确定对后续有没有影响

closed 2019-12-30 by coolsnowwolf