coolsnowwolf/lede #2096: qBittorrent配置文件会在重启之后或者管理界面里保存配置会变化。

yenping08, 2019-11-18:
固件版本 | OpenWrt R9.11.9 / LuCI Master (git-19.299.52050-0b735ea)
-- | --
内核版本 | 4.19.82

原配置：[AutoRun]
enabled=false
program=

[Core]
AutoDeleteAddedTorrentFile=Never

[LegalNotice]
Accepted=true

[Network]
Cookies=@Invalid()

[Preferences]
Bittorrent\MaxConnecs=-1
Bittorrent\MaxConnecsPerTorrent=-1
Downloads\SavePath=/mnt/sdb1/download
General\Locale=zh
Queueing\QueueingEnabled=false
WebUI\Port=8082
WebUI\CSRFProtection=false
WebUI\ClickjackingProtection=false
WebUI\HTTPS\Enabled=false
WebUI\HostHeaderValidation=false
WebUI\LocalHostAuth=false

重启路由或者重新保存之后的配置：
[LegalNotice]
Accepted=true

[Preferences]
Bittorrent\MaxConnecs=-1
Bittorrent\MaxConnecsPerTorrent=-1
1/download
General\Locale=
Queueing\QueueingEnabled=false
WebUI\Port=48484
WebUI\CSRFProtection=false
WebUI\ClickjackingProtection=false
WebUI\HTTPS\Enabled=false
WebUI\HostHeaderValidation=false
WebUI\LocalHostAuth=false
Downloads\SavePath=

Her0R0cket, 2019-11-19:
默认替换路径的正则表达式只替换了小写字符和右斜杠,但实际情况是:路径里还可能有一些大写数字和符号.所以这个正则表达式无法正确处理.
临时解决办法是把/etc/init.d/qbittorrent的文件17,18行注释了(如果你用的最新版本的话),这样只会在第一次生成配置文件时处理存储路径

hong0980, 2019-11-19:
在打开WebUI下设置保存下都解决了

closed 2019-11-28 by coolsnowwolf