coolsnowwolf/lede #2103: NAT回流问题

swfz2008, 2019-11-19:
防火墙端口映射，NAT回流不起作用了，局域网ip无法访问WAN的ip

vlanbin, 2019-11-20:
遇到了同样的问题(R9.10.24,内核4.19.81)，使用的以前编译的版本(R9.8.17,内核4.14.138)未发现这个问题。今天打算编译最新源码再试一次。

coolsnowwolf, 2019-11-20:
docker 的内核 NS 原因

norikohxb, 2019-11-20:
其实可以自己用指定的域名解析为内网地址先替代，我的ros是双线负载均衡，怎么弄回流都弄不好网上也找不到教程关于双线的。最后只能是定语外网域名对应内网ip，做dns劫持

swfz2008, 2019-11-23:
> docker 的内核 NS 原因

是否删除docker可以解决？

DockerCS, 2019-11-23:
> > docker 的内核 NS 原因
> 
> 是否删除docker可以解决？

我把docker删除了,而且安装Google给的各种设置方法,好像还是不起作用.内网访问不了

closed 2019-11-28 by coolsnowwolf

ppdragon16, 2020-12-11:
commit 479e7027c24c1b1a0d7b0d17fe841886bed6a145 的锅。
git revert 479e7027c24c1b1a0d7b0d17fe841886bed6a145可修复此问题。

woOzZ1, 2020-12-27:
> commit [479e702](https://github.com/coolsnowwolf/lede/commit/479e7027c24c1b1a0d7b0d17fe841886bed6a145) 的锅。
> git revert 479e7027c24c1b1a0d7b0d17fe841886bed6a145可修复此问题。

问题得到了解决，原因不是`NAT loopback`不起作用，而是bridge防火墙给阻挡了回流的流量，之前一直怀疑是这样，因为发现nmap一直显示`filtered`而不是`closed`
如已经编译可以在`/etc/sysctl.conf `和`/etc/sysctl.d/11-br-netfilter.conf` 修改
`net.bridge.bridge-nf-call-arptables=0`
`net.bridge.bridge-nf-call-ip6tables=0`
`net.bridge.bridge-nf-call-iptables=0`
然后`sysctl -p` 即可使用`NAT loopback`

jerrykkkkk, 2022-02-28:
> 

按这个方式做了，没有效果

zyallday, 2023-02-19:
试试在 Nat 表里添加如下规则
`iptables -t nat -A POSTROUTING -o br-lan -s 192.168.0.0/24 -d 192.168.0.0/24 -j MASQUERADE`
我也是试了sysctl不行 使用上面的可行

alucardzlh, 2023-12-31:
> 试试在 Nat 表里添加如下规则 `iptables -t nat -A POSTROUTING -o br-lan -s 192.168.0.0/24 -d 192.168.0.0/24 -j MASQUERADE` 我也是试了sysctl不行 使用上面的可行 192.168.0.0/24 按需改写 请查看下自己的网段

困扰半年的问题终于解决了，感谢

EvanxGo, 2024-08-30:
> 试试在 Nat 表里添加如下规则 `iptables -t nat -A POSTROUTING -o br-lan -s 192.168.0.0/24 -d 192.168.0.0/24 -j MASQUERADE` 我也是试了sysctl不行 使用上面的可行 192.168.0.0/24 按需改写 请查看下自己的网段

感谢，网上各种方法试了都没成功，就这条命令生效了
我查了一下，这条命令大概是确保同一子网内通信也经过NAT处理的意思，但是理解得真切，请问这个会带来其他的影响吗

hkn-uestc, 2025-03-02:
> > > 试试在 Nat 表里添加如下规则 `iptables -t nat -A POSTROUTING -o br-lan -s 192.168.0.0/24 -d 192.168.0.0/24 -j MASQUERADE` 我也是试了sysctl不行 使用上面的可行 192.168.0.0/24 按需改写 请查看下自己的网段
> > 
> > 
> > 感谢，网上各种方法试了都没成功，就这条命令生效了 我查了一下，这条命令大概是确保同一子网内通信也经过NAT处理的意思，但是理解得真切，请问这个会带来其他的影响吗
> 
> 最好别这么配吧, 子网之通信没必要走NAT, [@woOzZ2](https://github.com/woOzZ2) 的方法测试可行, 需要注意同时修改 `/etc/sysctl.conf`, 没有文件就新建一下, 没有相关条目就加上. 完成后使用 `service sysctl restart` 重启一下服务后再看看.

非常有用，感谢感谢