coolsnowwolf/lede #2111: 突然断网,内核报了一堆日志，有没有高手帮忙看下

82kg, 2019-11-21:
[526307.946145] igb 0000:01:00.0: changing MTU from 1492 to 1500
[526307.969850] igb 0000:01:00.0: changing MTU from 1500 to 1492
[526308.013854] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[526308.020780] 8021q: adding VLAN 0 to HW filter on device eth0
[526308.117986] igb 0000:01:00.0: changing MTU from 1492 to 1500
[526308.142619] igb 0000:01:00.0: changing MTU from 1500 to 1492
[526308.183742] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[526308.190633] 8021q: adding VLAN 0 to HW filter on device eth0
[526311.393806] igb 0000:01:00.0 eth0: igb: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
[526311.643500] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[526327.020046] igb 0000:01:00.0: changing MTU from 1492 to 1500
[526327.044775] igb 0000:01:00.0: changing MTU from 1500 to 1492
[526327.092853] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[526327.099746] 8021q: adding VLAN 0 to HW filter on device eth0
[526327.180646] igb 0000:01:00.0: changing MTU from 1492 to 1500
[526327.205687] igb 0000:01:00.0: changing MTU from 1500 to 1492
[526327.252633] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[526327.259493] 8021q: adding VLAN 0 to HW filter on device eth0
[526330.492901] igb 0000:01:00.0 eth0: igb: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
[526330.742390] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[526346.089902] igb 0000:01:00.0: changing MTU from 1492 to 1500
[526346.114991] igb 0000:01:00.0: changing MTU from 1500 to 1492
[526346.162498] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[526346.169418] 8021q: adding VLAN 0 to HW filter on device eth0
[526346.259999] igb 0000:01:00.0: changing MTU from 1492 to 1500
[526346.284676] igb 0000:01:00.0: changing MTU from 1500 to 1492
[526346.331820] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[526346.338698] 8021q: adding VLAN 0 to HW filter on device eth0
[526349.572019] igb 0000:01:00.0 eth0: igb: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
[526349.821885] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[526365.161948] igb 0000:01:00.0: changing MTU from 1492 to 1500
[526365.186348] igb 0000:01:00.0: changing MTU from 1500 to 1492
[526365.231560] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[526365.238491] 8021q: adding VLAN 0 to HW filter on device eth0
[526365.330919] igb 0000:01:00.0: changing MTU from 1492 to 1500
[526365.355596] igb 0000:01:00.0: changing MTU from 1500 to 1492
[526365.400828] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[526365.407684] 8021q: adding VLAN 0 to HW filter on device eth0
[526368.601144] igb 0000:01:00.0 eth0: igb: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
[526368.850935] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[526384.192870] igb 0000:01:00.0: changing MTU from 1492 to 1500
[526384.218557] igb 0000:01:00.0: changing MTU from 1500 to 1492
[526384.260897] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[526384.267826] 8021q: adding VLAN 0 to HW filter on device eth0
[526384.363800] igb 0000:01:00.0: changing MTU from 1492 to 1500
[526384.388457] igb 0000:01:00.0: changing MTU from 1500 to 1492
[526384.430209] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[526384.437107] 8021q: adding VLAN 0 to HW filter on device eth0
[526387.560045] igb 0000:01:00.0 eth0: igb: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
[526387.809771] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready

还有很长，估计持续了10分钟断网。 eth0是wan口，这是路由的问题吗
有没有类似情况的兄弟啊？

coolsnowwolf, 2019-11-21:
检查和光猫的链接

82kg, 2019-11-21:
> 检查和光猫的链接

网线应该没问题。新换的线。我把wan口里的ipv6 全禁用试试，不知道是不是猫要挂。再观察下

closed 2019-11-21 by 82kg