coolsnowwolf/lede #2112: web ui not found some app menu

buxiaomo, 2019-11-21:
I select the luci-app-aria2, aria2ng, and aria2 on `make menuconfig` command , and `make` command not have error message, but i'm not found the aria2 menu in the web ui, what's the problem

[image]

buxiaomo, 2019-11-21:
I try to delete the `/tmp/luci-indexcache` file, but is not ok

EnnawYang, 2019-11-21:
跟我树莓派4B一样，没有网络硬盘选项

buxiaomo, 2019-11-21:
samba have this problem too

buxiaomo, 2019-11-21:
> 跟我树莓派4B一样，没有网络硬盘选项

How did you solve it

EnnawYang, 2019-11-21:
> > 跟我树莓派4B一样，没有网络硬盘选项
> 
> How did you solve it
no（

EnnawYang, 2019-11-21:
> > 跟我树莓派4B一样，没有网络硬盘选项
> 
> How did you solve it

No （

EnnawYang, 2019-11-21:
> samba have this problem too

“网络硬盘”选项下的app都不会显示，我之前编译过好几次都是这样，我现在换了设备了不知道有没有修复

buxiaomo, 2019-11-21:
> > samba have this problem too
> 
> “网络硬盘”选项下的app都不会显示，我之前编译过好几次都是这样，我现在换了设备了不知道有没有修复

some time, I saw samba on this menu, but more time I not found it

KFERMercer, 2019-11-21:
不兼容luci2 需要手动切换到services下

EnnawYang, 2019-11-21:
> 不需要自己手动添加luci-app-aria2, luci trunk 本身就有这个插件.
> 
> lean/openwrt下的aria2不兼容luci2, 硬要用的话需要在源码中手动切换到services下

那类似于asmba、kodexploer、硬盘休眠之类的也需要这么操作对吧

KFERMercer, 2019-11-21:
嗯

buxiaomo, 2019-11-21:
> 不需要自己手动添加luci-app-aria2, luci trunk 本身就有这个插件.
> 
> lean/openwrt下的aria2不兼容luci2, 硬要用的话需要在源码中手动切换到services下

how to change the app to services mode

coolsnowwolf, 2019-11-25:
不要漏了 vsftp 的 luci

buxiaomo, 2019-11-25:
> 不要漏了 vsftp 的 luci

 i select luci-app-vsftp, this ok , but samba and aria2 is not ok

buxiaomo, 2019-11-25:
how to change the app to services mode

EnnawYang, 2019-11-26:
> > 不要漏了 vsftp 的 luci
> 
> i select luci-app-vsftp, this ok , but samba and aria2 is not ok

[image]
我的加上ftp之后就能正常显示了

closed 2019-11-28 by coolsnowwolf