coolsnowwolf/lede #2118: nano pi neo2 编译失败，请大神指导

gouff, 2019-11-22:
编译环境：Ubuntu 14.04 LTS
插件：默认
错误信息如下：
checking whether mknod can create fifo without root privileges... configure: error: in `/data/lede/build_dir/host/tar-1.30':
configure: error: you should not run configure as root (set FORCE_UNSAFE_CONFIGURE=1 in environment to bypass this check)
See `config.log' for more details
make[3]: *** [/data/lede/build_dir/host/tar-1.30/.configured] Error 1
make[3]: Leaving directory `/data/lede/tools/tar'
time: tools/tar/compile#7.87#3.93#39.40
make[2]: *** [tools/tar/compile] Error 2
make[2]: Leaving directory `/data/lede'
make[1]: *** [/data/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.tools_compile_yynyyyyynyyyyynyynnnyyyynyyyyyyyyyyyyyyynnyynynnyyynnyy] Error 2
make[1]: Leaving directory `/data/lede'
make: *** [world] Error 2

EnnawYang, 2019-11-22:
不要使用root用户编译

outemancut, 2019-11-22:
出现这个输入export FORCE_UNSAFE_CONFIGURE=1再编译就好了

gouff, 2019-11-22:
解决了！确实是root用户的问题。

closed 2019-11-22 by gouff