coolsnowwolf/lede #2122: X86编译失败，删除整个lede文件无法成功

EstFoxs, 2019-11-22:
Building backport-include/backport/autoconf.h ... done.
  CC [M]  /home/bbq/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.4-rc2-1/compat/main.o
In file included from ./include/linux/kmod.h:22:0,
                 from ./include/linux/module.h:13,
                 from /home/bbq/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.4-rc2-1/backport-include/linux/module.h:3,
                 from /home/bbq/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.4-rc2-1/compat/main.c:1:
/home/bbq/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.4-rc2-1/backport-include/linux/gfp.h: In function 'backport_page_frag_alloc':
/home/bbq/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.4-rc2-1/backport-include/linux/gfp.h:18:9: error: implicit declaration of function '__alloc_page_frag'; did you mean '__alloc_pages_node'? [-Werror=implicit-function-declaration]
  return __alloc_page_frag(nc, fragsz, gfp_mask);
         ^~~~~~~~~~~~~~~~~
         __alloc_pages_node
/home/bbq/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.4-rc2-1/backport-include/linux/gfp.h:18:9: warning: return makes pointer from integer without a cast [-Wint-conversion]
  return __alloc_page_frag(nc, fragsz, gfp_mask);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[9]: *** [/home/bbq/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.4-rc2-1/compat/main.o] Error 1
make[8]: *** [/home/bbq/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.4-rc2-1/compat] Error 2
make[7]: *** [_module_/home/bbq/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.4-rc2-1] Error 2
make[6]: *** [modules] Error 2
make[5]: *** [modules] Error 2
make[4]: *** [modules] Error 2
make[4]: Leaving directory `/home/bbq/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.4-rc2-1'
make[3]: *** [/home/bbq/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.4-rc2-1/.built] Error 2
make[3]: Leaving directory `/home/bbq/lede/package/kernel/mac80211'
time: package/kernel/mac80211/compile#2.49#0.49#5.36
make[2]: *** [package/kernel/mac80211/compile] Error 2
make[2]: Leaving directory `/home/bbq/lede'
make[1]: *** [/home/bbq/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/bbq/lede'
make: *** [world] Error 2
bbq@bbq:~/lede$

royalwei, 2019-11-22:
我也是，编译ar71xx出现的

coolsnowwolf, 2019-11-23:
4.9 内核问题，修复了

closed 2019-11-23 by coolsnowwolf

mengxd, 2020-04-05:
问题有出现了