coolsnowwolf/lede #2129: R7800最新源码编译失败

pk49800, 2019-11-23:
acer@acer-Aspire-E1-571G:~/lede$ make -j1 V=s
make[1]: Entering directory '/home/acer/lede'
make[2]: Entering directory '/home/acer/lede'
make[3]: Entering directory '/home/acer/lede/target/linux'
make[4]: Entering directory '/home/acer/lede/target/linux/ipq806x'
rm -f /home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.155/localversion
/home/acer/lede/scripts/kconfig.pl  + /home/acer/lede/target/linux/generic/config-4.14 /home/acer/lede/target/linux/ipq806x/config-4.14 > /home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.155/.config.target
awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /home/acer/lede/.config >> /home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.155/.config.target
echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.155/.config.target
echo "# CONFIG_KALLSYMS_ALL is not set" >> /home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.155/.config.target
echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.155/.config.target
/home/acer/lede/scripts/package-metadata.pl kconfig /home/acer/lede/tmp/.packageinfo /home/acer/lede/.config 4.14 > /home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.155/.config.override
/home/acer/lede/scripts/kconfig.pl 'm+' '+' /home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.155/.config.target /dev/null /home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.155/.config.override > /home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.155/.config.set
mv /home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.155/.config.set /home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.155/.config.old
grep -v INITRAMFS /home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.155/.config.old > /home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.155/.config.set
echo 'CONFIG_INITRAMFS_SOURCE=""' >> /home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.155/.config.set
echo '# CONFIG_INITRAMFS_FORCE is not set' >> /home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.155/.config.set
rm -rf /home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/modules
cmp -s /home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.155/.config.set /home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.155/.config.prev || { cp /home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.155/.config.set /home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.155/.config; cp /home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.155/.config.set /home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.155/.config.prev; }
export MAKEFLAGS= ; [ -d /home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.155/user_headers ] || make -C /home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.155 HOSTCFLAGS="-O2 -I/home/acer/lede/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="arm-openwrt-linux-muslgnueabi-" ARCH="arm" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Fri Nov 22 17:38:03 2019" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/acer/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KERNELRELEASE=4.14.155 CC="arm-openwrt-linux-muslgnueabi-gcc" INSTALL_HDR_PATH=/home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.155/user_headers headers_install
grep '=[ym]' /home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.155/.config.set | LC_ALL=C sort | mkhash md5 > /home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.155/.vermagic
touch /home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.155/.configured
rm -f /home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.155/vmlinux /home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.155/System.map
make -C /home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.155 HOSTCFLAGS="-O2 -I/home/acer/lede/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="arm-openwrt-linux-muslgnueabi-" ARCH="arm" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Fri Nov 22 17:38:03 2019" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/acer/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KERNELRELEASE=4.14.155 CC="arm-openwrt-linux-muslgnueabi-gcc" modules
make[5]: Entering directory '/home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.155'
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  CHK     include/generated/bounds.h
  CHK     include/generated/timeconst.h
  CHK     include/generated/asm-offsets.h
  CALL    scripts/checksyscalls.sh
  CHK     scripts/mod/devicetable-offsets.h
  Building modules, stage 2.
  MODPOST 168 modules
make[5]: Leaving directory '/home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.155'
touch /home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.155/.modules
make -C image compile TARGET_BUILD=
make[5]: Entering directory '/home/acer/lede/target/linux/ipq806x/image'
make[5]: Nothing to be done for 'compile'.
make[5]: Leaving directory '/home/acer/lede/target/linux/ipq806x/image'
make[4]: Leaving directory '/home/acer/lede/target/linux/ipq806x'
make[3]: Leaving directory '/home/acer/lede/target/linux'
time: target/linux/compile#4.65#1.72#6.28
make[2]: Leaving directory '/home/acer/lede'
export MAKEFLAGS= ;make -w -r diffconfig
make[2]: Entering directory '/home/acer/lede'
make[2]: Leaving directory '/home/acer/lede'
make[2]: Entering directory '/home/acer/lede'
rm -rf /home/acer/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x
make[2]: Leaving directory '/home/acer/lede'
make[2]: Entering directory '/home/acer/lede'
make[3]: Entering directory '/home/acer/lede/package/libs/libjson-c'
make[3]: Leaving directory '/home/acer/lede/package/libs/libjson-c'
time: package/libs/libjson-c/host-compile#0.21#0.10#0.28
make[3]: Entering directory '/home/acer/lede/package/libs/libubox'
make[3]: Leaving directory '/home/acer/lede/package/libs/libubox'
time: package/libs/libubox/host-compile#0.67#0.36#0.96
make[3]: Entering directory '/home/acer/lede/package/system/opkg'
make[3]: Leaving directory '/home/acer/lede/package/system/opkg'
time: package/system/opkg/host-compile#0.56#0.28#0.79
make[3]: Entering directory '/home/acer/lede/package/libs/toolchain'
mkdir -p /home/acer/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/stamp
SHELL= flock /home/acer/lede/tmp/.root-copy.flock -c 'cp -fpR /home/acer/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/toolchain/.pkgdir/libc/. /home/acer/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/'
touch /home/acer/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/stamp/.libc_installed
echo "libc" >> /home/acer/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/toolchain.default.install
mkdir -p /home/acer/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/stamp
[... cut ...]
make[2]: *** [package/network/services/uhttpd/compile] Error 2
make[2]: Leaving directory '/home/acer/lede'
package/Makefile:103: recipe for target '/home/acer/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/home/acer/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/acer/lede'
/home/acer/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

coolsnowwolf, 2019-11-23:
make clean

closed 2019-11-23 by coolsnowwolf

kktt007, 2019-11-25:
兄弟，我最近换ubuntu编译的，R7800 没问题。之间我非要用archlinux这样的编译，今年9月初问题，解决的各种编译错误，还是放弃了。您就用ubuntu编译，应该没问题的，我确定。 最后编译可以默念:
L有大屌