coolsnowwolf/lede #2131: 第二次编译时，make meauconfig失败

gouff, 2019-11-23:
gouff@ubuntu:/data/lede$ make clean
 make[1] clean
gouff@ubuntu:/data/lede$ ls
bin          dl                  key-build.pub           package      target
BSDmakefile  feeds               key-build.ucert         README       tmp
build_dir    feeds.conf.default  key-build.ucert.revoke  rules.mk     toolchain
config       include             LICENSE                 scripts      tools
Config.in    key-build           Makefile                staging_dir
gouff@ubuntu:/data/lede$ make meauconfig
time: target/linux/prereq#0.02#0.05#0.13
 make[1] meauconfig
make -r meauconfig: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [meauconfig] Error 1

EnnawYang, 2019-11-23:
命令打错了，兄弟

gouff, 2019-11-23:
感谢

closed 2019-11-23 by gouff