coolsnowwolf/lede #2146: 最新的版本Acrh17的，无线会定时无故断开，看日志，无线重启了

chenwenqiu2013, 2019-11-24:
最新的版本Acrh17的，无线会定时无故断开，看日志，无线重启了

star-o0O, 2019-11-24:
我也遇到了，体现为不定时重启wifi/卡死/连不上网，固件有自己编译的也有大雕发群里的。看群里也有几个人遇到重启/断流之类的问题。你的图好像挂了。
[   15.011395] ath10k_pci 0000:01:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[   15.184560] ath10k_pci 0000:01:00.0: Direct firmware load for ath10k/QCA9984/hw1.0/firmware-6.bin failed with error -2
[   15.184649] ath10k_pci 0000:01:00.0: Falling back to user helper
[   15.239813] firmware ath10k!QCA9984!hw1.0!firmware-6.bin: firmware_loading_store: map pages failed
[   16.811004] ath10k_pci 0000:01:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe

star-o0O, 2019-11-24:
目前刷回原厂固件，虽然好像还是报错，但是能用。下面是原厂固件的完整日志

May  5 05:05:15 syslogd started: BusyBox v1.17.4
May  5 05:05:15 kernel: klogd started: BusyBox v1.17.4 (2019-03-25 18:25:47 CST)
May  5 05:05:15 INIT: firmware version: 3.0.0.4_382_51569-g66b0d4f
May  5 05:05:15 kernel: Linux version 3.14.77 (jenkins@asuswrt-build-server) (gcc version 4.6.3 20120201 (prerelease) (Linaro GCC 4.6-2012.02) ) #1 SMP PREEMPT Mon Mar 25 18:27:59 CST 2019
May  5 05:05:16 kernel: Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 64512
May  5 05:05:16 kernel: Kernel command line: root_rfs=0x00203140 flash_type=nand ubi.mtd=UBI_DEV  clk_ignore_unused
May  5 05:05:16 kernel: Memory: 251504K/260096K available (3990K kernel code, 194K rwdata, 1152K rodata, 172K init, 537K bss, 8592K reserved, 0K highmem)
May  5 05:05:16 kernel: Virtual kernel memory layout:
May  5 05:05:16 kernel:     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
May  5 05:05:16 kernel:     fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
May  5 05:05:16 kernel:     vmalloc : 0xd0800000 - 0xff000000   ( 744 MB)
May  5 05:05:16 kernel:     lowmem  : 0xc0000000 - 0xd0000000   ( 256 MB)
May  5 05:05:16 kernel:     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
May  5 05:05:16 kernel:     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
May  5 05:05:16 kernel:       .text : 0xc0208000 - 0xc070dc1c   (5144 kB)
May  5 05:05:16 kernel:       .init : 0xc070e000 - 0xc0739180   ( 173 kB)
May  5 05:05:16 kernel:       .data : 0xc073a000 - 0xc076a844   ( 195 kB)
May  5 05:05:16 kernel:        .bss : 0xc076a844 - 0xc07f0e58   ( 538 kB)
May  5 05:05:16 kernel: Invalid signature of oopsbuf: FF-FF-FF-FF-DF-FF-E7-FE (len -262145)
May  5 05:05:16 kernel: CPU1: Booted secondary processor
May  5 05:05:16 kernel: CPU2: Booted secondary processor
May  5 05:05:16 kernel: CPU3: Booted secondary processor
May  5 05:05:16 kernel: hw-breakpoint: Debug register access (0xee003e17) caused undefined instruction on CPU 2
May  5 05:05:16 kernel: hw-breakpoint: Debug register access (0xee003e17) caused undefined instruction on CPU 1
May  5 05:05:16 kernel: hw-breakpoint: Debug register access (0xee003e17) caused undefined instruction on CPU 3
May  5 05:05:16 kernel: hw-breakpoint: CPU 1 failed to disable vector catch
May  5 05:05:16 kernel: hw-breakpoint: Debug register access (0xee003e17) caused undefined instruction on CPU 0
May  5 05:05:16 kernel: SCSI subsystem initialized
May  5 05:05:16 kernel: msm_bus_fabric_init_driver
May  5 05:05:16 kernel: msm_bus_device 580000.ad-hoc-bus: Failed to get bus clk for bus4096 ctx1
May  5 05:05:16 kernel: msm_bus_device 580000.ad-hoc-bus: Failed to get bus clk for bus1024 ctx1
May  5 05:05:16 kernel: 80000.qcom,pcie supply vreg-3.3 not found, using dummy regulator
May  5 05:05:16 kernel: 80000.qcom,pcie supply vreg-1.8 not found, using dummy regulator
May  5 05:05:16 kernel: 80000.qcom,pcie supply vreg-0.9 not found, using dummy regulator
May  5 05:05:16 kernel: 80000.qcom,pcie supply gdsc-vdd not found, using dummy regulator
May  5 05:05:16 kernel: msm_pcie_oper_conf: msm_pcie_oper_conf: Read of RC0 0:0x00 + 0x0014[4] is all FFs
May  5 05:05:16 kernel: msm_pcie_oper_conf: msm_pcie_oper_conf: Read of RC0 1:0x00 + 0x0014[4] is all FFs
May  5 05:05:16 kernel: msm_pcie_oper_conf: msm_pcie_oper_conf: Read of RC0 0:0x00 + 0x0028[4] is all FFs
May  5 05:05:16 nat: apply redirect rules
May  5 05:05:16 kernel: msm_pcie_probe: msm_pcie_probe: RC0 is enabled in bootup
May  5 05:05:16 WAN Connection: Fail to connect with some issues.
May  5 05:05:16 kernel: Key type asymmetric registered
May  5 05:05:16 kernel: Asymmetric key parser 'x509' registered
May  5 05:05:16 kernel: sps: BAM device 0x07984000 is not registered yet.
May  5 05:05:16 kernel: 2 ofpart partitions found on MTD device 7980000.qcom,nand
May  5 05:05:16 kernel: Creating 2 MTD partitions on "7980000.qcom,nand":
May  5 05:05:16 kernel: 0x000000000000-0x000000400000 : "Bootloader"
May  5 05:05:16 kernel: 0x000000400000-0x000008000000 : "UBI_DEV"
May  5 05:05:17 kernel: sps: BAM device 0x07884000 is not registered yet.
May  5 05:05:17 kernel: phy_reset_gpio is 47
May  5 05:05:17 kernel: requesting memory region failed
May  5 05:05:17 kernel: Bridge firewalling registered
May  5 05:05:17 kernel: Registering SWP/SWPB emulation handler
May  5 05:05:17 kernel: UBI: attaching mtd1 to ubi0
May  5 05:05:17 kernel: UBI: scanning is finished
May  5 05:05:17 kernel: UBI: attached mtd1 (name "UBI_DEV", size 124 MiB) to ubi0
May  5 05:05:17 kernel: UBI: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
May  5 05:05:17 kernel: UBI: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
May  5 05:05:17 kernel: UBI: VID header offset: 2048 (aligned 2048), data offset: 4096
May  5 05:05:17 kernel: UBI: good PEBs: 992, bad PEBs: 0, corrupted PEBs: 0
May  5 05:05:17 kernel: UBI: user volume: 6, internal volumes: 1, max. volumes count: 128
May  5 05:05:17 kernel: UBI: max/mean erase counter: 600/16, WL threshold: 4096, image sequence number: 551267818
May  5 05:05:17 kernel: UBI: available PEBs: 19, total reserved PEBs: 973, PEBs reserved for bad PEB handling: 20
May  5 05:05:17 kernel: UBI: background thread "ubi_bgt0d" started, PID 65
May  5 05:05:17 kernel: gluebi_create: Hook read/write method of Factory.
May  5 05:05:17 kernel: gluebi_create: Hook read/write method of Factory2.
May  5 05:05:17 kernel: [Factory-0]: OK (ver: 15)
May  5 05:05:17 kernel: [Factory2-0]: OK (ver: 15)
[... 2380 lines cut ...]
[... cut ...]
Nov 24 17:55:17 kernel: [UNSPECIFIED] ol_scan_unregister_event_handler: Failed to unregister evhandler=bf5fa164 arg=cc3714c0
Nov 24 17:55:17 kernel: [UNSPECIFIED] ol_scan_unregister_event_handler: Failed to unregister evhandler=bf5fa730 arg=cc3714c0
Nov 24 17:55:17 kernel: send_vdev_delete_cmd_non_tlv for vap 0
Nov 24 17:55:17 kernel: ol_ath_vap_delete: wmi_unified_vdev_delete_send done ID = 0 vap (ca488000) VAP Addr = 4c:ed:fb:8e:55:8c:
Nov 24 17:55:17 kernel: ieee80211_mbo_vdetach: MBO terminated
Nov 24 17:55:17 kernel: ieee80211_oce_vdetach: OCE terminated
Nov 24 17:55:17 kernel: Suspending Target  scn=ce3804

wilson-chan2019, 2019-11-24:
> 目前刷回原厂固件，虽然好像还是报错，但是能用。下面是原厂固件的完整日志
>
> May 5 05:05:15 syslogd started: BusyBox v1.17.4
> May 5 05:05:15 kernel: klogd started: BusyBox v1.17.4 (2019-03-25 18:25:47 CST)
> May 5 05:05:15 INIT: firmware [...]

kid-oh, 2019-11-24:
> `最新的版本Acrh17的，无线会定时无故断开，看日志，无线重启了，看图片[https://s2.ax1x.com/2019/11/24/MObnaR.png](url)

对 我也遇到了 很烦

star-o0O, 2019-11-25:
11/25最新版本日志

Tue Nov 26 00:10:42 2019 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Tue Nov 26 00:10:42 2019 kern.notice kernel: [    0.000000] Linux version 4.14.155 (star@ubuntu) (gcc version 7.4.0 (OpenWrt GCC 7.4.0 r1708-359d56d5)) #0 SMP Mon Nov 25 12:40:32 2019
Tue Nov 26 00:10:42 2019 kern.info kernel: [    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
Tue Nov 26 00:10:42 2019 kern.info kernel: [    0.000000] CPU: div instructions available: patching division code
Tue Nov 26 00:10:42 2019 kern.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Tue Nov 26 00:10:42 2019 kern.info kernel: [    0.000000] OF: fdt: Machine model: ASUS RT-ACRH17
Tue Nov 26 00:10:42 2019 kern.info kernel: [    0.000000] Memory policy: Data cache writealloc
Tue Nov 26 00:10:42 2019 kern.debug kernel: [    0.000000] On node 0 totalpages: 65024
Tue Nov 26 00:10:42 2019 kern.debug kernel: [    0.000000] free_area_init_node: node 0, pgdat c0a268c0, node_mem_map cfdb9000
Tue Nov 26 00:10:42 2019 kern.debug kernel: [    0.000000]   Normal zone: 576 pages used for memmap
Tue Nov 26 00:10:42 2019 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Tue Nov 26 00:10:42 2019 kern.debug kernel: [    0.000000]   Normal zone: 65024 pages, LIFO batch:15
Tue Nov 26 00:10:42 2019 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x8c/0x3e8 with crng_init=0
Tue Nov 26 00:10:42 2019 kern.info kernel: [    0.000000] percpu: Embedded 15 pages/cpu s29452 r8192 d23796 u61440
Tue Nov 26 00:10:42 2019 kern.debug kernel: [    0.000000] pcpu-alloc: s29452 r8192 d23796 u61440 alloc=15*4096
Tue Nov 26 00:10:42 2019 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Tue Nov 26 00:10:42 2019 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64448
Tue Nov 26 00:10:42 2019 kern.notice kernel: [    0.000000] Kernel command line: root_rfs=0x00000000 flash_type=nand  ubi.mtd=UBI_DEV
Tue Nov 26 00:10:42 2019 kern.info kernel: [    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
Tue Nov 26 00:10:42 2019 kern.info kernel: [    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
Tue Nov 26 00:10:42 2019 kern.info kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
Tue Nov 26 00:10:42 2019 kern.info kernel: [    0.000000] Memory: 249668K/260096K available (4526K kernel code, 156K rwdata, 1236K rodata, 1024K init, 234K bss, 10428K reserved, 0K cma-reserved, 0K highmem)
Tue Nov 26 00:10:42 2019 kern.notice kernel: [    0.000000] Virtual kernel memory layout:
Tue Nov 26 00:10:42 2019 kern.notice kernel: [    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
Tue Nov 26 00:10:42 2019 kern.notice kernel: [    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
Tue Nov 26 00:10:42 2019 kern.notice kernel: [    0.000000]     vmalloc : 0xd0800000 - 0xff800000   ( 752 MB)
Tue Nov 26 00:10:42 2019 kern.notice kernel: [    0.000000]     lowmem  : 0xc0000000 - 0xd0000000   ( 256 MB)
Tue Nov 26 00:10:42 2019 kern.notice kernel: [    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
Tue Nov 26 00:10:42 2019 kern.notice kernel: [    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
Tue Nov 26 00:10:42 2019 kern.notice kernel: [    0.000000]       .text : 0xc0208000 - 0xc076b988   (5519 kB)
Tue Nov 26 00:10:42 2019 kern.notice kernel: [    0.000000]       .init : 0xc0900000 - 0xc0a00000   (1024 kB)
Tue Nov 26 00:10:42 2019 kern.notice kernel: [    0.000000]       .data : 0xc0a00000 - 0xc0a271c0   ( 157 kB)
Tue Nov 26 00:10:42 2019 kern.notice kernel: [    0.000000]        .bss : 0xc0a29000 - 0xc0a63a98   ( 235 kB)
Tue Nov 26 00:10:42 2019 kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Tue Nov 26 00:10:42 2019 kern.info kernel: [    0.000000] Hierarchical RCU implementation.
Tue Nov 26 00:10:42 2019 kern.info kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Tue Nov 26 00:10:42 2019 kern.info kernel: [    0.000000] arch_timer: cp15 timer(s) running at 48.00MHz (virt).
Tue Nov 26 00:10:42 2019 kern.info kernel: [    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb11fd3bfb, max_idle_ns: 440795203732 ns
Tue Nov 26 00:10:42 2019 kern.info kernel: [    0.000007] sched_clock: 56 bits at 48MHz, resolution 20ns, wraps every 4398046511096ns
Tue Nov 26 00:10:42 2019 kern.info kernel: [    0.000021] Switching to timer-based delay loop, resolution 20ns
Tue Nov 26 00:10:42 2019 kern.info kernel: [    0.000243] Calibrating delay loop (skipped), value calculated using timer frequency.. 96.00 BogoMIPS (lpj=480000)
Tue Nov 26 00:10:42 2019 kern.info kernel: [    0.000261] pid_max: default: 32768 minimum: 301
Tue Nov 26 00:10:42 2019 kern.info kernel: [    0.000424] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Tue Nov 26 00:10:42 2019 kern.info kernel: [    0.000441] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
Tue Nov 26 00:10:42 2019 kern.info kernel: [    0.001229] CPU: Testing write buffer coherency: ok
Tue Nov 26 00:10:42 2019 kern.info kernel: [    0.001979] Setting up static identity map for 0x80300000 - 0x80300060
Tue Nov 26 00:10:42 2019 kern.info kernel: [    0.002130] Hierarchical SRCU implementation.
[... cut ...]
Tue Nov 26 00:10:57 2019 daemon.notice procd: /etc/rc.d/S96led: setting up led LAN4
Tue Nov 26 00:10:57 2019 user.notice ddns-scripts[3672]: myddns_ipv4: PID '3672' started at 2019-11-26 00:10
Tue Nov 26 00:10:57 2019 daemon.info procd: - init complete -
Tue Nov 26 00:10:58 2019 daemon.info urandom_seed[4169]: Seed saved (/etc/urandom.seed)
Tue Nov 26 00:11:00 2019 user.warn ddns-scripts[3672]: myddns_ipv4: Service section disabled! - TERMINATE
Tue Nov 26 00:11:00 2019 user.warn ddns-scripts[3672]: myddns_ipv4: PID '3672' exit WITH ERROR '1' at 2019-11-26 00:11

aesirteam, 2019-11-27:
拉取最新代码，此问题解决！！！

coolsnowwolf, 2019-12-06:
fixed

closed 2019-12-06 by coolsnowwolf