coolsnowwolf/lede #2149: 更新新版本设置无线重启后无线起不来

z826540272, 2019-11-24:
设备 WR1200JS
MT7621A+MT7603E+MT7612E
日志:
Mon Nov 25 04:58:10 2019 daemon.notice netifd: radio1 (3675): sh: out of range
Mon Nov 25 04:58:10 2019 daemon.notice netifd: radio1 (3675): command failed: Too many open files in system (-23)
Mon Nov 25 04:58:12 2019 daemon.notice netifd: radio1 (3675): command failed: Too many open files in system (-23)
Mon Nov 25 04:58:13 2019 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Nov 25 04:58:13 2019 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Nov 25 04:58:13 2019 daemon.notice netifd: radio1 (3675): Command failed: Not found
Mon Nov 25 04:58:13 2019 daemon.notice netifd: radio1 (3675): Device setup failed: HOSTAPD_START_FAILED

z826540272, 2019-11-24:
手动切换一次频宽保存才可以开启WIFI.
WIFI无法启动时LUCI界面显示
SSID: OpenWrt | 模式: Master
BSSID: D4:5F:25:EB:A8:D2 | 加密: -
WIFI正确启动后是
SSID: OpenWrt | 模式: Master
BSSID: D4:5F:25:EB:A8:D2 | 加密: WPA2 PSK (CCMP)

jxph1123, 2019-11-25:
同样的，无线起不来

yasum2006, 2019-11-25:
K3 也是遇到同样问题，手动切换频宽或者频道可以开启无线，但重启后无线起不来

Mattaclp, 2019-11-25:
N1使用了新源碼編譯,無線啟動不了,11月19日編譯的正常
Mon Nov 25 13:17:51 2019 daemon.notice netifd: radio0 (8750): command failed: I/O error (-5)
Mon Nov 25 13:17:51 2019 daemon.notice netifd: radio0 (8750): sh: out of range
Mon Nov 25 13:17:53 2019 kern.err kernel: [  396.165626] brcmfmac: brcmf_ap_add_vif: timeout occurred
Mon Nov 25 13:17:53 2019 kern.err kernel: [  396.165937] brcmfmac: brcmf_cfg80211_add_iface: add iface wlan0 type 3 failed: err=-5
Mon Nov 25 13:17:53 2019 daemon.notice netifd: radio0 (8750): command failed: I/O error (-5)
Mon Nov 25 13:17:53 2019 daemon.notice netifd: radio0 (8750): Device setup failed: INTERFACE_CREATION_FAILED
Mon Nov 25 13:18:43 2019 daemon.notice netifd: radio0 (9532): command failed: I/O error (-5)
Mon Nov 25 13:18:43 2019 daemon.notice netifd: radio0 (9532): sh: out of range
Mon Nov 25 13:18:44 2019 kern.err kernel: [  447.749660] brcmfmac: brcmf_ap_add_vif: timeout occurred
Mon Nov 25 13:18:44 2019 kern.err kernel: [  447.749965] brcmfmac: brcmf_cfg80211_add_iface: add iface wlan0 type 3 failed: err=-5
Mon Nov 25 13:18:44 2019 daemon.notice netifd: radio0 (9532): command failed: I/O error (-5)
Mon Nov 25 13:18:44 2019 daemon.notice netifd: radio0 (9532): Device setup failed: INTERFACE_CREATION_FAILED

hcym, 2019-11-25:
我中午编译的正常，n1无线和大雕的源码基本没有关系，和armbian关系大一些

hcym, 2019-11-25:
我靠，是有点问题，前面没开，看到有无线口，接k3用吧，暖和

sxlehua, 2019-11-26:
设备HC5661，同样的问题。
之前版本好的，最新编译的固件启动后无法打开wifi，需要手动才能打开。

jxph1123, 2019-11-26:
新版本问题解决了吗？

sxlehua, 2019-11-26:
还没有解决

coolsnowwolf, 2019-11-28:
已经不支持支持 BCM 的无线了

closed 2019-11-28 by coolsnowwolf

hcwang, 2020-04-13:
大家好，有任何解决方案吗？目前小米3Pro也有同样的问题。

z826540272, 2020-04-13:
再重启一次

>

hcwang, 2020-04-13:
再重启也没用，但使用/sbin/wifi up 就可以启动了。很奇怪！