coolsnowwolf/lede #2155: SSR +的V2ray加密方式，allowInsecure，TLS导入参数时不正确

willaytoun, 2019-11-25:
刷11月24号l大编译的固件（openwrt-R9.11.24-x64-combined-squashfs）
1、V2用网址导入时实际是AUTO（导入是NONE），allowInsecure实际是√（导入没有），TLS实际是√ （引入没有）
2、下面的Url是随便填测试几个参数问题，不能正常连接:，如果需要正常用的可以私发作者做测试
具体信息如下图：固件概况
[image]
V2ray客户端导入时参数
[image]
SSR+导入时问题
[image]
测试用的Url：vmess://ew0KICAidiI6ICIyIiwNCiAgInBzIjogIua1i+ivleWvvOWFpeWPguaVsCjkuI3lj6/ov54pIiwNCiAgImFkZCI6ICJhYmMuc2F2ZWZpbGUudGsiLA0KICAicG9ydCI6ICI4MDgwIiwNCiAgImlkIjogIjM2NGNlODFjLTcyNjAtNDNlYi1hZmVkLTg0YzI1N2NkY2MxMyIsDQogICJhaWQiOiAiMiIsDQogICJuZXQiOiAid3MiLA0KICAidHlwZSI6ICJub25lIiwNCiAgImhvc3QiOiAid3d3LmJhaWR1LmNvbSIsDQogICJwYXRoIjogIi92MnJheSIsDQogICJ0bHMiOiAidGxzIg0KfQ==

LitCcc, 2019-11-28:
确实是这样，需要手动修改，另外最好给关键参数打下码

willaytoun, 2019-11-28:
这些参数都已经改过了并不能连接，你那边用手机看概况里接口以下会不会出现抖动

coolsnowwolf, 2019-12-11:
[image]
自己解密看就知道了，并没有你说的auto和 insecure

willaytoun, 2019-12-11:
这就奇怪了，您用windows的v2rayN导入看一下，加密方式确实是auto，allowInsecure有可能是默认的话是true。ws+tls加密方式为auto我用v2rayN和ssr+试过很多次了，解密也确实是您的那样，但是二者导入后的参数就是不一样。

closed 2019-12-24 by coolsnowwolf