coolsnowwolf/lede #2162: x86_64最新代码便有libevent_pthreads有问题

zcluo, 2019-11-25:
make[3]: Entering directory '/root/lede/feeds/packages/net/miniupnpc'
mkdir -p /root/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /root/lede/tmp/.root-copy.flock -c 'cp -fpR /root/lede/build_dir/target-x86_64_musl/miniupnpc-2.0.20170509/.pkgdir/libminiupnpc/. /root/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /root/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.libminiupnpc_installed
echo "libminiupnpc" >> /root/lede/staging_dir/target-x86_64_musl/pkginfo/miniupnpc.default.install
make[3]: Leaving directory '/root/lede/feeds/packages/net/miniupnpc'
time: package/feeds/packages/miniupnpc/compile#0.18#0.15#0.32
make[3]: Entering directory '/root/lede/package/libs/libevent2'
mkdir -p /root/lede/staging_dir/target-x86_64_musl/root-x86/stamp
echo '7' | cmp -s - /root/lede/staging_dir/target-x86_64_musl/pkginfo/libevent2.version || echo '7' > /root/lede/staging_dir/target-x86_64_musl/pkginfo/libevent2.version
SHELL= flock /root/lede/tmp/.root-copy.flock -c 'cp -fpR /root/lede/build_dir/target-x86_64_musl/libevent-2.1.11-stable/.pkgdir/libevent2/. /root/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /root/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.libevent2_installed
mkdir -p /root/lede/bin/targets/x86/64/packages /root/lede/build_dir/target-x86_64_musl/libevent-2.1.11-stable/ipkg-x86_64/libevent2-pthreads/CONTROL /root/lede/staging_dir/target-x86_64_musl/pkginfo
install -d -m0755 /root/lede/build_dir/target-x86_64_musl/libevent-2.1.11-stable/ipkg-x86_64/libevent2-pthreads/usr/lib
cp -fpR /root/lede/build_dir/target-x86_64_musl/libevent-2.1.11-stable/ipkg-install/usr/lib/libevent_pthreads-2.1.so.* /root/lede/build_dir/target-x86_64_musl/libevent-2.1.11-stable/ipkg-x86_64/libevent2-pthreads/usr/lib/
cp: cannot stat '/root/lede/build_dir/target-x86_64_musl/libevent-2.1.11-stable/ipkg-install/usr/lib/libevent_pthreads-2.1.so.*': No such file or directory
make[3]: *** [Makefile:168: /root/lede/bin/packages/x86_64/base/libevent2-pthreads_2.1.11-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/root/lede/package/libs/libevent2'
time: package/libs/libevent2/compile#0.22#0.13#0.32
make[2]: *** [package/Makefile:109: package/libs/libevent2/compile] Error 2
make[2]: Leaving directory '/root/lede'
make[1]: *** [package/Makefile:103: /root/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/root/lede'
make: *** [/root/lede/include/toplevel.mk:218: world] Error 2

coolsnowwolf, 2019-11-26:
make clean

zcluo, 2019-11-27:
搞定谢谢

closed 2019-11-27 by zcluo