coolsnowwolf/lede #2165: acrh17更新最新固件后wifi ap无法找到或会断开

aesirteam, 2019-11-26:
系统日志：
Tue Nov 26 08:58:04 2019 kern.warn kernel: [  296.920858] ------------[ cut here ]------------
Tue Nov 26 08:58:04 2019 kern.warn kernel: [  296.925682] WARNING: CPU: 3 PID: 25 at backports-5.4-rc2-1/net/mac80211/util.c:2422 ieee80211_reconfig+0x8e0/0xba8 [mac80211]
Tue Nov 26 08:58:04 2019 kern.warn kernel: [  296.930195] Modules linked in: xt_FULLCONENAT pppoe ppp_async ath10k_pci ath10k_core ath pppox ppp_generic nf_nat_pptp nf_conntrack_pptp nf_conntrack_ipv6 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_socket xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TPROXY xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD xt_CT ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm slhc nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_redirect nf_nat_proto_gre nf_nat_masquerade_ipv4 nf_nat_irc nf_conntrack_ipv4 nf_nat_ipv4 nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4
Tue Nov 26 08:58:04 2019 kern.warn kernel: [  296.989519]  nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conntrack macvlan iptable_raw iptable_mangle iptable_filter ip_tables crc_ccitt compat fuse tcp_bbr cryptodev xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables dummy vfat fat nls_utf8 nls_iso8859_1 nls_cp437 authenc usb_storage leds_gpio
Tue Nov 26 08:58:04 2019 kern.warn kernel: [  297.060598]  xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_of_simple sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache exfat crc32c_generic
Tue Nov 26 08:58:04 2019 kern.warn kernel: [  297.082832] CPU: 3 PID: 25 Comm: kworker/3:0 Tainted: G        W       4.14.155 #0
Tue Nov 26 08:58:04 2019 kern.warn kernel: [  297.095207] Hardware name: Generic DT based system
Tue Nov 26 08:58:04 2019 kern.warn kernel: [  297.102928] Workqueue: events_freezable ieee80211_restart_work [mac80211]
Tue Nov 26 08:58:04 2019 kern.warn kernel: [  297.107636] [<c030e81c>] (unwind_backtrace) from [<c030ad3c>] (show_stack+0x10/0x14)
Tue Nov 26 08:58:04 2019 kern.warn kernel: [  297.114487] [<c030ad3c>] (show_stack) from [<c0780234>] (dump_stack+0x94/0xa8)
Tue Nov 26 08:58:04 2019 kern.warn kernel: [  297.122297] [<c0780234>] (dump_stack) from [<c031a8e8>] (__warn+0xe4/0x100)
Tue Nov 26 08:58:04 2019 kern.warn kernel: [  297.129328] [<c031a8e8>] (__warn) from [<c031a9b4>] (warn_slowpath_null+0x20/0x28)
Tue Nov 26 08:58:04 2019 kern.warn kernel: [  297.136239] [<c031a9b4>] (warn_slowpath_null) from [<bf5ab22c>] (ieee80211_reconfig+0x8e0/0xba8 [mac80211])
Tue Nov 26 08:58:04 2019 kern.warn kernel: [  297.143921] [<bf5ab22c>] (ieee80211_reconfig [mac80211]) from [<bf57881c>] (ieee80211_restart_work+0xb4/0xc0 [mac80211])
Tue Nov 26 08:58:04 2019 kern.warn kernel: [  297.153516] [<bf57881c>] (ieee80211_restart_work [mac80211]) from [<c03305f8>] (process_one_work+0x240/0x3d0)
Tue Nov 26 08:58:04 2019 kern.warn kernel: [  297.164579] [<c03305f8>] (process_one_work) from [<c0331594>] (worker_thread+0x330/0x560)
Tue Nov 26 08:58:04 2019 kern.warn kernel: [  297.174385] [<c0331594>] (worker_thread) from [<c0335df0>] (kthread+0x148/0x154)
Tue Nov 26 08:58:04 2019 kern.warn kernel: [  297.182543] [<c0335df0>] (kthread) from [<c0307768>] (ret_from_fork+0x14/0x2c)
Tue Nov 26 08:58:04 2019 kern.warn kernel: [  297.190034] ---[ end trace 7dc153624346dcee ]---
Tue Nov 26 08:58:04 2019 kern.err kernel: [  297.197294] wlan0: failed to set key (1, ff:ff:ff:ff:ff:ff) to hardware (-70)
Tue Nov 26 08:58:06 2019 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Tue Nov 26 08:58:06 2019 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Tue Nov 26 08:58:10 2019 kern.warn kernel: [  303.230020] ath10k_warn: 26 callbacks suppressed
Tue Nov 26 08:58:10 2019 kern.warn kernel: [  303.230179] ath10k_pci 0000:01:00.0: failed to set beacon mode for vdev 0: -70
Tue Nov 26 08:58:10 2019 kern.warn kernel: [  303.234231] ath10k_pci 0000:01:00.0: failed to set dtim period for vdev 0: -70
Tue Nov 26 08:58:10 2019 kern.warn kernel: [  303.243499] ath10k_pci 0000:01:00.0: failed to recalculate rts/cts prot for vdev 0: -70
Tue Nov 26 08:58:10 2019 kern.warn kernel: [  303.248274] ath10k_pci 0000:01:00.0: failed to set preamble for vdev 0: -70
Tue Nov 26 08:58:10 2019 kern.warn kernel: [  303.256159] ath10k_pci 0000:01:00.0: failed to set mgmt tx rate -70
Tue Nov 26 08:58:16 2019 kern.warn kernel: [  309.270276] ath10k_pci 0000:01:00.0: failed to set beacon mode for vdev 0: -70
Tue Nov 26 08:58:16 2019 kern.warn kernel: [  309.270693] ath10k_pci 0000:01:00.0: failed to set dtim period for vdev 0: -70
Tue Nov 26 08:58:16 2019 kern.warn kernel: [  309.278258] ath10k_pci 0000:01:00.0: failed to recalculate rts/cts prot for vdev 0: -70
Tue Nov 26 08:58:16 2019 kern.warn kernel: [  309.283975] ath10k_pci 0000:01:00.0: failed to set preamble for vdev 0: -70
Tue Nov 26 08:58:16 2019 kern.warn kernel: [  309.292556] ath10k_pci 0000:01:00.0: failed to set mgmt tx rate -70
Tue Nov 26 08:58:22 2019 cron.info crond[15310]: crond (busybox 1.30.0) started, log level 8
Tue Nov 26 08:58:22 2019 kern.warn kernel: [  315.304744] ath10k_pci 0000:01:00.0: failed to set beacon mode for vdev 0: -70
Tue Nov 26 08:58:22 2019 kern.warn kernel: [  315.304863] ath10k_pci 0000:01:00.0: failed to set dtim period for vdev 0: -70
Tue Nov 26 08:58:22 2019 kern.warn kernel: [  315.311105] ath10k_pci 0000:01:00.0: failed to recalculate rts/cts prot for vdev 0: -70
Tue Nov 26 08:58:22 2019 kern.warn kernel: [  315.318217] ath10k_pci 0000:01:00.0: failed to set preamble for vdev 0: -70
Tue Nov 26 08:58:23 2019 kern.warn kernel: [  315.326074] ath10k_pci 0000:01:00.0: failed to set mgmt tx rate -70
Tue Nov 26 08:58:23 2019 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Tue Nov 26 08:58:23 2019 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Tue Nov 26 08:58:25 2019 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Tue Nov 26 08:58:25 2019 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Tue Nov 26 08:58:29 2019 kern.warn kernel: [  321.339093] ath10k_pci 0000:01:00.0: failed to set beacon mode for vdev 0: -70
Tue Nov 26 08:58:29 2019 kern.warn kernel: [  321.339503] ath10k_pci 0000:01:00.0: failed to set dtim period for vdev 0: -70
Tue Nov 26 08:58:29 2019 kern.warn kernel: [  321.348467] ath10k_pci 0000:01:00.0: failed to recalculate rts/cts prot for vdev 0: -70
Tue Nov 26 08:58:29 2019 kern.warn kernel: [  321.352795] ath10k_pci 0000:01:00.0: failed to set preamble for vdev 0: -70
Tue Nov 26 08:58:29 2019 kern.warn kernel: [  321.361501] ath10k_pci 0000:01:00.0: failed to set mgmt tx rate -70
Tue Nov 26 08:58:35 2019 kern.warn kernel: [  327.380608] ath10k_pci 0000:01:00.0: failed to set beacon mode for vdev 0: -70
Tue Nov 26 08:58:35 2019 kern.warn kernel: [  327.381020] ath10k_pci 0000:01:00.0: failed to set dtim period for vdev 0: -70
Tue Nov 26 08:58:35 2019 kern.warn kernel: [  327.388844] ath10k_pci 0000:01:00.0: failed to recalculate rts/cts prot for vdev 0: -70
Tue Nov 26 08:58:35 2019 kern.warn kernel: [  327.394306] ath10k_pci 0000:01:00.0: failed to set preamble for vdev 0: -70
Tue Nov 26 08:58:35 2019 kern.warn kernel: [  327.402737] ath10k_pci 0000:01:00.0: failed to set mgmt tx rate -70
Tue Nov 26 08:58:36 2019 cron.info crond[15990]: crond (busybox 1.30.0) started, log level 8
Tue Nov 26 08:58:36 2019 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Tue Nov 26 08:58:36 2019 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Tue Nov 26 08:58:38 2019 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Tue Nov 26 08:58:38 2019 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
[... cut ...]
[  574.907812] ath10k_pci 0000:01:00.0: failed to set preamble for vdev 0: -70
[  574.915546] ath10k_pci 0000:01:00.0: failed to set mgmt tx rate -70
[  580.929069] ath10k_pci 0000:01:00.0: failed to set beacon mode for vdev 0: -70
[  580.929119] ath10k_pci 0000:01:00.0: failed to set dtim period for vdev 0: -70
[  580.935283] ath10k_pci 0000:01:00.0: failed to recalculate rts/cts prot for vdev 0: -70
[  580.942495] ath10k_pci 0000:01:00.0: failed to set preamble for vdev 0: -70
[  580.950383] ath10k_pci 0000:01:00.0: failed to set mgmt tx rate -70

solionw, 2019-11-26:
一样 换为ath10k-ct就没这个问题了

coolsnowwolf, 2019-11-26:
看起来是新版的 ath10k 的问题，fw 崩溃了

aesirteam, 2019-11-27:
拉取最新代码，此问题解决！！！close #2165

aesirteam, 2019-11-27:
拉取最新代码，此问题解决！！！

closed 2019-11-27 by aesirteam