coolsnowwolf/lede #2167: 编译固件没选vpn，为什么ipsec的进程还会启动呢

82kg, 2019-11-26:
[image]
如图上述两个进程，luci都没vpn 还是会启动。

[image]
不用VPN的话 ，防火墙里的这两个规则可以删掉么

Her0R0cket, 2019-11-26:
luci只是web界面,你取消了luci-app,但没取消它所依赖的应用

kktt007, 2019-11-26:
我也看到了。我们取消的只是他的luci-app，但是系统最小配合和你选择的依赖需要他。Lean大的固件其实蛮精简的了。我这个强迫症也不挑剔了。你看默认需要设置的都被lean强制选择了。

82kg, 2019-11-26:
> 我也看到了。我们取消的只是他的luci-app，但是系统最小配合和你选择的依赖需要他。Lean大的固件其实蛮精简的了。我这个强迫症也不挑剔了。你看默认需要设置的都被lean强制选择了。

我把这两个进程关掉了，好像也没啥影响么，防火墙里的这两个规则可以删除么

EnnawYang, 2019-11-26:
vpn没有需求的话network里面可以取消掉的

82kg, 2019-11-26:
> vpn没有需求的话network里面可以取消掉的

我以为只要luci里取消掉就行了，多谢啊

closed 2019-11-26 by 82kg