coolsnowwolf/lede #2168: 游戏模式UDP中继服务器BUG

cntzjjs, 2019-11-26:
路由：Lean R9.11.9
插件：SSR Plus+
现象：V2ray ws+tls下 游戏模式UDP中继服务器 游戏可以连接 但是会出现突然延迟增高，闪断的情况。测试了几款游戏均为此表现，但是同样配置换做软件均正常。

trysh, 2019-11-27:
Lean R9.11.23
SSR Plus+
ss节点用客户端测试是支持udp的。但是在SSR Plus+中启用了udp，游戏中的udp却不走代理

Iy204, 2019-11-30:
这个问题，我在原版Ss-libev也遇到了，一开始以为是lean代码的问题，
后来跑回op也是。应该是其他包跟随上游出问题了。Nat测试都是Udpblocked
SSr plus SS-libev 更换SS SSR协议 都不行。就这几天的版本之后开始的。

trysh, 2019-11-30:
> 这个问题，我在原版Ss-libev也遇到了，一开始以为是lean代码的问题，
> 后来跑回op也是。应该是其他包跟随上游出问题了。Nat测试都是Udpblocked
> SSr plus SS-libev 更换SS SSR协议 都不行。就这几天的版本之后开始的。

感谢跟踪。十分感谢。如果过几天后的版本解决这个问题了，麻烦回复此消息，让我知道。谢谢

Iy204, 2019-12-01:
> 
> 
> > 这个问题，我在原版Ss-libev也遇到了，一开始以为是lean代码的问题，
> > 后来跑回op也是。应该是其他包跟随上游出问题了。Nat测试都是Udpblocked
> > SSr plus SS-libev 更换SS SSR协议 都不行。就这几天的版本之后开始的。
> 
> 感谢跟踪。十分感谢。如果过几天后的版本解决这个问题了，麻烦回复此消息，让我知道。谢谢

上游openwrt已经更新到4.14.156已经好了。

trysh, 2019-12-01:
> > > 这个问题，我在原版Ss-libev也遇到了，一开始以为是lean代码的问题，
> > > 后来跑回op也是。应该是其他包跟随上游出问题了。Nat测试都是Udpblocked
> > > SSr plus SS-libev 更换SS SSR协议 都不行。就这几天的版本之后开始的。
> > 
> > 
> > 感谢跟踪。十分感谢。如果过几天后的版本解决这个问题了，麻烦回复此消息，让我知道。谢谢
> 
> 上游openwrt已经更新到4.14.156已经好了。

十分感谢！

Iy204, 2019-12-01:
> 
> 
> > > > 这个问题，我在原版Ss-libev也遇到了，一开始以为是lean代码的问题，
> > > > 后来跑回op也是。应该是其他包跟随上游出问题了。Nat测试都是Udpblocked
> > > > SSr plus SS-libev 更换SS SSR协议 都不行。就这几天的版本之后开始的。
> > > 
> > > 
> > > 感谢跟踪。十分感谢。如果过几天后的版本解决这个问题了，麻烦回复此消息，让我知道。谢谢
> > 
> > 
> > 上游openwrt已经更新到4.14.156已经好了。
> 
> 十分感谢！

我看错了。是18.06.5，没跑18.06。
所以回退回去是正常的。

trysh, 2019-12-01:
> > > > > 这个问题，我在原版Ss-libev也遇到了，一开始以为是lean代码的问题，
> > > > > 后来跑回op也是。应该是其他包跟随上游出问题了。Nat测试都是Udpblocked
> > > > > SSr plus SS-libev 更换SS SSR协议 都不行。就这几天的版本之后开始的。
> > > > 
> > > > 
> > > > 感谢跟踪。十分感谢。如果过几天后的版本解决这个问题了，麻烦回复此消息，让我知道。谢谢
> > > 
> > > 
> > > 上游openwrt已经更新到4.14.156已经好了。
> > 
> > 
> > 十分感谢！
> 
> 我看错了。是18.06.5，没跑18.06。
> 所以回退回去是正常的。

回退？ 我刚编译好 github.com/coolsnowwolf/lede 的最新git pull

Iy204, 2019-12-01:
> 回退？ 我刚编译好 [github.com/coolsnowwolf/lede](http://github.com/coolsnowwolf/lede) 的最新git pull

抄的自动编译没看对方的版本搞错了。lean这里还没更新内核呢。等我看看op最新的内核再告诉你吧。

trysh, 2019-12-01:
> 抄的自动编译没看对方的版本搞错了。lean这里还没更新内核呢。等我看看op最新的内核再告诉你吧。

好的。十分感谢。

Iy204, 2019-12-02:
> 好的。十分感谢。

Openwrt在新内核下正常了。昨晚lean跟随升了内核了，现在都正常了，

trysh, 2019-12-02:
> Openwrt在新内核下正常了。昨晚lean跟随升了内核了，现在都正常了，

哇哦！我这就更新lean lede，感谢你，感谢lean

trysh, 2019-12-02:
> > 
> > 
> > Openwrt在新内核下正常了。昨晚lean跟随升了内核了，现在都正常了，
> 
> 哇哦！我这就更新lean lede，感谢你，感谢lean

我是git pull
然后./scripts/feeds update -a 
   ./scripts/feeds install -a
make clean
然后编译的。
安装好了之后发现好多地方有版本号啊
[image]
我该从哪里看版本号确认我已经是用上了最新的版本？

EnnawYang, 2019-12-02:
> > > Openwrt在新内核下正常了。昨晚lean跟随升了内核了，现在都正常了，
> > 
> > 
> > 哇哦！我这就更新lean lede，感谢你，感谢lean
> 
> 我是git pull
> 然后./scripts/feeds update -a
> ./scripts/feeds install -a
> make clean
> 然后编译的。
> 安装好了之后发现好多地方有版本号啊
> [image]
> 我该从哪里看版本号确认我已经是用上了最新的版本？

4.19.86

trysh, 2019-12-02:
> > > > Openwrt在新内核下正常了。昨晚lean跟随升了内核了，现在都正常了，
> > > 
> > > 
> > > 哇哦！我这就更新lean lede，感谢你，感谢lean
> > 
> > 
> > 我是git pull
> > 然后./scripts/feeds update -a
> > ./scripts/feeds install -a
> > make clean
> > 然后编译的。
> > 安装好了之后发现好多地方有版本号啊
> > [image]
> > 我该从哪里看版本号确认我已经是用上了最新的版本？
> 
> 4.19.86

可是他提到好多次版本啊，又是“上游openwrt已经更新到4.14.156已经好了。”，然后又是“我看错了。是18.06.5，没跑18.06。”
我都看晕了

Iy204, 2019-12-02:
> 
> 
> > > > > Openwrt在新内核下正常了。昨晚lean跟随升了内核了，现在都正常了，
> > > > 
> > > > 
> > > > 哇哦！我这就更新lean lede，感谢你，感谢lean
> > > 
> > > 
> > > 我是git pull
> > > 然后./scripts/feeds update -a
> > > ./scripts/feeds install -a
> > > make clean
> > > 然后编译的。
> > > 安装好了之后发现好多地方有版本号啊
> > > [image]
> > > 我该从哪里看版本号确认我已经是用上了最新的版本？
> > 
> > 
> > 4.19.86
> 
> 可是他提到好多次版本啊，又是“上游openwrt已经更新到4.14.156已经好了。”，然后又是“我看错了。是18.06.5，没跑18.06。”
> 我都看晕了

4.14.156是linux内核版本，不过你这个编译的是4.19内核，这个4.19.86也是最新的。
18.06是openwrt的版本号。

trysh, 2019-12-02:
> 
> 4.14.156是linux内核版本，不过你这个编译的是4.19内核，这个4.19.86也是最新的。
> 18.06是openwrt的版本号。

“18.06是openwrt的版本号”从路由器web后台里看不到吗？ 不管了，反正我是今天git pull 然后make clean 然后编译的，应该可以udp了吧？

cntzjjs, 2019-12-02:
我这边v2ray的udp还是不正常。ss正常

closed 2019-12-07 by coolsnowwolf