coolsnowwolf/lede #2170: EXSi虚拟里添加4个及以上的网口时显示不正常

yocosweet, 2019-11-26:
EXSi虚拟里添加4个及以上的网口时显示不正常

EnnawYang, 2019-11-26:
我的是实体机，四个网口，最后一个接口显示不正常

yocosweet, 2019-11-26:
> 我的是实体机，四个网口，最后一个接口显示不正常

如果添加4个网口，最后一个显示半双工，如果添加5个网口，概况界面的内存，接口等界面不显示

coolsnowwolf, 2019-11-26:
运行 ethinfo

dolphinpaopao, 2019-11-26:
[image]
[image]
共3个网口，同样问题，最后一个没有使用的网口显示不正常。

dolphinpaopao, 2019-11-26:
[image]
[image]

coolsnowwolf, 2019-11-26:
你们都在用老版本

dolphinpaopao, 2019-11-26:
> 你们都在用老版本

老大，今天早上才全新git pull编译的啊，刚看了今天下午好像又更新了，我晚上再试试后汇报。

coolsnowwolf, 2019-11-26:
估计是从第5个网口开始的显示代码有问题。话说你们真变态啊，有需要那么多wan 么

dolphinpaopao, 2019-11-26:
> 估计是从第5个网口开始的显示代码有问题。话说你们真变态啊，有需要那么多wan 么

我只有3个网口，网上的成品软路由很多都是4网口或者更多啊。

closed 2019-11-27 by coolsnowwolf