coolsnowwolf/lede #2178: Unable to compile two .bin files TP-LINK WR1043ND

jeremy168, 2019-11-26:
Hello there
Would like to ask, I compiled a routing device and found a problem
Is that the following firmware files are not produced after compilation
* -squashfs-factory.bin
* -squashfs-sysupgrade.bin

I tried to adjust the menuconfig configuration and nothing improved
This device model TP-LINK WR1043ND V1
His FLASH space is 8MB, I don't know if it was compiled
Is the document too large? How to improve it? Thank you for your help

batmancar, 2019-11-26:
[image]
我也有同样的问题,求助

KFERMercer, 2019-11-26:
插件过多

jeremy168, 2019-11-27:
謝謝你的回答,不過拿掉一些差件我有想過試過,但還是一樣情況,
請問是不是menuconfig選完裝置硬體其他都保持預設不變直接編譯呢?
我試過選幾個luci內的app也是不行,可否有建議menuconfig配置選項建議呢??

jeremy168, 2019-11-27:
> [image]
> 我也有同样的问题,求助

你好
你也是TL-WR1043ND這一台嗎??

outemancut, 2019-11-27:
8M的玩意让你选了SSR全家桶还想生成？什么都不编译进去 单单内核就要6M 你还加个V2R  插件单独压缩过的就3.多M 能生成就有鬼了

xlf1234, 2019-11-27:
插件太多，换16M flash，或者去掉非必要插件以及附带组件make dirclean 重新编译

KFERMercer, 2019-11-27:
> 謝謝你的回答,不過拿掉一些插件我有想過試過,但還是一樣情況,
> 請問是不是menuconfig選完裝置硬體其他都保持預設不變直接編譯呢?
> 我試過選幾個luci內的app也是不行,可否有建議menuconfig配置選項建議呢??

删掉`/.config`, 重新生成设定档.

jeremy168, 2019-11-27:
> > 謝謝你的回答,不過拿掉一些插件我有想過試過,但還是一樣情況,
> > 請問是不是menuconfig選完裝置硬體其他都保持預設不變直接編譯呢?
> > 我試過選幾個luci內的app也是不行,可否有建議menuconfig配置選項建議呢??
> 
> 删掉`/.config`, 重新生成设定档.

重新生成.config有比較推的選項嗎?還是直接生成直接編譯呢??

jeremy168, 2019-11-27:
> 8M的玩意让你选了SSR全家桶还想生成？什么都不编译进去 单单内核就要6M 你还加个V2R 插件单独压缩过的就3.多M 能生成就有鬼了

所以這樣說來,v2ray和ssr plus這些都不能用囉,如果重新生成.config有比較建議的選擇嗎?
還有如何判斷是否插件過多的問題呢??

jeremy168, 2019-11-27:
> […](#)

好的,看來也只有這樣囉!其實之前有刪除一些插件,但可能還是太多所以還是不行生成
主要還不太清楚這個再.config插件是否過多選擇判斷

closed 2019-11-28 by coolsnowwolf