coolsnowwolf/lede #2182: SSR PLUS 问题

rong2d, 2019-11-27:
大雕好， 最近编译的固件，发现SSR plus有个很诡异的问题，使用gfwlist模式，启动SSR后，正常访问墙内墙外，但过了几分钟后，似乎所有链路都走墙外，好似“全局模式”，最直接的反映就是优酷爱奇艺等网站显示IP在海外，无法播放，其他墙内网站明显访问变慢。重启SSR服务后恢复正常，过几分钟后该问题又重现。。。烦请看下这个问题。

coolsnowwolf, 2019-11-27:
看看 IPv6 是否去除了

rong2d, 2019-11-27:
好，回去重新编译下再试试。谢谢

rong2d, 2019-11-27:
拉了最新的代码，把无线和ipv6的组件都去掉了重新编译了下，问题仍在，ip.cn，https://whoer.net/zh# 等显示的都是墙外IP。。。打开墙内的网站，明显加载慢。。

快疯了。。求大雕帮忙看看是什么问题

coolsnowwolf, 2019-11-28:
估计你是所谓的“旁路由”之类了

rong2d, 2019-11-28:
就只是一个软路由裸装了openwrt
还可能是什么情况导致的？之前用的也是你的源码编译的固件，一切都好好的。。就上周末更新好之后。。。T_T

closed 2019-11-28 by coolsnowwolf

WoodC2011, 2020-10-19:
> 大雕好， 最近编译的固件，发现SSR plus有个很诡异的问题，使用gfwlist模式，启动SSR后，正常访问墙内墙外，但过了几分钟后，似乎所有链路都走墙外，好似“全局模式”，最直接的反映就是优酷爱奇艺等网站显示IP在海外，无法播放，其他墙内网站明显访问变慢。重启SSR服务后恢复正常，过几分钟后该问题又重现。。。烦请看下这个问题。

你最终怎么搞定的

rong2d, 2020-11-01:
> > 大雕好， 最近编译的固件，发现SSR plus有个很诡异的问题，使用gfwlist模式，启动SSR后，正常访问墙内墙外，但过了几分钟后，似乎所有链路都走墙外，好似“全局模式”，最直接的反映就是优酷爱奇艺等网站显示IP在海外，无法播放，其他墙内网站明显访问变慢。重启SSR服务后恢复正常，过几分钟后该问题又重现。。。烦请看下这个问题。
> 
> 你最终怎么搞定的

是那个黑白名单有BUG，把黑白名单清空就好了。

WoodC2011, 2020-11-02:
> > > 大雕好，最近编译的固件，发现SSR加有个很奇怪的怪异的问题，使用gfwlist模式，启动SSR后，正常访问墙内墙外，但过了几分钟后，似乎所有都都走墙外，好似“层次模式”，最直接的反映就是优酷爱奇艺等网站显示IP在海外，无法播放，其他墙内网站明显访问变慢。重新启动SSR服务后恢复正常，过几分钟后该问题又重现。。。烦请看下这个问题。
> > 
> > 
> > 你最终怎么搞定的
> 
> 是那个黑白名单有BUG，把黑白名单清空就好了。

谢谢，我后来发现绕过中国IP模式下，iptables -t nat -I POSTROUTING -o eth0 -j MASQUERADE这条防火墙规则会影响国内上网，去掉就没事了，搞不懂