coolsnowwolf/lede #2183: 最新代码x86_64固件编译失败！mac80211.ko missing

aesirteam, 2019-11-27:
ake[3]: Entering directory '/home/zhongkui/lede/package/kernel/mac80211'
mkdir -p /home/zhongkui/lede/bin/targets/x86/64/packages /home/zhongkui/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.85-1/ipkg-x86_64/kmod-mac80211/CONTROL /home/zhongkui/lede/staging_dir/target-x86_64_musl/pkginfo
ERROR: module '/home/zhongkui/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.85-1/net/mac80211/mac80211.ko' is missing.
Makefile:561: recipe for target '/home/zhongkui/lede/bin/targets/x86/64/packages/kmod-mac80211_4.19.85+4.19.85-1-2_x86_64.ipk' failed
make[3]: *** [/home/zhongkui/lede/bin/targets/x86/64/packages/kmod-mac80211_4.19.85+4.19.85-1-2_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/zhongkui/lede/package/kernel/mac80211'
time: package/kernel/mac80211/compile#0.35#0.11#0.46
package/Makefile:107: recipe for target 'package/kernel/mac80211/compile' failed
make[2]: *** [package/kernel/mac80211/compile] Error 2
make[2]: Leaving directory '/home/zhongkui/lede'
package/Makefile:103: recipe for target '/home/zhongkui/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/zhongkui/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zhongkui/lede'
/home/zhongkui/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

EnnawYang, 2019-11-27:
同，编译了4次都一样

f3404923, 2019-11-27:
我有时候也会出现，make dirclean、make clean、rm -rf tmp后在update & install基本都能解决。你可以去试试，第一次用-j1, 别图快

jacklykc, 2019-11-27:
我今天编译也出现同样的问题。将官方/package/kernel/mac80211目录替换lean的相应文件，重新编译后OK。

xinxijishuwyq, 2019-11-27:
同

coolsnowwolf, 2019-11-27:
系统用 Ubuntu 14 即可，其他系统都不能编译 4.19 内核的 mac80211

vaxoyik58, 2019-11-27:
16.04应该也可以，今天刚编译的，全新的，用的应该是这个版本。[https://github.com/coolsnowwolf/lede/commit/6773af4631adee965938d90d9164a003e46e127d](url)

aesirteam, 2019-11-28:
> 系统用 Ubuntu 14 即可，其他系统都不能编译 4.19 内核的 mac80211
没拉取最新代码前，ubuntu18编译成功过x86_64 kernel 4.19的

EnnawYang, 2019-11-28:
> 系统用 Ubuntu 14 即可，其他系统都不能编译 4.19 内核的 mac80211

用Ubuntu14.04编译还是同样的错误，之前ubuntu18编译也没问题么

helianling, 2019-11-28:
和ubuntu版本没有关系，用默认.config就可以编译通过，自己改了就不行了。以前也这样。以前是先用默认配置先编译过一次，再自己选择自己需要的。现在默认配置也可以过，但自己一改就不行了。

BOBINIUNIU, 2019-11-28:
把wpad 那个组件去掉就可以正常编译了

f3404923, 2019-11-28:
> 系统用 Ubuntu 14 即可，其他系统都不能编译 4.19 内核的 mac80211

同样，18.04，14.04都是报这个错误。好像就是从11.26更新过后就出现这个问题，之前编译都没问题

f3404923, 2019-11-28:
> 把wpad 那个组件去掉就可以正常编译了

能说的详细点吗，wpad在那个menu里面啊？

EnnawYang, 2019-11-28:
> > 把wpad 那个组件去掉就可以正常编译了
> 
> 能说的详细点吗，wpad在那个menu里面啊？

network

f3404923, 2019-11-28:
> > > 把wpad 那个组件去掉就可以正常编译了
> > 
> > 
> > 能说的详细点吗，wpad在那个menu里面啊？
> 
> network - wpad，基础组件不建议移除，除非你用不到

我的是x86软路由用，没有wifi模块

EnnawYang, 2019-11-28:
刚刚把原有.config删除并使用默认的配置重新编译编译成功，后来加上自己需要的组件编译也成功，系统ubuntu14.04

aesirteam, 2019-11-28:
修改target/linux/x86/Makefile中的KERNEL_PATCHVER为4.14,
执行编译成功

closed 2019-11-28 by aesirteam