coolsnowwolf/lede #2184: x86固件编译提示qemu-ga_4.1.1-1_x86_64.ipk安装失败

happyzhang1995, 2019-11-27:
[image]
提示找不到/home/pve/openwrt/lede/build_dir/target-x86_64_musl/qemu-4.1.1/ipkg-install/qemu-ga文件夹，用winscp连接查看后文件地址的确没有qemu-ga

iamggs, 2019-11-27:
make menuconfig的时候，到Utilities ---> Virtualizatio ---> 中去把qemu-ga去掉。

其实我也需要qemu-ga，但只要选中这项再编译就会报错，目前只有不选中，部署后再手动安装。
`opkg qemu-ga`

iamggs, 2019-11-27:
可以参考这个：
https://github.com/coolsnowwolf/lede/issues/2179

happyzhang1995, 2019-11-27:
> 可以参考这个：
> #2179

L大已经修复了，重新pull没有这个问题了

closed 2019-11-27 by happyzhang1995

RainCat1998, 2019-11-29:
> > 可以参考这个：
> > #2179
> 
> L大已经修复了，重新pull没有这个问题了

还是有问题，路径还是不正确。重新pull的。

RainCat1998, 2019-11-29:
@happyzhang1995 
[image]
最后qemu-ga只生成在根目录，ipkg_install目录下面没有。

RainCat1998, 2019-11-29:
解决方法很简单，出错了，就把编译过程中生成的qemu-ga文件放到target-x86_64_musl\ipkg-install\usr\bin 目录下，再继续编译就行了。

iamggs, 2019-11-30:
对的，还是有问题，希望@coolsnowwolf能彻底修复。