coolsnowwolf/lede #2188: 重新clone后编译出错，翻出去了的。

jxph1123, 2019-11-27:
make[6]: Leaving directory '/home/etek/lede/build_dir/toolchain-mipsel_24kc_gcc-7.4.0_musl/gcc-7.4.0-initial/libbacktrace'
make[5]: Leaving directory '/home/etek/lede/build_dir/toolchain-mipsel_24kc_gcc-7.4.0_musl/gcc-7.4.0-initial/libbacktrace'
make[5]: Entering directory '/home/etek/lede/build_dir/toolchain-mipsel_24kc_gcc-7.4.0_musl/gcc-7.4.0-initial/libcpp'
test -f config.h || (rm -f stamp-h1 && make stamp-h1)
make[5]: Leaving directory '/home/etek/lede/build_dir/toolchain-mipsel_24kc_gcc-7.4.0_musl/gcc-7.4.0-initial/libcpp'
make[5]: Entering directory '/home/etek/lede/build_dir/toolchain-mipsel_24kc_gcc-7.4.0_musl/gcc-7.4.0-initial/libdecnumber'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/etek/lede/build_dir/toolchain-mipsel_24kc_gcc-7.4.0_musl/gcc-7.4.0-initial/libdecnumber'
make[5]: Entering directory '/home/etek/lede/build_dir/toolchain-mipsel_24kc_gcc-7.4.0_musl/gcc-7.4.0-initial/fixincludes'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/etek/lede/build_dir/toolchain-mipsel_24kc_gcc-7.4.0_musl/gcc-7.4.0-initial/fixincludes'
make[5]: Entering directory '/home/etek/lede/build_dir/toolchain-mipsel_24kc_gcc-7.4.0_musl/gcc-7.4.0-initial/gcc'
build/genautomata /home/etek/lede/build_dir/toolchain-mipsel_24kc_gcc-7.4.0_musl/gcc-7.4.0/gcc/common.md /home/etek/lede/build_dir/toolchain-mipsel_24kc_gcc-7.4.0_musl/gcc-7.4.0/gcc/config/mips/mips.md \
  insn-conditions.md > tmp-automata.c
/bin/bash: line 1:  7176 Killed                  build/genautomata /home/etek/lede/build_dir/toolchain-mipsel_24kc_gcc-7.4.0_musl/gcc-7.4.0/gcc/common.md /home/etek/lede/build_dir/toolchain-mipsel_24kc_gcc-7.4.0_musl/gcc-7.4.0/gcc/config/mips/mips.md insn-conditions.md > tmp-automata.c
Makefile:2278: recipe for target 's-automata' failed
make[5]: *** [s-automata] Error 137
make[5]: Leaving directory '/home/etek/lede/build_dir/toolchain-mipsel_24kc_gcc-7.4.0_musl/gcc-7.4.0-initial/gcc'
Makefile:4210: recipe for target 'all-gcc' failed
make[4]: *** [all-gcc] Error 2
make[4]: Leaving directory '/home/etek/lede/build_dir/toolchain-mipsel_24kc_gcc-7.4.0_musl/gcc-7.4.0-initial'
Makefile:36: recipe for target '/home/etek/lede/build_dir/toolchain-mipsel_24kc_gcc-7.4.0_musl/gcc-7.4.0-initial/.built' failed
make[3]: *** [/home/etek/lede/build_dir/toolchain-mipsel_24kc_gcc-7.4.0_musl/gcc-7.4.0-initial/.built] Error 2
make[3]: Leaving directory '/home/etek/lede/toolchain/gcc/initial'
time: toolchain/gcc/initial/compile#8.65#1.80#14.25
toolchain/Makefile:98: recipe for target 'toolchain/gcc/initial/compile' failed
make[2]: *** [toolchain/gcc/initial/compile] Error 2
make[2]: Leaving directory '/home/etek/lede'
toolchain/Makefile:96: recipe for target '/home/etek/lede/staging_dir/toolchain-mipsel_24kc_gcc-7.4.0_musl/stamp/.toolchain_compile' failed
make[1]: *** [/home/etek/lede/staging_dir/toolchain-mipsel_24kc_gcc-7.4.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/etek/lede'
/home/etek/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

jxph1123, 2019-11-27:
编译了几次都这样

jeremy168, 2019-11-27:
我遇到這一種诶都砍掉整個lede然後重頭過,有時是檔案下載不完整
下make clean再編一次看看

BOBINIUNIU, 2019-11-28:
wpad那个组件去掉就编译成功了

jacklykc, 2019-11-28:
> wpad那个组件去掉就编译成功了

wpad那个组件具体在哪个位置呢？没找到这个组件。谢谢！

EnnawYang, 2019-11-28:
> wpad那个组件去掉就编译成功了

兄弟你看过他的错误跟你一样了吗？

jacklykc, 2019-11-28:
> > wpad那个组件去掉就编译成功了
> 
> 兄弟你看过他的错误跟你一样了吗？

不一样，但我看过他在80211错误的回复也是这样操作，借鉴一下。

jeremy168, 2019-11-28:
其實有些插件可能源碼都失效囉,我要編譯openflow或openvswitch沒有一次成功的
好像curl 都get不到openvswutch.tar.gz檔案,連結失效囉!

jxph1123, 2019-11-28:
还是不行。。。。

BOBINIUNIU, 2019-11-29:
wpad 在主菜单的Network里 wpad.............................. IEEE 802.1Xxxxx

closed 2019-12-07 by coolsnowwolf

tufu9441, 2019-12-08:
@jxph1123 请问现在解决了么？我这两天也遇到了同样的问题，不知道这个toolchain是个什么东西😂……不得已我退回到了12月2日的版本，才编译通过的。
[image]

tufu9441, 2019-12-11:
最终发现是内存不够的问题，改大了就好了。See #528