coolsnowwolf/lede #2189: 全新编译r6190ac  报错

axgaoyu, 2019-11-27:
Created:         Wed Nov 27 05:51:20 2019
 Image 0 (kernel@1)
  Description:  ARM OpenWrt Linux-4.14.155
  Created:      Wed Nov 27 05:51:20 2019
  Type:         Kernel Image
  Compression:  uncompressed
  Data Size:    2271728 Bytes = 2218.48 KiB = 2.17 MiB
  Architecture: ARM
  OS:           Linux
  Load Address: 0x80208000
  Entry Point:  0x80208000
  Hash algo:    crc32
  Hash value:   09459da9
  Hash algo:    sha1
  Hash value:   56383b234c47cdc90d64301a1b099ca581f47680
 Image 1 (fdt@1)
  Description:  ARM OpenWrt p2w_r619ac device tree blob
  Created:      Wed Nov 27 05:51:20 2019
  Type:         Flat Device Tree
  Compression:  uncompressed
  Data Size:    16653 Bytes = 16.26 KiB = 0.02 MiB
  Architecture: ARM
  Hash algo:    crc32
  Hash value:   e61b2d1d
  Hash algo:    sha1
  Hash value:   c29dfa804b0d38b01ff0a28f63b7e56b780bfabf
 Default Configuration: 'config@10'
 Configuration 0 (config@10)
  Description:  OpenWrt
  Kernel:       kernel@1
  FDT:          fdt@1
[ -f /home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/p2w_r619ac-fit-zImage.itb -a -f /home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/root.squashfs ]
sh /home/openwrt/lede/scripts/ubinize-image.sh  --kernel /home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/p2w_r619ac-fit-zImage.itb  /home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/root.squashfs /home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-squashfs-nand-factory.ubi.tmp -p 128KiB -m 2048   
[kernel]
mode=ubi
vol_id=0
vol_type=dynamic
vol_name=kernel
image=/home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/p2w_r619ac-fit-zImage.itb
[rootfs]
mode=ubi
vol_id=1
vol_type=dynamic
vol_name=rootfs
image=/home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/root.squashfs
[rootfs_data]
mode=ubi
vol_id=2
vol_type=dynamic
vol_name=rootfs_data
vol_size=1MiB
vol_flags=autoresize
ubinize: volume size was not specified in section "kernel", assume minimum to fit image "/home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/p2w_r619ac-fit-zImage.itb"2290248 bytes (2.1 MiB)
ubinize: volume size was not specified in section "rootfs", assume minimum to fit image "/home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/root.squashfs"16762259 bytes (15.9 MiB)
cat /home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-squashfs-nand-factory.ubi.tmp >> /home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-squashfs-nand-factory.ubi
rm /home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-squashfs-nand-factory.ubi.tmp
cp /home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-squashfs-nand-factory.ubi /home/openwrt/lede/bin/targets/ipq40xx/generic/openwrt-ipq40xx-p2w_r619ac-squashfs-nand-factory.ubi
GZ_SUFFIX := 
bash: GZ_SUFFIX: command not found
Makefile:349: recipe for target '/home/openwrt/lede/bin/targets/ipq40xx/generic/openwrt-ipq40xx-p2w_r619ac-squashfs-nand-factory.ubi' failed
make[5]: [/home/openwrt/lede/bin/targets/ipq40xx/generic/openwrt-ipq40xx-p2w_r619ac-squashfs-nand-factory.ubi] Error 127 (ignored)
[ -f /home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/p2w_r619ac-fit-zImage.itb -a -f /home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/root.squashfs ]
sh /home/openwrt/lede/scripts/sysupgrade-tar.sh --board p2w_r619ac --kernel /home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/p2w_r619ac-fit-zImage.itb --rootfs /home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/root.squashfs /home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-squashfs-nand-sysupgrade.bin
tar: Option --mtime: Treating date '@1574833880' as 2019-11-27 05:51:20
sysupgrade-p2w_r619ac/
sysupgrade-p2w_r619ac/CONTROL
sysupgrade-p2w_r619ac/kernel
sysupgrade-p2w_r619ac/root
echo '{  "supported_devices":["p2w,r619ac"], "version": { "dist": "OpenWrt", "version": "SNAPSHOT", "revision": "r1723-049a40ee", "board": "ipq40xx" } }' | fwtool -I - /home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-squashfs-nand-sysupgrade.bin
[ ! -s "/home/openwrt/lede/key-build" -o ! -s "/home/openwrt/lede/key-build.ucert" -o ! -s "/home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-squashfs-nand-sysupgrade.bin" ] || { cp "/home/openwrt/lede/key-build.ucert" "/home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-squashfs-nand-sysupgrade.bin.ucert" ; usign -S -m "/home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-squashfs-nand-sysupgrade.bin" -s "/home/openwrt/lede/key-build" -x "/home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-squashfs-nand-sysupgrade.bin.sig" ; ucert -A -c "/home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-squashfs-nand-sysupgrade.bin.ucert" -x "/home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-squashfs-nand-sysupgrade.bin.sig" ; fwtool -S "/home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-squashfs-nand-sysupgrade.bin.ucert" "/home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-squashfs-nand-sysupgrade.bin" ; }
[... 27 lines cut ...]
ubinize: volume size was not specified in section "rootfs", assume minimum to fit image "/home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/root.squashfs"16762259 bytes (15.9 MiB)
cat /home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-squashfs-nand-factory.bin.tmp >> /home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-squashfs-nand-factory.bin
rm /home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-squashfs-nand-factory.bin.tmp
/home/openwrt/lede/scripts/mkits-qsdk-ipq-image.sh /home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-squashfs-nand-factory.bin.its ubi /home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-squashfs-nand-factory.bin
mkimage -f /home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-squashfs-nand-factory.bin.its /home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-squashfs-nand-factory.bin.new
sh: 1: dtc: not found
mkimage: Can't open /home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-squashfs-nand-factory.bin.new.tmp: No such file or directory
Makefile:349: recipe for target '/home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-squashfs-nand-factory.bin' failed
make[5]: *** [/home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/tmp/openwrt-ipq40xx-p2w_r619ac-squashfs-nand-factory.bin] Error 255
make[5]: Leaving directory '/home/openwrt/lede/target/linux/ipq40xx/image'
Makefile:25: recipe for target 'install' failed
make[4]: *** [install] Error 2
make[4]: Leaving directory '/home/openwrt/lede/target/linux/ipq40xx'
Makefile:13: recipe for target 'install' failed
make[3]: *** [install] Error 2
make[3]: Leaving directory '/home/openwrt/lede/target/linux'
time: target/linux/install#346.85#39.20#370.48
target/Makefile:23: recipe for target 'target/linux/install' failed
make[2]: *** [target/linux/install] Error 2
make[2]: Leaving directory '/home/openwrt/lede'
target/Makefile:19: recipe for target '/home/openwrt/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.target_install' failed
make[1]: *** [/home/openwrt/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/openwrt/lede'
/home/openwrt/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

Aes64X, 2019-11-27:
device-tree-compiler 依赖没有安装

axgaoyu, 2019-11-27:
> device-tree-compiler依赖没有安装

好的 我在试试

closed 2019-11-27 by coolsnowwolf