coolsnowwolf/lede #2192: 首页cpuinfo,主机型号不显示

waynesg, 2019-11-27:
已更新最新固件,其他部分均正常,model name无法正常显示.
<a>[image]
[image]

BOBINIUNIU, 2019-11-28:
最新的代码编译后已经正常显示了，编译前最好执行一次 make dirclean

closed 2019-11-28 by coolsnowwolf

waynesg, 2019-11-28:
更新了最新代码,执行了dirclean
model name还是显示?

waynesg, 2019-11-28:
同样的固件在其他机器上又是显示正常的

waynesg, 2019-11-29:
> 最新的代码编译后已经正常显示了，编译前最好执行一次 make dirclean

大概找到了原因,我的路由无法自己创建/tmp/sysinfo目录,所以抓取不到生成的model信息
请问应该如何解决?

Narizgnaw, 2019-12-01:
您好，我在编译的过程中也遇到了同样的问题，请问您最后如何解决的此问题？
我尝试了以下方法：

1.  make clean并pull了最新的源码进行编译，不保留配置刷机。
2.  更换了编译环境重新设置.config进行编译，不保留配置刷机。
3.  使用OpwnWrt CI重新进行编译，不保留配置刷机。

仍然出现此问题。
坐等和我遇到同样问题的大佬。

Narizgnaw, 2019-12-01:
[image]
附截图如上图所示

waynesg, 2019-12-01:
并没有解决。
路由器无法创建/tmp/sysinfo这个目录，所以无法生成目录下的model文件。
我是自己手动创建的。

Narizgnaw, 2019-12-02:
请问一下，如何手动创建这个？只是mkdir我试了好像不行。。