coolsnowwolf/lede #2200: OpenWrt CI 在线集成编译环境使用教程

KFERMercer, 2019-11-28:
# 使用 GitHub Actions 的 OpenWrt 在线集成自动编译环境.

## 自动定制固件, 自动生成配置文件, 无需上传配置, 依赖自动调整.
## 每天凌晨四点自动编译 coolsnowwolf/lede, 兼容 OpenWrt trunk.

[KFERMercer/OpenWrt-CI](https://github.com/KFERMercer/OpenWrt-CI)

感谢[P3TERX](https://github.com/P3TERX/Actions-OpenWrt)珠玉在前.

### 好消息! 好消息!
### 你还在使用别人编译的加料固件吗?!
### 你还在因为学不会Linux命令而苦苦挣扎吗?!
### 你还在用你那双核赛扬装虚拟机吗?!
### 你还在花费半辈子攒下的论坛币只为了下载别人编译好的附件吗?!
### 你还在因为自身原因导致的编译失败而发一些不知所云的issue吗?!
### 生命冇 take two! 我们浪费在编译上的时间已经够多了!
### 21世纪, 自己的才是最好的! 现在人人都可以定制自己的专属固件了! <sub><sub>(不包含三岁及以下儿童)</sub></sub>

---

## 麻瓜级使用教程:

> ### 这个CI脚本和[P3TERX/Actions-OpenWrt](https://github.com/P3TERX/Actions-OpenWrt)的同样是云编译, 有什么不同?

这个 CI 脚本可以帮助你在你的 OpenWrt 分支下构建当前库的Op固件, 无需上传配置文件, 不依赖外挂配置脚本, 所以不会污染库文件树, 对Op开发者比较友好.

P3TERX/Actions-OpenWrt 的优势在于, 其可以独立于文件库存在, 可以快速搭建起可用的编译配置, 但是由于 OpenWrt make configure 自身机制的缘故, 生成的配置文件鲁棒性较差, 需要经常重新生成上传配置, 有形中加重了维护负担, 使得使用起来不是很`优雅`.

> ### 这个CI脚本适合哪些人?

- 伸手党 (访问者可以每日下载到当日最新的完全基于大雕代码的原味固件)
- 正在维护自己的 OpenWrt 分支的大佬/初学者/玩家 (可以快速测试自己的代码)
- 没有精力维持一个专用编译机的佛系人士 (梯子到期了?)

> ### 开始吧.

在一切开始前, 你需要的是:

- GitHub 账号
- 申请使用 GitHub Actions
- 自己的OpenWrt分支(fork大雕源或者官方源)
- 识字能力以及基本电脑操作技能
- 脑子

你不需要的是:

- Linux技能
- 对编译失败的处置能力 (脚本由我维护)
- 梯子

### 1. 注册GitHub账号并开启GitHub Actions (百度).

### 2. fork [coolsnowwolf/lede](https://github.com/coolsnowwolf/lede) 或者其他OpenWrt源.

### 3. (如果你fork的是大雕源,请忽略此步) 点击页面上方的`Create new file`按钮, 打开后在文件名处填入`.github/workflows/openwrt-ci.yml`(github 会自动创建路径).

### 4. (如果你fork的是大雕源,请忽略此步) 打开[这个链接](https://raw.githubusercontent.com/KFERMercer/OpenWrt-CI/master/openwrt-ci.yml), `Ctrl+a`选泽复制页面所有内容黏贴在刚才的下方的大文本框内.

> ### 现在暂停一下, 代码里需要留意一些东西...

如果你只希望生成默认配置, 不经过任何修改的固件, 那你不需要对代码进行任何修改. 自动集成工具会在每次推送后自动编译出最新默认配置的固件.

如果你希望定制你的固件:\
代码里的注释部分详细介绍了如何在脚本中客制化你的固件. 简单来说, 你只需要解除注释相应行即可.

现在我们继续:

### 5. 假设你已经完成了脚本修改, 现在点击页面最下方的绿色按钮. 你是色盲也没关系, 因为按钮上写着`Commit new file`.

### 6. 大功告成, 集成编译环境会自动开始编译. 现在返回你的库首页, 点击页面上方的`Actions`按钮就可以查看进度.

> ### 如何下载到编译完成的固件?

进入`Actions`标签页后, 如果相应的集成活动全部完成 (打勾) , 点击页面右上方的`Artifacts`即可看到你的固件 (通常是一个压缩包). 点击压缩包即可开始下载. (别人也可以下载你的固件, 是不是很神奇?)

pingod, 2019-11-29:
我的workflow一直显示 (Unnamed workflow),不知道为啥,GitHub官方问题??

KFERMercer, 2019-11-29:
> 我的workflow一直显示 (Unnamed workflow),不知道为啥,GitHub官方问题??

你改了什么地方?

pingod, 2019-11-29:
添加了一个workflow示例模板文件,我把这个文件添加或者删除都不行,都是显示Unnamed workflow

d975xbx, 2019-11-29:
感谢大神，可以删除虚拟机了。

lyafancn, 2019-11-29:
大佬，我不用X64，你的实例脚本里的x64怎么替换别的CPU(就是给个其他的CPU设备的TARGET_参数列表吧。)不知道哪里有TARGTE_的不同参数。

lyafancn, 2019-11-29:
[image]
我想要修改x64为A53处理器的架构，但是这里貌似没有arm v8

redchenjs, 2019-11-29:
armv8就是aarch64

KFERMercer, 2019-11-29:
> [image]
> 我想要修改x64为A53处理器的架构，但是这里貌似没有arm v8

建议直接复制config.seed或者.config最开始的target部分

lyafancn, 2019-11-29:
> > [image]
> > 我想要修改x64为A53处理器的架构，但是这里貌似没有arm v8
> 
> 建议直接复制config.seed或者.config最开始的target部分

我强烈建议您再写一个文档，列出不同设备的这个部分：
[image]

lyafancn, 2019-11-29:
[image]
只需要这三项对吗？其他的像这样的要吗？
CONFIG_TARGET_BOARD="sunxi"
CONFIG_TARGET_SUBTARGET="cortexa53"
CONFIG_TARGET_PROFILE="DEVICE_sun50i-h5-nanopi-neo2"
CONFIG_TARGET_ARCH_PACKAGES="aarch64_cortex-a53"

lyafancn, 2019-11-29:
CONFIG_TARGET_PREINIT_IFNAME=""
CONFIG_TARGET_PREINIT_IP="192.168.1.1"
CONFIG_TARGET_PREINIT_NETMASK="255.255.255.0"
CONFIG_TARGET_PREINIT_BROADCAST="192.168.1.255"
CONFIG_TARGET_INIT_PATH="/usr/sbin:/usr/bin:/sbin:/bin"
CONFIG_TARGET_INIT_ENV=""
CONFIG_TARGET_INIT_CMD="/sbin/init"
CONFIG_TARGET_INIT_SUPPRESS_STDERR=y
还有这些要吗？

KFERMercer, 2019-11-29:
> [image]
> 只需要这三项对吗？其他的像这样的要吗？
> CONFIG_TARGET_BOARD="sunxi"
> CONFIG_TARGET_SUBTARGET="cortexa53"
> CONFIG_TARGET_PROFILE="DEVICE_sun50i-h5-nanopi-neo2"
> CONFIG_TARGET_ARCH_PACKAGES="aarch64_cortex-a53"

比如要编译ACRH 17的固件, 只需要这两行:
```
CONFIG_TARGET_ipq40xx=y
CONFIG_TARGET_ipq40xx_DEVICE_asus_rt-acrh17=y
```

lyafancn, 2019-11-29:
> > CONFIG_TARGET_PREINIT_IFNAME=""
> > CONFIG_TARGET_PREINIT_IP="192.168.1.1"
> > CONFIG_TARGET_PREINIT_NETMASK="255.255.255.0"
> > CONFIG_TARGET_PREINIT_BROADCAST="192.168.1.255"
> > CONFIG_TARGET_INIT_PATH="/usr/sbin:/usr/bin:/sbin:/bin"
> > CONFIG_TARGET_INIT_ENV=""
> > CONFIG_TARGET_INIT_CMD="/sbin/init"
> > CONFIG_TARGET_INIT_SUPPRESS_STDERR=y
> > 还有这些要吗？
> 
> 不需要.
> 
> > [image]
> > 只需要这三项对吗？其他的像这样的要吗？
> > CONFIG_TARGET_BOARD="sunxi"
> > CONFIG_TARGET_SUBTARGET="cortexa53"
> > CONFIG_TARGET_PROFILE="DEVICE_sun50i-h5-nanopi-neo2"
> > CONFIG_TARGET_ARCH_PACKAGES="aarch64_cortex-a53"
> 
> 对. 通常来说你在menuconfig下需要做出多少次选泽就需要多少行, 所以nanopi 2只需要这三行就够了:
> 
> ```
> CONFIG_TARGET_sunxi_cortexa53_DEVICE_sun50i-h5-nanopi-neo2=y
> CONFIG_TARGET_sunxi_cortexa53=y
> CONFIG_TARGET_sunxi=y
> ```
> 
> 比如要编译ACRH 17的固件, 只需要这两行:
> 
> ```
> CONFIG_TARGET_ipq40xx=y
> CONFIG_TARGET_ipq40xx_DEVICE_asus_rt-acrh17=y
> ```

我觉得这些都背不下来，所有设备的。。。您看看再写个这块的所有设备参数文档？嘻嘻嘻嘻

KFERMercer, 2019-11-29:
> 我觉得这些都背不下来，所有设备的。。。您看看再写个这块的所有设备参数文档？嘻嘻嘻嘻
我也没背, 需要什么设备只需要去make一个配置文件看一下就好了啊, 反正这种事是一劳永逸的.

lyafancn, 2019-11-29:
> > 我觉得这些都背不下来，所有设备的。。。您看看再写个这块的所有设备参数文档？嘻嘻嘻嘻
> 
> 我也没背, 需要什么设备只需要去make一个配置文件看一下就好了啊, 反正这种事是一劳永逸的.
> 话说回来, 配置选项的命名仔细看一下你就会发现规律.

嗯嗯。
那我再问下几个问题：
1.我配置的自定义的Neo2固件编译，那么可以同时再添加一个编译其他的设备固件的workflow脚本吗？（可以的话，是怎么添加呢？在workflow目录下新建一个其他的比如op-acrh17.yml就可以了？）
2.FORK一次代码后，任意一个workflow脚本都可以自动同步大雕的commit并且开始按照脚本编译吗？不需要重新FORK吧?.....我不太会用git，不知道怎么同步大雕的源码且不覆盖我的自定义的workflow的yml。

pingod, 2019-11-29:
现在可以了,是GitHub官方的锅
https://www.githubstatus.com/incidents/39ynqr1xb3s0

KFERMercer, 2019-11-29:
> 嗯嗯。
> 那我再问下几个问题：
> 1.我配置的自定义的Neo2固件编译，那么可以同时再添加一个编译其他的设备固件的workflow脚本吗？（可以的话，是怎么添加呢？在workflow目录下新建一个其他的比如op-acrh17.yml就可以了？） [...]

① 对
② 不能, 需要手动合并. 这个脚本专注当前库, 如果想要代码, 可以再脚本中添加一个拉取代码的命令, 但是我不建议你这么做. 同步上游不会覆盖下游更改.

> 现在可以了,是GitHub官方的锅
> https://www.githubstatus.com/incidents/39ynqr1xb3s0

github的bug都让你找着了😂

z826540272, 2019-11-29:
> > > > 我觉得这些都背不下来，所有设备的。。。您看看再写个这块的所有设备参数文档？嘻嘻嘻嘻
> > > 
> > > 
> > > 我也没背, 需要什么设备只需要去make一个配置文件看一下就好了啊, 反正这种事是一劳永逸的.
> > > 话说回来, 配置选项的命名仔细看一下你就会发现规律.
> > 
> > 
> > 嗯嗯。
> > 那我再问下几个问题：
> > 1.我配置的自定义的Neo2固件编译，那么可以同时再添加一个编译其他的设备固件的workflow脚本吗？（可以的话，是怎么添加呢？在workflow目录下新建一个其他的比如op-acrh17.yml就可以了？）
> > 2.FORK一次代码后，任意一个workflow脚本都可以自动同步大雕的commit并且开始按照脚本编译吗？不需要重新FORK吧?.....我不太会用git，不知道怎么同步大雕的源码且不覆盖我的自定义的workflow的yml。
> 
> ① 对
> ② 不能, 需要手动合并. 这个脚本专注当前库, 如果想要代码, 可以再脚本中添加一个拉取代码的命令, 但是我不建议你这么做. 同步上游不会覆盖下游更改.
> 
> 如果你只是想编译最新固件, 可以用[P3TERX/Actions-OpenWrt](https://github.com/P3TERX/Actions-OpenWrt)
> 
> > 现在可以了,是GitHub官方的锅
> > https://www.githubstatus.com/incidents/39ynqr1xb3s0
> 
> github的bug都让你找着了😂

4点同时编译,GitHub算力被撸完

d975xbx, 2019-11-30:
如果大雕源码更新，要怎么自动编译？

BROBIRD, 2019-11-30:
每个项目每个月只有2000分钟的免费时间，如果每天都运行一次的话，是不够用的。

KFERMercer, 2019-11-30:
> 每个项目每个月只有2000分钟的免费时间，如果每天都运行一次的话，是不够用的。

请问可否提供一下信息来源?
我查到的文档里是这样说明的, 没有看到每月使用时间相关的限制:
[image]

BROBIRD, 2019-11-30:
@KFERMercer 抱歉看错了，上述限制是针对私有仓库的

d975xbx, 2019-12-01:
大神，怎么在CI脚本中添加pull上游代码的动作？

z826540272, 2019-12-01:
> 大神，怎么在CI脚本中添加pull上游代码的动作？

你用另一个脚本就是最新版编译

d975xbx, 2019-12-01:
这个吗？https://github.com/P3TERX/Actions-OpenWrt

xunqiuhuiyi, 2019-12-01:
谢谢UP主的分享 想请教下 如果我拥有多台路由 是否可以同时自动化编译多个固件呢？还是说需要建立多个自动编译化脚本？如果我修改了openwrt-ci.yml 在Action建立多个脚本 更新以后 他是按照我最近修改还是？

KFERMercer, 2019-12-01:
> 谢谢UP主的分享 想请教下 如果我拥有多台路由 是否可以同时自动化编译多个固件呢？还是说需要建立多个自动编译化脚本？如果我修改了openwrt-ci.yml 在Action建立多个脚本 更新以后 他是按照我最近修改还是？主要是我不太会使用git 如果我workflow目录下创建多个.yml脚本，是不是workflow脚本都可以自动同步大雕的commit并且开始按照脚本编译呢？

同时建立多个flow, 每个flow都会执行. 不过既然你不太会用git, 那这种需求建议使用[P3TERX/Actions-OpenWrt](https://github.com/P3TERX/Actions-OpenWrt).

adaaaaaa, 2019-12-02:
老哥，不行啊。。。你最近一次改代码之后，GG了。。。

` Uploading artifact 'OpenWrt firmware' from '/home/runner/work/lede/lede/artifact/' for run #21
Uploading 0 files
File upload complete.
Uploaded '0' bytes from '/home/runner/work/lede/lede/artifact/' to server`

固件大小直接为0了。。。

KFERMercer, 2019-12-02:
> 老哥，不行啊。。。你最近一次改代码之后，GG了。。。
> 
> ` Uploading artifact 'OpenWrt firmware' from '/home/runner/work/lede/lede/artifact/' for run #21 Uploading 0 files File upload complete. Uploaded '0' bytes from '/home/runner/work/lede/lede/artifact/' to server`
> 
> 固件大小直接为0了。。。

刚才又更新了,, 之前一条命令写错了

adaaaaaa, 2019-12-02:
不行啊，你是说前6，7小时那个改动吗？。。。还是这样啊。。。😂😂😂
你那边可以了？？

KFERMercer, 2019-12-02:
> 不行啊，你是说前6，7小时那个改动吗？。。。还是这样啊。。。😂😂😂
> 你那边可以了？？

刚刚更新

adaaaaaa, 2019-12-02:
好的，我又去抄作业了。。。😂😂😂

adaaaaaa, 2019-12-02:
你确定这次没问题了吧？？😂😂😂

KFERMercer, 2019-12-02:
> 你确定这次没问题了吧？？😂😂😂

刚看了下你的CI, 写的非常山炮. 建议去看小白教程: https://github.com/coolsnowwolf/lede/issues/2225

adaaaaaa, 2019-12-02:
结合你们两家所长啊。。。怎么就山炮了呢？？😂😂😂

KFERMercer, 2019-12-02:
> 结合你们两家所长啊。。。怎么就山炮了呢？？😂😂😂

基本语法都错了

adaaaaaa, 2019-12-02:
最初的模板就是你的呢。。。整个编译过程也没有报错啊？？🙃

adaaaaaa, 2019-12-02:
给个栗子我看看哪里不对了？？

adaaaaaa, 2019-12-02:
除了中间加一点插件，软件，驱动。。。其他基本完全照抄啊。。。😂😂😂

KFERMercer, 2019-12-02:
> 最初的模板就是你的呢。。。整个编译过程也没有报错啊？？🙃

>给个栗子我看看哪里不对了？？

>除了中间加一点插件，软件，驱动。。。其他基本完全照抄啊。。。😂😂😂

我怀疑你这个编出来的配置根本就不会生效, 而且有多个包之间互相冲突. 如果你不需要某项配置, 除了你想强制关闭应该在后面添加` is not set`外, 其余的应该就直接删掉对应行而不是注释.

adaaaaaa, 2019-12-02:
收到，谢谢，我马上去改。。。至少也是简介好看一点。。。😂😂😂

adaaaaaa, 2019-12-02:
对了，我设置是计划9点UTC，每星期五执行的，为什么没有生效，而是立刻就开始编译了呢？？

adaaaaaa, 2019-12-02:
老哥，好像还是不太行哦。。。编译出的固件大小有问题。。。之前都是190多M的，现在只剩90多了。。。无论是按照你后来让我改的，还是之前照你修改的配置文件改的。。。都是一样大小。。。
然后，压缩包解压出错。。。

KFERMercer, 2019-12-02:
> 老哥，好像还是不太行哦。。。编译出的固件大小有问题。。。之前都是190多M的，现在只剩90多了。。。无论是按照你后来让我改的，还是之前照你修改的配置文件改的。。。都是一样大小。。。
> 然后，压缩包解压出错。。。

[image]

adaaaaaa, 2019-12-03:
cat >> .config <<EOF
          CONFIG_TARGET_IMAGES_PAD=y
          CONFIG_VDI_IMAGES=y
          EOF

这里有什么问题不？为什么不能编译出VDI镜像？

powskey, 2019-12-04:
# 编译ipq40XX固件:

          # cat >> .config <<EOF

          # CONFIG_TARGET_ipq40xx=y

          # CONFIG_TARGET_ipq40xx_DEVICE_p2w_r619ac-128m=y

          # EOF
我这样写的，为什么编译出来的不是竞斗云的固件？是不是哪里错了？

redchenjs, 2019-12-04:
就因为这个CI好多萌新乱开PR都开到官方OpenWrt上去了

z826540272, 2019-12-04:
> 就因为这个CI好多萌新乱开PR都开到官方OpenWrt上去了

说实话这ci难用多此一举，不如原来的那个

KFERMercer, 2019-12-04:
> 就因为这个CI好多萌新乱开PR都开到官方OpenWrt上去了

今天新加了自动更新CI, 不再需要手动pr了.

adaaaaaa, 2019-12-05:
能不能把编译时候实际用到的config文件也导出来，一起打包下载了？？
这次看到又添加了一些新的驱动，这些是我不需要的。。。
我想要下次编译的时候，禁止这些新的驱动添加到固件里面。。。
但是，我又不想在本地启动虚拟机来git pull。。。我都打算删掉虚拟机了。。。

kang-mk, 2019-12-05:
牛逼感谢

peter-tank, 2019-12-05:
照着撸了个最近的官包，打128M 大内存补丁后的 WNDR4300 和 WNDR3700 好像哪里不对，2小时后出来 firmware 600M，熟悉的[指点指点](https://github.com/peter-tank/openwrt/blob/v19.07.0-rc2_netgear_wndr/.github/workflows/main.yml)

KFERMercer, 2019-12-05:
> 照着撸了个最近的官包，打128M 大内存补丁后的 WNDR4300 和 WNDR3700 好像哪里不对，2小时后出来 firmware 600M，熟悉的[指点指点](https://github.com/peter-tank/openwrt/blob/v19.07.0-rc2_netgear_wndr/.github/workflows/main.yml)

槽点太多以至于不知道从哪里开始吐. 

简单来说就是逻辑混乱, 拉取了错误的库到错误的位置, 配置文件根本没有, 加上你的CI和你的库不兼容. 但是神奇的是, 最后的编译失败并不是CI脚本造成的. 

玩转高阶操作之前, 希望能做好前期知识积累.

> 能不能把编译时候实际用到的完整版的config文件也导出来，一起打包下载了？？
> 这次看到又添加了一些新的驱动，这些是我不需要的。。。
> 我想要下次编译的时候，禁止这些新的驱动添加到固件里面。。。
> 但是，我又不想在本地启动虚拟机来git pull。。。我都打算删掉虚拟机了。。。

这个很容易就可以做到. 
说句题外话, 有件事可能需要告诉你一下, 我会更新在最近你提的那个issue里, 希望你这段时间能保持留意.

bonnyandsky, 2019-12-06:
你好，请问怎么把你的merge-upstream.yml文件放到我的目录下？我看好像没办法下载，

peter-tank, 2019-12-07:
@KFERMercer 怎么从trunk抓指定tag的版本 ？
我从 github 和 git.openwrt.org clone 或wget *.tar.gz 后 git checkout 各种玩法，死活都是内核不对版。。。只有 master 才是最正确的

KFERMercer, 2019-12-07:
> @KFERMercer 怎么从trunk抓指定tag的版本 ？
> 我从 github 和 git.openwrt.org clone 或wget *.tar.gz 后 git checkout 各种玩法，死活都是内核不对版。。。只有 master 才是最正确的

由于没看到错误信息所以猜不出来

sunlanhao, 2019-12-08:
你好，请问我想通过云编译加入魔改版的ssr，这样添加代码可以吗？ 
[image]
[image]

KFERMercer, 2019-12-08:
> 你好，请问我想通过云编译加入魔改版的ssr，这样添加代码可以吗？
> [image]
> [image]

不知道, 没用过这个

i6o6i, 2019-12-09:
Nano pi R1s的CPU的配置在哪？

bonnyandsky, 2019-12-09:
我做了一个分支，名字叫ledenew，我打算以后同步L大的代码都用分支来同步，请问openwrt-ci.yml和merge-upstream.yml需要修改对应那些字段

[23 more replies]

closed 2019-12-24 by coolsnowwolf