coolsnowwolf/lede #2206: 使用coolsnowwolf/lede最新代码编译OpenWrt-mt7621-mr3g刷入后：内存提示 0.00 mb

dingpengyu, 2019-11-29:
机器型号：
小米3g
处理器：
联发科 mt7621

dingpengyu, 2019-11-29:
经过测试 使用PandoraBox-ralink-mt7621-xiaomi-r3g-2019-02-01-git-0231ad4b5-squashfs-sysupgrade.bin刷入是正常的，内存还有89mb

hanandjun, 2019-12-08:
nand 的话缺一个补丁，加上就好了，可以看我的fork

dingpengyu, 2019-12-08:
> nand 的话缺一个补丁，加上就好了，可以看我的fork
能发个链接吗？

dingpengyu, 2019-12-08:
我把0038-mtd-ralink-add-mt7620-nand-driver.patch复制到了我代码的相同目录下，做了编译，编译完刷到了3g做了测试，软件包可用内存依然提示0.00mb可用。

hanandjun, 2019-12-08:
> 我把0038-mtd-ralink-add-mt7620-nand-driver.patch复制到了我代码的相同目录下，做了编译，编译完刷到了3g做了测试，软件包可用内存依然提示0.00mb可用。

你直接cherry-pick 我那条提交就可以了

pepsigold, 2019-12-13:
@hanandjun 我也是昨天编译K2刷入后空闲空间: 0% (0.00 B) ，但是能正常使用，今天重新拉源码再编译一次试一试。

dingpengyu, 2019-12-13:
> @hanandjun 我也是昨天编译K2刷入后空闲空间: 0% (0.00 B) ，但是能正常使用，今天重新拉源码再编译一次试一试。

你可以试试看，配置完重启路由器配置会不会丢。

pepsigold, 2019-12-13:
@dingpengyu 还不清楚，我现在还在下载编译环境，我把原来的文件夹全部del了

pepsigold, 2019-12-13:
系统
主机名	OpenWrt
主机型号	Phicomm PSG1218 rev.A
架构	MediaTek MT7620A ver:2 eco:6 x 1
固件版本	OpenWrt R9.12.03 / LuCI Master (git-19.338.43082-9e87e69)
内核版本	4.14.158
本地时间	Fri Dec 13 17:01:16 2019
运行时间	0h 12m 5s
平均负载	0.20, 0.33, 0.28
~~~~~~~~重新编译后恢复正常！
软件列表 空闲空间: 89% (1.04 MB)

dingpengyu, 2019-12-13:
> 系统
> 主机名 OpenWrt
> 主机型号 Phicomm PSG1218 rev.A
> 架构 MediaTek MT7620A ver:2 eco:6 x 1
> 固件版本 OpenWrt R9.12.03 / LuCI Master (git-19.338.43082-9e87e69)
> 内核版本 4.14.158
> 本地时间 Fri Dec 13 17:01:16 2019
> 运行时间 0h 12m 5s
> 平均负载 0.20, 0.33, 0.28
> 
> ```
> 软件列表 空闲空间: 89% (1.04 MB) 
> ```

看来问题是修复了，等等去编一版，刷进去看看。

closed 2019-12-15 by coolsnowwolf

designer-fanfan, 2021-01-09:
MT1300使用coolsnowwolf/lede最新代码编译OpenWrt-mt7621刷入后：内存提示 0.00 mb 。不知道怎么解决，求助

mjysci, 2021-02-12:
r3g的kernel0是专门用来恢复出厂设置的，用breed刷[需要一些额外步骤](https://www.right.com.cn/forum/forum.php?mod=viewthread&tid=267455&page=1#pid2314800)。具体如下：  
1. Breed的Xiaomi R3G设置中清空`normal_firmware_md5`的内容（不要删除，保留该键）。  
2. Breed的环境变量中增加`xiaomi.r3g.bootfw`，值为`2`。
3. 固件用`initramfs-kernel.bin`这个，闪存布局选`2`。
4. 升级启动后空闲空间为0，且重启无法保存配置。此时去`系统——备份/升级`处升级固件，选`sysupgrade.bin`这个。升级重启后空闲空间会正常显示。