coolsnowwolf/lede #2216: daemon.err uhttpd[1663]: cat: can't open '/etc/bench.log': No such file or directory

SouHong, 2019-11-30:
daemon.err uhttpd[1663]: cat: can't open '/etc/bench.log': No such file or directory
请问日志文件中出现这个错误是个例吗？

dolphinpaopao, 2019-11-30:
偶然发现最新的版本还有类似的这个报错log
daemon.err uhttpd[2966]: Cannot get device settings: No such device

EnnawYang, 2019-11-30:
> 偶然发现最新的版本还有类似的这个报错log
> daemon.err uhttpd[2966]: Cannot get device settings: No such device

这个不影响使用 我也有

dolphinpaopao, 2019-12-04:
> > 偶然发现最新的版本还有类似的这个报错log
> > daemon.err uhttpd[2966]: Cannot get device settings: No such device
> 
> 这个不影响使用 我也有

观察了两天，也尝试编译了最新版本，不管是否勾选uhttpd，都会有这个错误，而且一直出现一大堆，没有什么头绪。

closed 2019-12-07 by coolsnowwolf

norikohxb, 2020-04-18:
> daemon.err uhttpd[1663]: cat: can't open '/etc/bench.log': No such file or directory
> 请问日志文件中出现这个错误是个例吗？

Sat Apr 18 10:35:29 2020 daemon.err uhttpd[2090]: cat: can't open '/etc/bench.log': No such file or directory
Sat Apr 18 10:48:48 2020 daemon.err uhttpd[2090]: cat: can't open '/etc/bench.log': No such file or directory
Sat Apr 18 10:48:48 2020 daemon.err uhttpd[2090]: cat: can't open '/etc/bench.log': No such file or directory
Sat Apr 18 11:37:38 2020 daemon.err uhttpd[2090]: cat: can't open '/etc/bench.log': No such file or directory
Sat Apr 18 11:37:38 2020 daemon.err uhttpd[2090]: cat: can't open '/etc/bench.log': No such file or directory

请问楼上的后来怎么解决这个问题

norikohxb, 2020-04-18:
> > 偶然发现最新的版本还有类似的这个报错log
> > daemon.err uhttpd[2966]: Cannot get device settings: No such device
> 
> 这个不影响使用 我也有

Sat Apr 18 10:35:29 2020 daemon.err uhttpd[2090]: cat: can't open '/etc/bench.log': No such file or directory
Sat Apr 18 10:48:48 2020 daemon.err uhttpd[2090]: cat: can't open '/etc/bench.log': No such file or directory
Sat Apr 18 10:48:48 2020 daemon.err uhttpd[2090]: cat: can't open '/etc/bench.log': No such file or directory
Sat Apr 18 11:37:38 2020 daemon.err uhttpd[2090]: cat: can't open '/etc/bench.log': No such file or directory
Sat Apr 18 11:37:38 2020 daemon.err uhttpd[2090]: cat: can't open '/etc/bench.log': No such file or directory

请问楼上的后来怎么解决这个问题

randyho, 2021-09-19:
感觉是coremark的问题，我不想要这个插件，选上后就不会要问题

sayidly, 2021-11-29:
```
cd /etc
touch bench.log
coremark >> bench.log
```
should work