coolsnowwolf/lede #2219: transmission持续崩溃。

once375ml, 2019-11-30:
Sat Nov 30 12:36:53 2019 user.notice transmission: Starting with 126464000 virt mem
Sat Nov 30 12:37:42 2019 kern.info kernel: [ 1074.457685] transmission-da[10827]: segfault (11) at 0 nip 10028320 lr 100282b0 code 1 in transmission-daemon[10000000+58000]
Sat Nov 30 12:37:42 2019 kern.info kernel: [ 1074.469049] transmission-da[10827]: code: 83e1002c 38210030 480093d8 80ff0070 1d09000c 7d474214 814a0004 7f8ad800
Sat Nov 30 12:37:42 2019 kern.info kernel: [ 1074.480228] transmission-da[10827]: code: 40be0014 7d07402e 57ca103a 3bde0001 <7d1c512e> 39290001 4bffff90 57a9103a
Sat Nov 30 12:38:31 2019 kern.info kernel: [ 1123.762628] transmission-da[10983]: segfault (11) at 0 nip b7e11078 lr b7d16964 code 1 in libc.so[b7daf000+85000]
Sat Nov 30 12:38:31 2019 kern.info kernel: [ 1123.772947] transmission-da[10983]: code: 7d2a4b78 54e70036 54e8e13e 38e70010 39080001 7d0903a6 81640000 394a0010
Sat Nov 30 12:38:31 2019 kern.info kernel: [ 1123.784700] transmission-da[10983]: code: 80c40004 38840010 8004fff8 8104fffc <916afff0> 90cafff4 900afff8 910afffc
Sat Nov 30 12:39:12 2019 kern.info kernel: [ 1164.390897] transmission-da[11089]: segfault (11) at 0 nip 10028320 lr 100282b0 code 1 in transmission-daemon[10000000+58000]
Sat Nov 30 12:39:12 2019 kern.info kernel: [ 1164.402302] transmission-da[11089]: code: 83e1002c 38210030 480093d8 80ff0070 1d09000c 7d474214 814a0004 7f8ad800
Sat Nov 30 12:39:12 2019 kern.info kernel: [ 1164.414498] transmission-da[11089]: code: 40be0014 7d07402e 57ca103a 3bde0001 <7d1c512e> 39290001 4bffff90 57a9103a
Sat Nov 30 12:40:23 2019 kern.info kernel: [ 1235.358033] transmission-da[11186]: segfault (11) at 0 nip b7bf402c lr b7af9964 code 1 in libc.so[b7b92000+85000]
Sat Nov 30 12:40:23 2019 kern.info kernel: [ 1235.368426] transmission-da[11186]: code: 418204a0 2f850000 419e0104 7c691b78 7ca903a6 48000008 424000f4 38840001
Sat Nov 30 12:40:23 2019 kern.info kernel: [ 1235.380092] transmission-da[11186]: code: 8944ffff 70880003 39290001 38a5ffff <9949ffff> 4082ffe4 71280003 408200e4
Sat Nov 30 12:41:24 2019 kern.info kernel: [ 1297.034616] transmission-da[11355]: segfault (11) at 0 nip b78ce02c lr b77d3964 code 1 in libc.so[b786c000+85000]
Sat Nov 30 12:41:24 2019 kern.info kernel: [ 1297.044955] transmission-da[11355]: code: 418204a0 2f850000 419e0104 7c691b78 7ca903a6 48000008 424000f4 38840001
Sat Nov 30 12:41:24 2019 kern.info kernel: [ 1297.055811] transmission-da[11355]: code: 8944ffff 70880003 39290001 38a5ffff <9949ffff> 4082ffe4 71280003 408200e4
Sat Nov 30 12:42:20 2019 kern.info kernel: [ 1352.672276] transmission-da[12109]: segfault (11) at 200 nip 1003e248 lr 1003e234 code 1 in transmission-daemon[10000000+58000]
Sat Nov 30 12:42:20 2019 kern.info kernel: [ 1352.683843] transmission-da[12109]: code: 80630000 2f840020 409c0008 38800020 909f0008 5484103a 4bff7b6d 907f0000
Sat Nov 30 12:42:20 2019 kern.info kernel: [ 1352.695838] transmission-da[12109]: code: 807f0004 815f0000 80010024 5469103a <7fca492e> 39230001 913f0004 7c0803a6
Sat Nov 30 12:42:20 2019 daemon.info procd: Instance transmission::instance1 s in a crash loop 6 crashes, 50 seconds since last crash

WNDR4700 挂一块2T。 现在qbittorent可以，但是开销太大，一个种子 10MB/s CPU 90%

transmission一直不行 前前后后git 好几次编译 都不行

HiGarfield, 2019-11-30:
transmission 要内存够大。
如果你的路由器不到512MB的内存，请编辑/etc/init.d/transmission
option mem_percentage '50' 这行 改成 option mem_percentage '100' 。但内存被transmission撑爆了路由器重启别找我。

coolsnowwolf, 2019-12-02:
内存不足导致分配失败

once375ml, 2019-12-02:
> 内存不足导致分配失败

设定的64M缓存是写缓存吧；设备为N手WNDR 4700  256M内存，系统还空闲128M

我编译了一个TR版 一个QB版 晚上回去再试试调整内存

once375ml, 2019-12-02:
Sun Dec  1 23:28:25 2019 kern.notice kernel: [    2.016602] scsi 0:0:0:0: Direct-Access     ATA      **WDC WD20EZRZ-00Z 0A80** PQ: 0 ANSI: 5
Sun Dec  1 23:28:25 2019 kern.notice kernel: [    2.031325] sd 0:0:0:0: [sda] 3907029168 512-byte logical blocks: (2.00 TB/1.82 TiB)
Sun Dec  1 23:28:25 2019 kern.notice kernel: [    2.039059] sd 0:0:0:0: [sda] 4096-byte physical blocks
Sun Dec  1 23:28:25 2019 kern.notice kernel: [    2.056306] sd 0:0:0:0: [sda] Write Protect is off
Sun Dec  1 23:28:25 2019 kern.debug kernel: [    2.061111] sd 0:0:0:0: [sda] Mode Sense: 00 3a 00 00
Sun Dec  1 23:28:25 2019 kern.notice kernel: [    2.065369] sd 0:0:0:0: [sda] **Write cache: enabled, read cache: enabled, **doesn't support DPO or FUA

查到了这块硬盘缓存是64M的，且开启了。

所以，关闭软件缓存，设定mem percentage 80%  200MB内存看看能不能跑起来了

closed 2019-12-04 by coolsnowwolf