coolsnowwolf/lede #2220: qt5又编译失败了，新版的编译有问题

feifan1113, 2019-11-30:


feifan1113, 2019-11-30:
[image]

Her0R0cket, 2019-11-30:
../../../../staging_dir/target-x86_64_musl/usr/include/QtCore/qprocess.h:49:1: error: 'QT_FEATURE_processenvironment' was not declared in this scope
 QT_REQUIRE_CONFIG(processenvironment);
 ^
../../../../staging_dir/target-x86_64_musl/usr/include/QtCore/qprocess.h:49:1: note: suggested alternative: 'QT_FEATURE_process'
In file included from ../../../../staging_dir/target-x86_64_musl/usr/include/QtNetwork/QNetworkAccessManager:1:0,
                 from ./base/net/downloadmanager.h:34,
                 from app/application.cpp:68:
../../../../staging_dir/target-x86_64_musl/usr/include/QtNetwork/qnetworkaccessmanager.h:161:45: error: 'QNetworkRequest::RedirectPolicy' has not been declared
     void setRedirectPolicy(QNetworkRequest::RedirectPolicy policy);
                                             ^~~~~~~~~~~~~~
../../../../staging_dir/target-x86_64_musl/usr/include/QtNetwork/qnetworkaccessmanager.h:162:22: error: 'RedirectPolicy' in 'class QNetworkRequest' does not name a type
     QNetworkRequest::RedirectPolicy redirectPolicy() const;
                      ^~~~~~~~~~~~~~
Makefile:2220: recipe for target 'application.o' failed

完整的是这样的奇怪的是6dcd971 升级4.1.9.1还编译成功,然后73b0fb4降级4.1.9还失败了

Her0R0cket, 2019-11-30:
应该是支持mips后挂了,早上已经clean了,等等改下makefile到前一版试试

Aes64X, 2019-11-30:
昨晚编译时 qt5 是正常的，请尝试重新拉取源码，记得保留dl文件，有几个小问题是 qt5 相关的几个依赖下载校验码有误，改skip后正常编译。

Her0R0cket, 2019-11-30:
qt5正常,这是qbittorrent的

feifan1113, 2019-11-30:
> 应该是支持mips后挂了,早上已经clean了,等等改下makefile到前一版试试

不搞了 试了编译两次都不行

Aes64X, 2019-11-30:
make download V=s 一遍，看看有没有回复校验码错误的，我记得有2个，估计是校验错误就没下载。

feifan1113, 2019-11-30:
> 校验错误

如果找到是检验码错误，那要怎么改才能编译通过？

Aes64X, 2019-11-30:
> > 校验错误
> 
> 如果找到是检验码错误，那要怎么改才能编译通过？

搜索出错的文件，定位到目录，校验值 =skip  就行了
估计是大雕改了版本号忘记了校验值

feifan1113, 2019-11-30:
> > > 校验错误
> > 
> > 
> > 如果找到是检验码错误，那要怎么改才能编译通过？
> 
> 搜索出错的文件，定位到目录，校验值 =skip 就行了
> 估计是大雕改了版本号忘记了校验值

[image]
这样算不算报错

feifan1113, 2019-11-30:
> > > > 校验错误
> > > 
> > > 
> > > 如果找到是检验码错误，那要怎么改才能编译通过？
> > 
> > 
> > 搜索出错的文件，定位到目录，校验值 =skip 就行了
> > 估计是大雕改了版本号忘记了校验值
> 
> [image]
> 这样算不算报错

[image]

feifan1113, 2019-11-30:
> > > > > 校验错误
> > > > 
> > > > 
> > > > 如果找到是检验码错误，那要怎么改才能编译通过？
> > > 
> > > 
> > > 搜索出错的文件，定位到目录，校验值 =skip 就行了
> > > 估计是大雕改了版本号忘记了校验值
> > 
> > 
> > [image]
> > 这样算不算报错
> 
> [image]

仔细看了一下 貌似没报错，算了不搞了 ，等大雕修复

EnnawYang, 2019-11-30:
建议重新git clone

d975xbx, 2019-11-30:
actions云编译没问题的

zcluo, 2019-11-30:
没问题，我之前用ubuntu19一直编译有问题，昨天搞个个ubuntu14 lts64位，没什么毛病了就

feifan1113, 2019-11-30:
> 没问题，我之前用ubuntu19一直编译有问题，昨天搞个个ubuntu14 lts64位，没什么毛病了就

我二次编译出的问题

dovegq, 2019-11-30:
ubuntu 19 一样的问题

Her0R0cket, 2019-12-01:
[image]

CONFIGURE_VARS += \
	QT_QMAKE="$(TOOLCHAIN_DIR)/bin" \
	PKG_CONFIG_PATH="$(TOOLCHAIN_DIR)/lib/pkgconfig"

编译失败的话在qbittorrent的makefile里加上这一段就可以了
我用的ubuntu18.04,看回复好像非16的系统需要这些变量

feifan1113, 2019-12-01:
> [image]
> 
> CONFIGURE_VARS +=
> QT_QMAKE="$(TOOLCHAIN_DIR)/bin"
> PKG_CONFIG_PATH="$(TOOLCHAIN_DIR)/lib/pkgconfig"
> 
> 编译失败的话在qbittorrent的makefile里加上这一段就可以了
> 我用的ubuntu18.04,看回复好像非16的系统需要这些变量

我又重建了虚拟机 重新编译，大雕说删除lede重新编译，就怕又不行，

Her0R0cket, 2019-12-01:
我直接编译失败了,然后加了这段就成功了,然后删了这段编译又失败,最好加上这段生成固件成功

Her0R0cket, 2019-12-01:
我直接编译失败了,然后加了这段就成功了,然后删了这段编译又失败,最后加上这段生成固件成功

feifan1113, 2019-12-01:
> 我直接编译失败了,然后加了这段就成功了,然后删了这段编译又失败,最后加上这段生成固件成功

。。。我的机子还在编译。。。。没加这段

feifan1113, 2019-12-01:
> > 我直接编译失败了,然后加了这段就成功了,然后删了这段编译又失败,最后加上这段生成固件成功
> 
> 。。。我的机子还在编译。。。。没加这段

我现在加上看看  还没到编译QB的部分 应该能赶上

feifan1113, 2019-12-01:
> 我直接编译失败了,然后加了这段就成功了,然后删了这段编译又失败,最后加上这段生成固件成功

[image]

[image]
是加到这个吧？

Her0R0cket, 2019-12-01:
是的,但前面那两条线是什么情况?finalshell按tab键的标志?

Her0R0cket, 2019-12-01:
刚虚拟机里试了下luci界面配置全了,但是会覆盖webui里的设置,这个就比较尬了

feifan1113, 2019-12-01:
> 是的,但前面那两条线是什么情况?finalshell按tab键的标志?

我用的Sublime Text2这个软件，能显示makefile的tab符，用来找出导致makefile报错的空格符合

feifan1113, 2019-12-01:
> 刚虚拟机里试了下luci界面配置全了,但是会覆盖webui里的设置,这个就比较尬了

意思是加的那两段会有问题？

Her0R0cket, 2019-12-01:
> 
> 
> > 刚虚拟机里试了下luci界面配置全了,但是会覆盖webui里的设置,这个就比较尬了
> 
> 意思是加的那两段会有问题？
不是,这次luci界面完善了,但这样就不能在webui里改了

feifan1113, 2019-12-01:
> > > 刚虚拟机里试了下luci界面配置全了,但是会覆盖webui里的设置,这个就比较尬了
> > 
> > 
> > 意思是加的那两段会有问题？
> 
> 不是,这次luci界面完善了,但这样在webui里改配置,再启动就被覆盖了

会不会是权限的原因

coolsnowwolf, 2019-12-02:
建议统一到 ubuntu 14 ，不然这里会有不同的行为

Her0R0cket, 2019-12-02:
> 
> 
> 建议统一到 ubuntu 14 ，不然这里会有不同的行为

之前一直用的16,这个月才升的18,我是从16复制过来的,16我还是保留了的.
其实就是想试试版本不一样到底有没有差别,没想到会在这个地方翻车!

dovegq, 2019-12-02:
> > 建议统一到 ubuntu 14 ，不然这里会有不同的行为
> 
> 之前一直用的16,这个月才升的18,我是从16复制过来的,16我还是保留了的.
> 其实就是想试试版本不一样到底有没有差别,没想到会在这个地方翻车!

我一直18编的好好的，也是在qt上翻车了

feifan1113, 2019-12-03:
> 建议统一到 ubuntu 14 ，不然这里会有不同的行为

[image]
二次编译最新的源码，qb会报错，然后用回之前我修改过的makefile就不会报错。很奇怪，对比过更新后的qb的makefile和我手动修改的makefile，也是加的那三行代码，内容一样的，是不是二次编译才会出问题？

coolsnowwolf, 2019-12-03:
更新 gcc 后解决了

closed 2019-12-05 by coolsnowwolf