coolsnowwolf/lede #2222: 竞斗云的ath10k-ct编译出错

xin122, 2019-11-30:
make[3]: Leaving directory '/home/zhao/lede/package/kernel/mac80211'
time: package/kernel/mac80211/compile#52.35#6.76#58.63
make[3]: Entering directory '/home/zhao/lede/package/kernel/ath10k-ct'
rm -f /home/zhao/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-2019-11-08-080cd781/.configured_*
rm -f /home/zhao/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.ath10k-ct_installed
cp /home/zhao/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/mac80211/ath/*.h /home/zhao/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-2019-11-08-080cd781
touch /home/zhao/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-2019-11-08-080cd781/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /home/zhao/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-2019-11-08-080cd781/.built
touch /home/zhao/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-2019-11-08-080cd781/.built_check
cat /dev/null > /home/zhao/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/ath10k-ct.symvers; for subdir in ath10k"-4.19"; do cat /home/zhao/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/*.symvers 2>/dev/null > /home/zhao/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-2019-11-08-080cd781/$subdir/Module.symvers; done
make CONFIG_ATH10K=m CONFIG_ATH10K_PCI=m CONFIG_ATH10K_CE=y CONFIG_ATH10K_AHB=m CONFIG_ATH10K_DEBUGFS=y CONFIG_MAC80211_DEBUGFS=y CONFIG_ATH10K_LEDS=y  -C "/home/zhao/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.14.155" HOSTCFLAGS="-O2 -I/home/zhao/lede/staging_dir/host/include -I/home/zhao/lede/staging_dir/hostpkg/include -I/home/zhao/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="arm-openwrt-linux-muslgnueabi-" ARCH="arm" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Fri Nov 29 10:23:25 2019" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/zhao/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KERNELRELEASE=4.14.155 SUBDIRS="/home/zhao/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-2019-11-08-080cd781/ath10k"-4.19"" NOSTDINC_FLAGS="-I/home/zhao/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-2019-11-08-080cd781 -I/home/zhao/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/mac80211-backport/uapi -I/home/zhao/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/mac80211-backport -I/home/zhao/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/mac80211/uapi -I/home/zhao/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/mac80211 -include backport/autoconf.h -include backport/backport.h -DCONFIG_MAC80211_MESH -DCONFIG_ATH10K_AHB -DSTANDALONE_CT -DCONFIG_MAC80211_DEBUGFS -DCONFIG_ATH10K_DEBUGFS -DCONFIG_ATH10K_DFS_CERTIFIED -DCONFIG_ATH10K_LEDS" modules
make[4]: Entering directory '/home/zhao/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.14.155'
  CC [M]  /home/zhao/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-2019-11-08-080cd781/ath10k-4.19/mac.o
/home/zhao/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-2019-11-08-080cd781/ath10k-4.19/mac.c:9038:30: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .cancel_remain_on_channel = ath10k_cancel_remain_on_channel,
                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/zhao/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-2019-11-08-080cd781/ath10k-4.19/mac.c:9038:30: note: (near initialization for 'ath10k_ops.cancel_remain_on_channel')
cc1: some warnings being treated as errors
scripts/Makefile.build:326: recipe for target '/home/zhao/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-2019-11-08-080cd781/ath10k-4.19/mac.o' failed
make[5]: *** [/home/zhao/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-2019-11-08-080cd781/ath10k-4.19/mac.o] Error 1
Makefile:1545: recipe for target '_module_/home/zhao/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-2019-11-08-080cd781/ath10k-4.19' failed
make[4]: *** [_module_/home/zhao/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-2019-11-08-080cd781/ath10k-4.19] Error 2
make[4]: Leaving directory '/home/zhao/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-4.14.155'
Makefile:108: recipe for target '/home/zhao/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-2019-11-08-080cd781/.built' failed
make[3]: *** [/home/zhao/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-2019-11-08-080cd781/.built] Error 2
make[3]: Leaving directory '/home/zhao/lede/package/kernel/ath10k-ct'
time: package/kernel/ath10k-ct/compile#3.27#0.59#3.84
package/Makefile:107: recipe for target 'package/kernel/ath10k-ct/compile' failed
make[2]: *** [package/kernel/ath10k-ct/compile] Error 2
make[2]: Leaving directory '/home/zhao/lede'
package/Makefile:103: recipe for target '/home/zhao/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/home/zhao/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zhao/lede'
/home/zhao/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

hzjnet, 2019-12-01:
ct从21日更新后到现在都不能用了

xin122, 2019-12-01:
> ct从21日更新后到现在都不能用了

为啥啊

closed 2019-12-03 by coolsnowwolf