coolsnowwolf/lede #2225: 【面向小白的重制版】OpenWrt CI 在线集成编译环境使用教程

Narizgnaw, 2019-12-01:
### 面向小白的Github_Action使用workflow自动编译lean_openwrt教程

https://zhuanlan.zhihu.com/p/94402324

### 关于Github Action自动编译Lean_Openwrt的配置修改问题

https://zhuanlan.zhihu.com/p/94527343

我用Word写的，先传渣乎了，不好意思。晚些时间我过来补充md版本。

感谢 @coolsnowwolf 大佬和 @KFERMercer 大佬的付出。

adaaaaaa, 2019-12-01:
如何在op-1907里面添加酸酸乳？？

mjyhj, 2019-12-01:
牛逼~~~~~~~，坐等补充files大法和涡轮增雕法的加入

Narizgnaw, 2019-12-02:
> 牛逼~~~~~~~，坐等补充files大法和涡轮增雕法的加入

？？？不是写了吗？第二篇第三节

Narizgnaw, 2019-12-02:
> 如何在op-1907里面添加酸酸乳？？

这个问题不太懂，
我估计是把这个文件夹按照第二篇第三节的方法，放在package/涡轮增雕文件夹里面。
https://github.com/coolsnowwolf/lede/tree/master/package/lean/luci-app-ssr-plus

btw：有人能帮忙测试下这个吗？
https://github.com/Lienol/openwrt-package/tree/master/lienol/luci-app-passwall

adaaaaaa, 2019-12-02:
> 我估计是把这个文件夹按照第二篇第三节的方法，放在package/涡轮增雕文件夹里面。
https://github.com/coolsnowwolf/lede/tree/master/package/lean/luci-app-ssr-plus

这玩意好像有几个依赖，单这个不行吧？

adaaaaaa, 2019-12-02:
怎么添加无线驱动？？编译过程报错了。。。

Run rm -f ./.config*
/home/runner/work/_temp/e3a2c1fc-8cfe-4620-ae34-f056d153cd57.sh: line 98: CONFIG_PACKAGE_kmod-ath10k=y: command not found
##[error]Process completed with exit code 127.

就添加一个6174是驱动。。。

mjyhj, 2019-12-03:
这配置文件只能默认没勾的加进去
如果要反向操作，默认打了勾的，想取消编译怎么搞
另外最好说明一下定时云编译，和手动云编译的介绍操作。
不然一堆人frok了，天天跟着同步云编译，浪费不少资源

adaaaaaa, 2019-12-03:
> 这配置文件只能默认没勾的加进去
> 如果要反向操作，默认打了勾的，想取消编译怎么搞

# xxx is not set
教程的配置文件里不是已经告诉你了吗？？认真看。。。

TUT123456, 2019-12-04:
本地编译k2p会默认勾选驱动，但是云编译没有在配置文件添加mt的驱动，这样编译出来的固件会不会没有wifi呀？

star-o0O, 2019-12-04:
- name: 同步最新的源码
        run: |
          git remote add openwrt https://github.com/coolsnowwolf/lede.git
          git pull openwrt master
          git remote -v
改动openwrt-ci.yml后会多一个或几个commit，用git pull命令无法合并，提示tell me who you are/refusing to merge unrelated histories之类的报错。查了一下用git fetch/stash/reset之类的命令好像也无法解决，请问有什么好办法吗？

lphgor, 2019-12-04:
> 本地编译k2p会默认勾选驱动，但是云编译没有在配置文件添加mt的驱动，这样编译出来的固件会不会没有wifi呀？

现在已经集成了那个闭源驱动，不再需要自己添加了。选k2p就会自动选中那个驱动的。

xynu617, 2019-12-06:
> ### 面向小白的Github_Action使用workflow自动编译lean_openwrt教程
> https://zhuanlan.zhihu.com/p/94402324
> 
> ### 关于Github Action自动编译Lean_Openwrt的配置修改问题
> https://zhuanlan.zhihu.com/p/94527343
> 
> 我用Word写的，先传渣乎了，不好意思。晚些时间我过来补充md版本。
> 
> 感谢 @coolsnowwolf 大佬和 @KFERMercer 大佬的付出。

报错了，可以帮忙看一下是什么问题吗？
[image]
只修改了下面两个地方
[image]
[image]

Narizgnaw, 2019-12-06:
> > ### 面向小白的Github_Action使用workflow自动编译lean_openwrt教程
> > https://zhuanlan.zhihu.com/p/94402324
> > ### 关于Github Action自动编译Lean_Openwrt的配置修改问题
> > https://zhuanlan.zhihu.com/p/94527343
> > 我用Word写的，先传渣乎了，不好意思。晚些时间我过来补充md版本。
> > 感谢 @coolsnowwolf 大佬和 @KFERMercer 大佬的付出。
> 
> 报错了，可以帮忙看一下是什么问题吗？
> [image]
> 只修改了下面两个地方
> [image]
> [image]

-a后面的一段代码被你给删了

xynu617, 2019-12-06:
> -a后面的一段代码被你给删了

> -a后面的一段代码被你给删了

刚才两个图片连在一起，看起来像是删了一段代码，其实没有，我把两个图隔开了，你再看一下

Narizgnaw, 2019-12-06:
> 刚才两个图片连在一起，看起来像是删了一段代码，其实没有，我把两个图隔开了，你再看一下

哦，这里我写的有问题，现在已删掉。KFER大佬重新写了一下，我已经换掉了，建议更换为我库里最新的版本，两个yml文件，一个用来自动pull push一个用来编译。

adaaaaaa, 2019-12-06:
VDI镜像还是搞不出来啊。。。用我的简短的配置移到130行附近和你提到的完整的配置拷下来放那里也不行。。。
你能弄出VDI镜像吗？？🙃🙃🙃

Narizgnaw, 2019-12-06:
> VDI镜像还是搞不出来啊。。。用我的简短的配置移到130行附近和你提到的完整的配置拷下来放那里也不行。。。
> 你能弄出VDI镜像吗？？🙃🙃🙃

没弄过，这个搞不定的话，还是建议本地编译好了

lphgor, 2019-12-06:
> VDI镜像还是搞不出来啊。。。用我的简短的配置移到130行附近和你提到的完整的配置拷下来放那里也不行。。。
> 你能弄出VDI镜像吗？？

你是不是用的KFERMercer的最新版呀？现在的脚本编译完成以后只会上传img和bin文件，vdi文件没有上传。
```
find ./bin/targets/ -name "*combined*img*" | xargs -i mv -f {} ./artifact/
find ./bin/targets/ -name "*sysupgrade*bin*" | xargs -i mv -f {} ./artifact/
```

KFERMercer, 2019-12-06:
> 你是不是用的KFERMercer的最新版呀？现在的脚本编译完成以后只会上传img和bin文件，vdi文件没有上传。

正解

adaaaaaa, 2019-12-07:
> 现在的脚本编译完成以后只会上传img和bin文件，vdi文件没有上传。

怎么改？？😬😬😬😬😬😬

bonnyandsky, 2019-12-07:
请问怎么自动同步你写的同步源码的yml文件？

adaaaaaa, 2019-12-07:
> 请问怎么自动同步你写的同步源码的yml文件？

禁止套娃。。。🤪🤪🤪🤪🤪🤪🤪

bonnyandsky, 2019-12-07:
> > 请问怎么自动同步你写的同步源码的yml文件？
> 
> 禁止套娃。。。🤪🤪🤪🤪🤪🤪🤪

我一开始没看懂你的回答，我以为开玩笑的，刚才查找actions的相关信息，算是明白了你说的套娃的意思。直接fork  Narizgnaw的项目就可以了，然后在他的项目里面修改openwrt-ci.yml文件就行了。

bonnyandsky, 2019-12-07:
但是我fork楼主的项目为啥在我仓库里没有？难道还要楼主同意下？？

bonnyandsky, 2019-12-07:
> > > 请问怎么自动同步你写的同步源码的yml文件？
> > 
> > 
> > 禁止套娃。。。🤪🤪🤪🤪🤪🤪🤪
> 
> 我一开始没看懂你的回答，我以为开玩笑的，刚才查找actions的相关信息，算是明白了你说的套娃的意思。直接fork Narizgnaw的项目就可以了，然后在他的项目里面修改openwrt-ci.yml文件就行了。

刚试了下，不能fork楼主的项目，不能套娃原来是这个意思，那fork KFERMercer的项目是不是就可以了，他的项目是每天同步lean_openwrt的么？

bonnyandsky, 2019-12-07:
我发现了一个好尴尬的问题，coolsnowwolf的.github/workflows是fork KFER的，，但是只有一个配置脚本，同步脚本没有。这样就意味着我fork coolsnowwolf的项目后还要同时fork KFER的脚本，这个我就不会了。。。楼主能否教程写一下？

sunlanhao, 2019-12-07:
请问想编译魔改版的ssr，添加了git clone 后 下面的定制固件还需要作什么修改吗？

sunlanhao, 2019-12-07:
请问想编译魔改版的ssr，添加了git clone 后 下面的定制固件还需要作什么修改吗？
[image]
[image]

adaaaaaa, 2019-12-08:
> [image]

adguardhome可以拦截手机广告，视频，https等的广告吗？？比电脑里的ublock origin效果怎样？？

sunlanhao, 2019-12-08:
> > [image]
> 
> adguardhome可以拦截手机，视频，https等的广告吗？？比电脑里的ublock origin效果怎样？？

我感觉拦截效果还是不错

peter-tank, 2019-12-08:
> btw：有人能帮忙测试下这个吗？
> https://github.com/Lienol/openwrt-package/tree/master/lienol/luci-app-passwall

我有[测试](https://github.com/peter-tank/openwrt)从最近的官网 release 编译，ubuntu 下，依赖的 redsocks2 [最近有更新](https://github.com/semigodking/redsocks/commits/master)，需要打补丁才能过。

saoba1982, 2019-12-09:
> ### 面向小白的Github_Action使用workflow自动编译lean_openwrt教程
> https://zhuanlan.zhihu.com/p/94402324
> 
> ### 关于Github Action自动编译Lean_Openwrt的配置修改问题
> https://zhuanlan.zhihu.com/p/94527343
> 
> 我用Word写的，先传渣乎了，不好意思。晚些时间我过来补充md版本。
> 
> 感谢 @coolsnowwolf 大佬和 @KFERMercer 大佬的付出。

> > 牛逼~~~~~~~，坐等补充files大法和涡轮增雕法的加入
> 
> 涡轮增雕法详见第二篇第三节。
> 
> ### 云编译的话，建议不要使用file大法，操作不当容易造成信息泄露。

你好，按照你的方法自动编译后，想加入的软件没有编译进去，只编译进去了一个皮肤是什么问题？

Narizgnaw, 2019-12-09:
> 你好，按照你的方法自动编译后，想加入的软件没有编译进去，只编译进去了一个皮肤是什么问题？

https://github.com/saoba1982/lede/blob/master/.github/workflows/openwrt-ci.yml
这个吗？这个你也没有进行修改啊

saoba1982, 2019-12-09:
> > 牛逼~~~~~~~，坐等补充files大法和涡轮增雕法的加入
> 
> 涡轮增雕法详见第二篇第三节。
> 
> ### 云编译的话，建议不要使用file大法，操作不当容易造成信息泄露。

> https://github.com/saoba1982/lede/blob/master/.github/workflows/openwrt-ci.yml
> 这个吗？这个你也没有进行修改啊

因为没有编译进去我就又恢复到没问题的了，我就是像编译皮肤一样的，编译两条命令进去，然后想加入一个魔改ssr和smartdns和adhome没编译进去，不知道什么原因.

Narizgnaw, 2019-12-09:
> 因为没有编译进去我就又恢复到没问题的了，我就是像编译皮肤一样的，编译两条命令进去，然后想加入一个魔改ssr和smartdns和adhome没编译进去，不知道什么原因.

魔改ssr和smartdns我没试过不太懂，如果编译adguardhome的话，需要加入luci-app才能在主界面看到该应用。如果不加luci-app的话，需要使用ssh命令开启该服务，确认3000端口通了才可以使用。

saoba1982, 2019-12-09:
> 魔改ssr和smartdns我没试过不太懂，如果编译adguardhome的话，需要加入luci-app才能在主界面看到该应用。如果不加luci-app的话，需要使用ssh命令开启该服务，确认3000端口通了才可以使用。
就是上边拷贝完地址，在下边软件那里加上luci-app-adguardhome=y
这样对吗？

saoba1982, 2019-12-09:
> ### 面向小白的Github_Action使用workflow自动编译lean_openwrt教程
> https://zhuanlan.zhihu.com/p/94402324
> 
> ### 关于Github Action自动编译Lean_Openwrt的配置修改问题
> https://zhuanlan.zhihu.com/p/94527343
> 
> 我用Word写的，先传渣乎了，不好意思。晚些时间我过来补充md版本。
> 
> 感谢 @coolsnowwolf 大佬和 @KFERMercer 大佬的付出。

可以把你添加adhome的添加源代码和地址给我发一下吗？我想编译进去.

sunlanhao, 2019-12-09:
> > ### 面向小白的Github_Action使用workflow自动编译lean_openwrt教程
> > https://zhuanlan.zhihu.com/p/94402324
> > ### 关于Github Action自动编译Lean_Openwrt的配置修改问题
> > https://zhuanlan.zhihu.com/p/94527343
> > 我用Word写的，先传渣乎了，不好意思。晚些时间我过来补充md版本。
> > 感谢 @coolsnowwolf 大佬和 @KFERMercer 大佬的付出。
> 
> 可以把你添加adhome的添加源代码和地址给我发一下吗？我想编译进去.

我的库里有，加了个adg 和serverchan，几个主题等。

Narizgnaw, 2019-12-09:
> 牛逼~~~~~~~，坐等补充files大法和涡轮增雕法的加入
>
>
> 涡轮增雕法详见第二篇第三节。
> ### 云编译的话，建议不要使用file大法，操作不当容易造成信息泄露。
>
>
> ### 面向小白的Github_Action使用workflow自动编译lean_openwrt教程
> https://zhuanlan.zhihu.com/p/94402324
> ### 关于Github [...]

你改完之后可以给我个链接，我帮你看一下改一下。

saoba1982, 2019-12-09:
> 你改完之后可以给我个链接，我帮你看一下改一下。

老大你还有别的联络方式吗？比如QQ或者微信啥的？

Narizgnaw, 2019-12-09:
> 老大你还有别的联络方式吗？比如QQ或者微信啥的？

QQ消息爆炸，还是别了

saoba1982, 2019-12-09:
> QQ消息爆炸，还是别了

这个我不大会用啊？微信可以吗？

Narizgnaw, 2019-12-09:
> 这个我不大会用啊？微信可以吗？

也不太行，这个有邮箱通知很方便

saoba1982, 2019-12-09:
> 也不太行，这个有邮箱通知很方便

这个可以私聊吗？如何私信？这感觉说话好难找.

saoba1982, 2019-12-09:
> ### 面向小白的Github_Action使用workflow自动编译lean_openwrt教程
> https://zhuanlan.zhihu.com/p/94402324
> 
> ### 关于Github Action自动编译Lean_Openwrt的配置修改问题
> https://zhuanlan.zhihu.com/p/94527343
> 
> 我用Word写的，先传渣乎了，不好意思。晚些时间我过来补充md版本。
> 
> 感谢 @coolsnowwolf 大佬和 @KFERMercer 大佬的付出。

给我一个链接，你编译代码带adghome的，我看看你是怎么编译 的

Narizgnaw, 2019-12-09:
> > ### 面向小白的Github_Action使用workflow自动编译lean_openwrt教程
> > https://zhuanlan.zhihu.com/p/94402324
> > ### 关于Github Action自动编译Lean_Openwrt的配置修改问题
> > https://zhuanlan.zhihu.com/p/94527343
> > 我用Word写的，先传渣乎了，不好意思。晚些时间我过来补充md版本。
> > 感谢 @coolsnowwolf 大佬和 @KFERMercer 大佬的付出。
> 
> 给我一个链接，你编译代码带adghome的，我看看你是怎么编译 的

上面加一组workflow：
`      - name: 构建自定义文件结构
        run: |
          cd package
          mkdir openwrt-packages
          cd openwrt-packages`

下面配置一栏加上
`CONFIG_PACKAGE_luci-app-adguardhome=y
CONFIG_PACKAGE_adguardhome=y`

就行

saoba1982, 2019-12-09:
> > > ### 面向小白的Github_Action使用workflow自动编译lean_openwrt教程
> > > https://zhuanlan.zhihu.com/p/94402324
> > > ### 关于Github Action自动编译Lean_Openwrt的配置修改问题
> > > https://zhuanlan.zhihu.com/p/94527343
> > > 我用Word写的，先传渣乎了，不好意思。晚些时间我过来补充md版本。
> > > 感谢 @coolsnowwolf 大佬和 @KFERMercer 大佬的付出。
> > 
> > 
> > 给我一个链接，你编译代码带adghome的，我看看你是怎么编译 的
> 
> 上面加一组workflow：
> 
> ```
>       - name: 加上adguardhome
>         run: |
>           cd package
>           mkdir openwrt-packages
>           cd openwrt-packages
>           git clone XXXXXXXXXXXXXXXXXXXXX（这里写ad的库地址）
> ```
> 
> 下面配置一栏加上
> 
> ```
> CONFIG_PACKAGE_luci-app-adguardhome=y
> CONFIG_PACKAGE_adguardhome=y
> ```
> 
> 就行

一会我编辑完打给你，帮我看看

saoba1982, 2019-12-09:
> 一会我编辑完打给你，帮我看看

      - name:  Software source
        run: |
          cd package
          mkdir openwrt-packages
          cd openwrt-packages
          git clone https://github.com/rosywrt/luci-theme-rosy
          git clone https://github.com/AdguardTeam/AdGuardHome.git
下边我打的和你一样
          CONFIG_PACKAGE_luci-app-v2ray-server=y
          CONFIG_PACKAGE_luci-app-verysync=y
          CONFIG_PACKAGE_luci-app-webadmin=y
          CONFIG_PACKAGE_luci-app-wireguard=y
          CONFIG_PACKAGE_luci-app-wrtbwmon=y
          CONFIG_PACKAGE_luci-app-AdGuardHome=y
          CONFIG_PACKAGE_AdGuardHome=y
这样可以吗？

Narizgnaw, 2019-12-09:
> ```
>   - name:  Software source
>     run: |
>       cd package
>       mkdir openwrt-packages
>       cd openwrt-packages
>       git clone https://github.com/rosywrt/luci-theme-rosy
>       git clone [...]

当然不可以，你没有添加适配openwrt的adguardhome项目

saoba1982, 2019-12-09:
> 当然不可以，你没有添加适配openwrt的adguardhome项目

      - name:  Software source
        run: |
          cd package
          mkdir openwrt-packages
          cd openwrt-packages
          git clone https://github.com/rosywrt/luci-theme-rosy
          git clone https://github.com/AdguardTeam/AdGuardHome.git
上边我这样打的

saoba1982, 2019-12-09:
>       - name: 加上adguardhome
>         run: |
>           cd package
>           mkdir openwrt-packages
>           cd openwrt-packages
>           git clone XXXXXXXXXXXXXXXXXXXXX（这里写ad的库地址）

      - name: Software source
        run: |
          cd package
          mkdir openwrt-packages
          cd openwrt-packages
          git clone https://github.com/rosywrt/luci-theme-rosy
     
      - name: AdGuardHome
        run: |
          cd package
          mkdir openwrt-packages
          cd openwrt-packages                 
          git clone https://github.com/AdguardTeam/AdGuardHome.git
这次 你看对不对？

KFERMercer, 2019-12-09:
加油, 再叠几层就超过哈里发塔了.

saoba1982, 2019-12-09:
> 加油, 再叠几层就超过哈里发塔了.

大神可以指导我一下吗？我真的是很小白。帮我看看该如何写这个代码

i6o6i, 2019-12-09:
没有TUI真的不知道在哪找对应CPU型号的TARGET参数。。。方便贴一下Nano R1s的这几行参数吗？（实在不想装ubuntu了, 要是Archlinux还可以，而且最近梯子有点抽风，下载镜像都要很久）

adaaaaaa, 2019-12-09:
> CONFIG_PACKAGE_adguardhome=y

好像这个不需要也行。。。还有就是直接去那个项目那里下载luci插件就行了，没必要从来编译这玩意。。。🤪🤪🤪

closed 2019-12-24 by coolsnowwolf