coolsnowwolf/lede #2244: qb在斐讯K3上不能出现web登录页面！！！

zxlhhyccc, 2019-12-02:
[image]

KFERMercer, 2019-12-02:
lean好像说不支持k3了

coolsnowwolf, 2019-12-03:
不再支持不带 FPU 的ARM 处理器了，例如 K3

zxlhhyccc, 2019-12-03:
> 不再支持不带 FPU 的ARM 处理器了，例如 K3

郁闷，请问以前的源码还支持吧

hcym, 2019-12-03:
都忘了怎么刷k3

coolsnowwolf, 2019-12-03:
最后一个支持的版本是 4.17

zxlhhyccc, 2019-12-03:
> 最后一个支持的版本是4.17

OK
谢谢大佬，我试试4.17，不过现在的luci适合这个版本吧？

zxlhhyccc, 2019-12-03:
> 最后一个支持的版本是 4.17

大佬，NPS在K3上面也是不运行，请问有可以支持k3的nps版本吗？

coolsnowwolf, 2019-12-03:
NPS 需要 带 FPU 的 ， K3 不支持

zxlhhyccc, 2019-12-03:
> NPS 需要 带 FPU 的 ， K3 不支持

NPS我向原作者提交了bug，原作者编译了v5版本，但运行提示配置文件不存在
[image]

zxlhhyccc, 2019-12-03:
> > 最后一个支持的版本是4.17
> 
> OK
> 谢谢大佬，我试试4.17，不过现在的luci适合这个版本吧？

大佬，经测试，你的最新qb4.19版本能在k3上运行，只是最新的luci无法启动它，降级为原1.04版本的luci可以启动该插件，麻烦修复一下：
[image]

coolsnowwolf, 2019-12-03:
K3 就让它死吧，这种老旧的 ARM CPU

zxlhhyccc, 2019-12-03:
> K3 就让它死吧，这种老旧的 ARM CPU

要是能修复一下最好了，起码我自己就在用，几台了，而且毕竟用k3的人还是很多。。。

closed 2019-12-04 by coolsnowwolf