coolsnowwolf/lede #2245: What is this fork about?

CodeFetch, 2019-12-02:
Why is this fork so popular? If there is a feature that OpenWrt does not have, it should be added.
Maintaining a heavily modified fork of an open-source project is absurd. What am I missing?

oudoubleyang, 2019-12-02:
This fork is about compiling custom ROMs in a quite easy way (for those who are not so familiar with Linux).

The fork itself seems to be automatically synced with the mainstream, so there might be no new features.

I don't know if the main OpenWrt project also have a similar building tool, but most tutorials about making your own custom OpenWrt (and automatically renew) lead to here.

rufengsuixing, 2019-12-03:
if you download openwrt，just as you will get a house with just walls ，you can make it a good house，but not easy.And now there is a house was decorated ,you just need bring your bag and live in.

oudoubleyang, 2019-12-03:
Just like what @rufengsuixing said, the most awesome feature of this fork should be the **user-friendly and robust** modules (packages) pre-compiler.

The OpenWrt dominates the router world because if its scalability, as I known. However such a thing is never easy for general users: the original system only contains basic packages, and manually install is needed. But it's difficult, since finding all dependencies, installing them and avoiding interrupts (with system libraries) are hard.

Besides, manually installed packages takes more spaces than pre installed ones.

In this fork you are able to add your preferred packages by a visual configuration program (`make menuconfig`). The program does the customizing, debugging and compiling things. The author also **guaranteed** that if you do follow the steps, you can surely get your own working system.

Here is an example:

* Checks: `wol` (base) or `luci-i18n-wol-zh-cn` (top)
* Installs:
    * * `wol`
    * * `etherwake`
    * * `luci-app-wol`
    * * `luci-i18n-wol-zh-cn` (based on your chosen language)

CodeFetch, 2019-12-03:
So it seems to me you were all lead here by some website who tells that this fork is any better than official OpenWrt?
Because all the features you've mentioned are ordinary in OpenWrt and not special to this fork... In OpenWrt you call "make menuconfig" and configure everything and then call "make" and it does the magic. That's how it always was.

> This fork is about compiling custom ROMs in a quite easy way (for those who are not so familiar with Linux).

I think the build process is identical to OpenWrt's.

> I don't know if the main OpenWrt project also have a similar building tool, but most tutorials about making your own custom OpenWrt (and automatically renew) lead to here.

Yes, it has. And that's bad, because tutorials should lead to https://openwrt.org/docs/guide-developer/build-system/use-buildsystem and not an unofficial fork.

> In this fork you are able to add your preferred packages by a visual configuration program (`make menuconfig`).

Please try official OpenWrt and tell me whether you see a difference.

> The author also **guaranteed** that if you do follow the tutorial steps, you can surely get your own working system.

It makes me very, very sceptical when someone **guarantees** anything for a collaborative open-source software project...

oudoubleyang, 2019-12-03:
I will try to make another by referring to the official documents recently, and report here if there's anything valuable...
(Besides I would always want a stable build, but this fork only offers snapshot versions)

Narizgnaw, 2019-12-04:
My English is not very proficlient.And I did know any programming languages.So I prefer Chinese-friendly project.This fork easy to facilitate Chinese user feedback and to provide program documentation and information.It helps me a lot.

hanwckf, 2019-12-04:
> So it seems to me you were all lead here by some website which tells that this fork is in any way better than official OpenWrt?
> Because all the features you've mentioned are ordinary in OpenWrt and not special to this fork... In OpenWrt you call "make menuconfig" and configure everything and then call "make" and it does the magic. That's how it always was.
> 
> > This fork is about compiling custom ROMs in a quite easy way (for those who are not so familiar with Linux).
> 
> I think the build process is identical to OpenWrt's.
> 
> > I don't know if the main OpenWrt project also have a similar building tool, but most tutorials about making your own custom OpenWrt (and automatically renew) lead to here.
> 
> Yes, it has. And that's bad, because tutorials should lead to https://openwrt.org/docs/guide-developer/build-system/use-buildsystem and not an unofficial fork.
> 
> > In this fork you are able to add your preferred packages by a visual configuration program (`make menuconfig`).
> 
> Please try official OpenWrt and tell me whether you see a difference.
> 
> > The author also **guaranteed** that if you do follow the tutorial steps, you can surely get your own working system.
> 
> It makes me very, very sceptical when someone **guarantees** anything for a collaborative open-source software project...

I agree your point. But the significance of its existence is that many Chinese people need some awesome packages which are not contained in official Openwrt sources. 

BTW, this repo supports lots of new devices which only can be found in China, which are not supported in official Openwrt

Promix953, 2019-12-04:
1.Communicate in Chinese
2.Chinese people need some special packages
3.Chinese documents and tutorials
Official OpenWrt can't provide these features

oudoubleyang, 2019-12-04:
> I agree your point. But the significance of its existence is that many Chinese people need some awesome packages which are not contained in official Openwrt source (these packages are not suitable for official repo).

Yes. The mentioned packages are in [package/lean](../tree/master/package/lean). They are highly customized and optimized mainly for some special user environments, namely in China. 

Since OpenWrt faces to the world, some packages there might (or surely would) violates some copyright laws in some region (for example, `vlmscd`), so they can't be added to the upstream.

Here are some examples:
```
adbyby: blocking ads
qBittorrent: BT client
shadowsocksr-libev / v2ray: proxy
vlmcsd: activating MS products
```

> BTW, lots of new devices which only can be found in China are better supported by this repo.

I don't think so. The supported devices list and configuration files are exactly the same, as I know.

So I would conclude that this fork is for providing packages which official OpenWrt can't (and shouldn't).

---

For those who love the official repo but also want these packages, I would recommend:
```bash
git clone https://github.com/openwrt/openwrt.git
git clone https://github.com/coolsnowwolf/lede.git
cp lede/package/lean openwrt/package
```

And then execute the `cd openwrt && ./scripts/feeds update -a &&  ./scripts/feeds install -a && make menuconfig` commands.

CodeFetch, 2019-12-04:
Thank you for the explanations. This shows me that there is a big demand for OpenWrt to open up to the Chinese community. I became aware of this repository because it seems to rank higher in Github's popularity than the official OpenWrt repository.  Thus a description in the README of what makes this fork special would be very nice. People seem to be confused about it like I am.

chchia, 2019-12-06:
> Thank you for the explanations. This shows me that there is a big demand for OpenWrt to open up to the Chinese community. I became aware of this repository because it seems to rank higher in [...]

if you are aware about the Great Firewall in China, then you will understand why this fork was so popular, it have some tools (shadowsocks and v2ray etc) that provide easy access for Chinese netizen for accessing those blocked website.
and this site is mainly communicated in Mandarin, so almost every OpenWRT user in China will come here. including me who are from Malaysia but traveling to China frequently also visting here. :)

CodeFetch, 2019-12-10:
> 1.Communicate in Chinese

I really don't want to start a discussion about that, but it's not in the sense of open-source as it limits the audience. I'm not a native English speaker, too, but I think Chinese people should use a translator on Github if they don't speak English. It would just be paradox if an OpenWrt fork that offers packages for circumventing the great firewall enforces the Chinese language. Language barriers divide people and English is the global language for computer science. It united people making projects like OpenWrt a reality. When Chinese developers only contribute their work in Mandarin it's likely lost for the rest of the world.
Let me make my point clear, because I didn't want to start a political discussion: OpenWrt developers (of which most are not native English speakers) put a whole lot of work into the development. If there are any changes in this project that could be integrated into official OpenWrt even if they are only useful for a minority of people, it would just be fair to contribute them in the way the OpenWrt project wishes to (the code in English on the official OpenWrt repository and the (possibly multi-lingual) documentation on the OpenWrt Wiki). There is a Chinese documentation in OpenWrt, but it's bad, because Chinese people seem to use sources like this Github repository or websites like CSDN where western people can't even register due to Kanji captchas. https://oldwiki.archive.openwrt.org/zh-cn/doc/howto/user.beginner

closed 2019-12-15 by coolsnowwolf

xkszltl, 2023-12-23:
@CodeFetch 
Happen to come across this thread.
It has been a long while but I think this is an interesting topic and wanna share a few thoughts.

I agree with your point of sharing, but you're largely underestimating the language barrier.

For Chinese people who don't use English in real life, it is not really feasible to directly dive into an English discussion just with a translator.
The difference between languages is not only the wording, but also in text layout, flow of thoughts, culture-specific focus point, and the atmosphere of inclusion.
The density of letter is much higher in English comparing to Chinese and inexperienced reader would be overwhelmed right at the beginning.
Besides, English education in China is focused in reading/formal writing, not speaking/informal discussion.
None of these can be shadowed by a translator, and it is not even trivial to figure out where to start the translation from.

What you saw on GitHub is already heavily bias.
People who are willing the discuss on GitHub - A English dominated website, is a very rare portion of users.
I would say the majority of people simply walk away when seeing English context.
So this problem, backed by the large population of Chinese users enough to create a sustainable fork of Internet, will still exist for long if not forever.
We would need a much stronger solution than just “English is the global language for computer science”.

xkszltl, 2023-12-23:
> English is the global language for computer science

And, this is not true anyway.
Majority of CS students are not capable of effective communication in English, even with translator in hand.
They may speak only Chinese and C for their entire life, do their job and develop awesome products just fine.

> and the (possibly multi-lingual) documentation on the OpenWrt Wiki

This is another issue.
Translation is a hard work without much credit, and will soon rust away without constant devotion.
Same for registrations and doc update request in maintainer’s language.

I always encourage people to contribute back to the very upstream, but not always getting positive response.
That’s not good, but I can understand their choice.