coolsnowwolf/lede #2247: 最新代码x86-64编qb报错

zcluo, 2019-12-03:
checking for Qt5 qmake >= 5.5.1... not found                                                           
configure: error: Could not find qmake                                                                 
Makefile:63: recipe for target '/home/zcluo/lede/build_dir/target-x86_64_musl/qBittorrent-release-4.1.9/.configured_68b329da9893e34099c7d8ad5cb9c940' failed                                                  
make[3]: *** [/home/zcluo/lede/build_dir/target-x86_64_musl/qBittorrent-release-4.1.9/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1                                                                   
make[3]: Leaving directory '/home/zcluo/lede/package/lean/qBittorrent'                                 
time: package/lean/qBittorrent/compile#5.54#0.26#5.68                                                  
package/Makefile:107: recipe for target 'package/lean/qBittorrent/compile' failed                      
make[2]: *** [package/lean/qBittorrent/compile] Error 2                                                
make[2]: Leaving directory '/home/zcluo/lede'                                                          
package/Makefile:103: recipe for target '/home/zcluo/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed                                                                                       
make[1]: *** [/home/zcluo/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2          
make[1]: Leaving directory '/home/zcluo/lede'                                                          
/home/zcluo/lede/include/toplevel.mk:216: recipe for target 'world' failed                             
make: *** [world] Error 2                                                                              
Start compile at 2019-12-02 23:28:43                                                                   
End compile at 2019-12-03 01:35:18                                                                     
Run time 126 mins

feifan1113, 2019-12-03:
[Makefile.zip](https://github.com/coolsnowwolf/lede/files/3914219/Makefile.zip)
你试试我这个qBittorrent的makefile，替换package-lean-qBittorrent的原文件，我二次编译也有这提示，用回我之前自己修改的编译就没问题，不知道为什么

zcluo, 2019-12-03:
谢谢

zcluo, 2019-12-03:
> [Makefile.zip](https://github.com/coolsnowwolf/lede/files/3914219/Makefile.zip)
> 你试试我这个qBittorrent的makefile，替换package-lean-qBittorrent的原文件，我二次编译也有这提示，用回我之前自己修改的编译就没问题，不知道为什么

用你这个确实可以了应该就是哪个斜杠的问题

zcluo, 2019-12-03:
qBittorrent的makefile
@coolsnowwolf 
[image]
没有斜杠哪个是可以编译通过的能否修复下，谢谢
@feifan1113

coolsnowwolf, 2019-12-03:
不是这个的问题，新版 qmake 安装到 toolchain 的 /usr/bin 下面了，除非你编译过老版本的 qt5

coolsnowwolf, 2019-12-03:
老版本 qmake 路径是 toochain 的 /bin

zcluo, 2019-12-03:
那我make clean重新编译一遍

zcluo, 2019-12-03:
可以了灭有问题了，谢谢 @coolsnowwolf

closed 2019-12-03 by zcluo