coolsnowwolf/lede #2250: 自动编译，uhttpd不能自动带上依赖。

Narizgnaw, 2019-12-03:
之前用手动编译的时候，luci-app-uhttpd是自动带ssl相关依赖的。
切换为自动编译勾选luci-app-uhttpd之后，默认不会自动带上ssl相关依赖，开启https会报错，需要再次手动安装ssl依赖。
希望勾选luci-app-uhttpd之后，可以自动勾选以下依赖：
luci-i18n-uhttpd-zh-cn
luci-ssl-openssl
px5g-mbedtls

adaaaaaa, 2019-12-03:
在op里面试试ssr呢？？🤪🤪🤪

adaaaaaa, 2019-12-03:
我编译的固件为何不包括VDI镜像？我明明已经开启的。。。
大佬有空看看我的源码。。。

Narizgnaw, 2019-12-03:
> 在op里面试试ssr呢？？🤪🤪🤪

？？？

Narizgnaw, 2019-12-03:
> 我编译的固件为何不包括VDI镜像？我明明已经开启的。。。
> 大佬有空看看我的配置文件。。。

### 1.你直接把.config里面的Root filesystem images拷过来试试，类似这样：
```
#
# Root filesystem images
#
# CONFIG_TARGET_ROOTFS_EXT4FS is not set
CONFIG_TARGET_ROOTFS_SQUASHFS=y
CONFIG_TARGET_SQUASHFS_BLOCK_SIZE=256
CONFIG_TARGET_UBIFS_FREE_SPACE_FIXUP=y
CONFIG_TARGET_UBIFS_JOURNAL_SIZE=""
CONFIG_GRUB_IMAGES=y
# CONFIG_EFI_IMAGES is not set
CONFIG_GRUB_CONSOLE=y
CONFIG_GRUB_SERIAL="ttyS0"
CONFIG_GRUB_BAUDRATE=115200
# CONFIG_GRUB_FLOWCONTROL is not set
CONFIG_GRUB_BOOTOPTS=""
CONFIG_GRUB_TIMEOUT="0"
CONFIG_GRUB_TITLE="OpenWrt"
CONFIG_VDI_IMAGES=y
CONFIG_VMDK_IMAGES=y
# CONFIG_VHD_IMAGES is not set
# CONFIG_QCOW2_IMAGES is not set
CONFIG_TARGET_IMAGES_PAD=y
# CONFIG_TARGET_IMAGES_GZIP is not set
```

### 2.把这一项放在Image Options前面
正常顺序应该是在Root filesystem archives和Image Options之间（你openwrt-ci.yml第130行左右的位置）。

### 3.我也不懂，建议仅供参考。

slitilanber, 2019-12-03:
op 18.06上面试过ssr, 基本可用。但新版新版Luci和lean的插件兼容性有些小问题。

adaaaaaa, 2019-12-04:
老哥，你的自主创新之路呢？？怎么不要了，抄别人的了？🤪🤪🤪

Narizgnaw, 2019-12-04:
> 老哥，你的自主创新之路呢？？怎么不要了，抄别人的了？🤪🤪🤪

我不会编程。。。。我只是到github上面找好玩的东西，看到哪个项目好就抄一抄。你看我库里昨晚又更新了KFER大佬做的一个自动同步大佬源码的脚本。

closed 2019-12-04 by coolsnowwolf