coolsnowwolf/lede #2254: mac80211 Error

iamggs, 2019-12-03:
`mkdir -p /home/iamgs/openwrt/lede/bin/targets/x86/64/packages /home/iamgs/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.85-1/ipkg-x86_64/kmod-mac80211/CONTROL /home/iamgs/openwrt/lede/staging_dir/target-x86_64_musl/pkginfo
ERROR: module '/home/iamgs/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.85-1/net/mac80211/mac80211.ko' is missing.
Makefile:561: recipe for target '/home/iamgs/openwrt/lede/bin/targets/x86/64/packages/kmod-mac80211_4.19.86+4.19.85-1-2_x86_64.ipk' failed
make[3]: *** [/home/iamgs/openwrt/lede/bin/targets/x86/64/packages/kmod-mac80211_4.19.86+4.19.85-1-2_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/iamgs/openwrt/lede/package/kernel/mac80211'
time: package/kernel/mac80211/compile#23.14#6.91#57.14
package/Makefile:107: recipe for target 'package/kernel/mac80211/compile' failed
make[2]: *** [package/kernel/mac80211/compile] Error 2
make[2]: Leaving directory '/home/iamgs/openwrt/lede'
package/Makefile:103: recipe for target '/home/iamgs/openwrt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/iamgs/openwrt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/iamgs/openwrt/lede'
/home/iamgs/openwrt/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2`

这周就没顺过，之前选中qemu-ga必报错（据说是路径有问题），把qemu-ga去掉后mac80211又报错。

iamggs, 2019-12-03:
我就给软路由用，其实我并不需要wifi驱动，请问我怎么能去掉这个mac80211？menuconfig里面去不掉，手动编辑.config屏蔽mac80211的话，编译的时候又会自动给加上，搞不懂。

keke1023, 2019-12-03:
一模一样的问题，是backports的版本不对么?

OtakuK, 2019-12-03:
如果不想要WiFi驱动，可以尝试下：
把Kernel Modules - Wireless Drivers下的wireless drivers取消，
还需要把Network下的 wpad. 取消。
建议在删减drivers时，尝试用 / 来搜索关键字看还有哪些相关依赖的drivers也一起取消

keke1023, 2019-12-03:
wpad，hostapd，还有所有无线驱动都去掉就好了

coolsnowwolf, 2019-12-03:
你们的 ubuntu 版本不是 14 吧？

coolsnowwolf, 2019-12-03:
qemu-ga 和 mac80211 都是路径问题，只有在 Ubuntu 14 中正确生成

mrblack0407, 2019-12-03:
是ubuntu 14.04 LTS的，但是一样mac80211报错

darkalexx, 2019-12-03:
专门换成了14试了一下，依然是这个错误

iamggs, 2019-12-04:
> qemu-ga 和 mac80211 都是路径问题，只有在 Ubuntu 14 中正确生成

的确不是14，是Ubuntu 18，但这个环境使用已有大半年了，之前多次编译中也没遇到过mac80211错误。

zeXtreme, 2019-12-04:
我之前一直用默认配置编译没问题，后来改动了配置后编译总是报这个错，用 ubuntu 14 也一样

aesirteam, 2019-12-04:
经过验证ubuntu14依然无法成功编译4.19内核
$ cat /etc/os-release
NAME="Ubuntu"
VERSION="14.04.6 LTS, Trusty Tahr"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 14.04.6 LTS"
VERSION_ID="14.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"

重新git clone了代码，全局网络模式， make -j1 V=s
编译时错误：
install -d -m0755 /home/dev/lede/bin/targets/ipq40xx/generic/packages
/home/dev/lede/scripts/ipkg-build -c -o 0 -g 0 /home/dev/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/backports-4.19.85-1/ipkg-arm_cortex-a7_neon-vfpv4/kmod-cfg80211 /home/dev/lede/bin/targets/ipq40xx/generic/packages
Packaged contents of /home/dev/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/backports-4.19.85-1/ipkg-arm_cortex-a7_neon-vfpv4/kmod-cfg80211 into /home/dev/lede/bin/targets/ipq40xx/generic/packages/kmod-cfg80211_4.19.86+4.19.85-1-3_arm_cortex-a7_neon-vfpv4.ipk
mkdir -p /home/dev/lede/bin/targets/ipq40xx/generic/packages /home/dev/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/backports-4.19.85-1/ipkg-arm_cortex-a7_neon-vfpv4/kmod-mac80211/CONTROL /home/dev/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo
ERROR: module '/home/dev/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/backports-4.19.85-1/net/mac80211/mac80211.ko' is missing.
make[3]: *** [/home/dev/lede/bin/targets/ipq40xx/generic/packages/kmod-mac80211_4.19.86+4.19.85-1-3_arm_cortex-a7_neon-vfpv4.ipk] Error 1
make[3]: Leaving directory `/home/dev/lede/package/kernel/mac80211'
time: package/kernel/mac80211/compile#26.80#4.47#120.34
make[2]: *** [package/kernel/mac80211/compile] Error 2
make[2]: Leaving directory `/home/dev/lede'
make[1]: *** [/home/dev/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/dev/lede'

烦请coolsnowwolf看看是否有法解决在这个问题，这个问题一直在ISSUE上重复提出

coolsnowwolf, 2019-12-04:
[image]
自己看看 mac80211.ko 放哪里了

darkalexx, 2019-12-04:
根本就搜不到mac80211.ko这个文件

wspzf, 2019-12-04:
x86编译时也出现同样问题

EnnawYang, 2019-12-04:
最近编译都没问题 x86_64

flyfeel, 2019-12-04:
大雕能不能解决下啊 这个mac80211报错 搞一天 什么方法都试了，崩溃了。。。

darkalexx, 2019-12-04:
[image]

iamggs, 2019-12-04:
> 最近编译都没问题 x86_64

1.编译环境cat /etc/os-release信息贴上来；
2.最近一次git pull是啥时候？

jxph1123, 2019-12-04:
重新下载编译。。

iamggs, 2019-12-04:
> qemu-ga 和 mac80211 都是路径问题，只有在 Ubuntu 14 中正确生成

准备全新部署一个Ubuntu14.04.6的编译环境来全新编译，等我消息。

darkalexx, 2019-12-04:
> > qemu-ga 和 mac80211 都是路径问题，只有在 Ubuntu 14 中正确生成
> 
> 准备全新部署一个Ubuntu14.04.6的编译环境来全新编译，等我消息。

我就是昨天重新装的14.04.6的虚拟机从头开始的，还是这个错误

iamggs, 2019-12-04:
> > > qemu-ga 和 mac80211 都是路径问题，只有在 Ubuntu 14 中正确生成
> > 
> > 
> > 准备全新部署一个Ubuntu14.04.6的编译环境来全新编译，等我消息。
> 
> 我就是昨天重新装的14.04.6的虚拟机从头开始的，还是这个错误

那完蛋，我已经开始了，等晚上再看看。

qyh214, 2019-12-04:
这个问题订阅下，我也遇到了，2天了，全局梯子+删掉库重新下都进行过了……第一个ac58u的固件都过不去……

star-o0O, 2019-12-04:
似乎找到问题所在了？默认的luci-app-pptp-server不要去掉，去掉后直接不生成mac80211.ko这个文件（搜索不到）我这本地试了3次，只要选中luci-app-pptp-server就能成功。不知道是不是个例，大雕可以看一下。编译环境ubuntu19.10桌面版，设备ACRH17。
**不要去掉luci-app-pptp-server**
下面是我的.config配置文件。

#
# Automatically generated file; DO NOT EDIT.
# OpenWrt Configuration
#
CONFIG_MODULES=y
CONFIG_HAVE_DOT_CONFIG=y
# CONFIG_TARGET_sunxi is not set
# CONFIG_TARGET_apm821xx is not set
# CONFIG_TARGET_ath25 is not set
# CONFIG_TARGET_ar71xx is not set
# CONFIG_TARGET_ath79 is not set
# CONFIG_TARGET_brcm2708 is not set
# CONFIG_TARGET_bcm53xx is not set
# CONFIG_TARGET_brcm47xx is not set
# CONFIG_TARGET_brcm63xx is not set
# CONFIG_TARGET_cns3xxx is not set
# CONFIG_TARGET_octeon is not set
# CONFIG_TARGET_gemini is not set
# CONFIG_TARGET_mpc85xx is not set
# CONFIG_TARGET_imx6 is not set
# CONFIG_TARGET_mxs is not set
# CONFIG_TARGET_adm8668 is not set
# CONFIG_TARGET_adm5120 is not set
# CONFIG_TARGET_xburst is not set
# CONFIG_TARGET_ixp4xx is not set
# CONFIG_TARGET_lantiq is not set
# CONFIG_TARGET_malta is not set
# CONFIG_TARGET_pistachio is not set
# CONFIG_TARGET_mvebu is not set
# CONFIG_TARGET_kirkwood is not set
# CONFIG_TARGET_mediatek is not set
# CONFIG_TARGET_ramips is not set
# CONFIG_TARGET_at91 is not set
# CONFIG_TARGET_rb532 is not set
# CONFIG_TARGET_mcs814x is not set
# CONFIG_TARGET_layerscape is not set
# CONFIG_TARGET_octeontx is not set
# CONFIG_TARGET_oxnas is not set
# CONFIG_TARGET_armvirt is not set
CONFIG_TARGET_ipq40xx=y
# CONFIG_TARGET_ipq806x is not set
# CONFIG_TARGET_au1000 is not set
# CONFIG_TARGET_samsung is not set
# CONFIG_TARGET_arc770 is not set
# CONFIG_TARGET_archs38 is not set
# CONFIG_TARGET_ar7 is not set
# CONFIG_TARGET_omap is not set
# CONFIG_TARGET_uml is not set
# CONFIG_TARGET_zynq is not set
# CONFIG_TARGET_x86 is not set
# CONFIG_TARGET_x86_64 is not set
# CONFIG_TARGET_x86_generic is not set
# CONFIG_TARGET_x86_core2 is not set
# CONFIG_TARGET_x86_zen is not set
# CONFIG_TARGET_x86_legacy is not set
# CONFIG_TARGET_x86_geode is not set
# CONFIG_TARGET_MULTI_PROFILE is not set
# CONFIG_TARGET_ipq40xx_Default is not set
# CONFIG_TARGET_ipq40xx_DEVICE_8dev_jalapeno is not set
# CONFIG_TARGET_ipq40xx_DEVICE_asus_map-ac2200 is not set
# CONFIG_TARGET_ipq40xx_DEVICE_avm_fritzbox-4040 is not set
# CONFIG_TARGET_ipq40xx_DEVICE_asus_rt-ac58u is not set
CONFIG_TARGET_ipq40xx_DEVICE_asus_rt-acrh17=y
# CONFIG_TARGET_ipq40xx_DEVICE_meraki_mr33 is not set
# CONFIG_TARGET_ipq40xx_DEVICE_compex_wpj428 is not set
# CONFIG_TARGET_ipq40xx_DEVICE_engenius_eap1300 is not set
[... 4897 lines cut ...]
# CONFIG_PACKAGE_forked-daapd is not set
# CONFIG_PACKAGE_ices is not set
# CONFIG_PACKAGE_lame is not set
# CONFIG_PACKAGE_lame-lib is not set
# CONFIG_PACKAGE_liblo-utils is not set
# CONFIG_PACKAGE_madplay is not set
# CONFIG_PACKAGE_madplay-alsa is not set
# CONFIG_PACKAGE_moc is not set
# CONFIG_PACKAGE_mpc is not set
# CONFIG_PACKAGE_mpd-avahi-service is not set
# CONFIG_PACKAGE_mpd-full is not set
# CONFIG_PACKAGE_mpd-mini is not set
# CONFIG_PACKAGE_mpg123 is not set
# CONFIG_PACKAGE_opus-tools is not set
# CONFIG_PACKAGE_pianod is not set
# CONFIG_PACKAGE_pianod-client is not set
# CONFIG_PACKAGE_portaudio is not set
# CONFIG_PACKAGE_pulseaudio-daemon is not set
# CONFIG_PACKAGE_pulseaudio-daemon-avahi is not set
# CONFIG_PACKAGE_pulseaudio-profiles is not set
# CONFIG_PACKAGE_pulseaudio-tools is not set
# CONFIG_PACKAGE_shairplay is not set
# CONFIG_PACKAGE_shairport-sync-mbedtls is not set
# CONFIG_PACKAGE_shairport-sync-mini is not set
# CONFIG_

coolsnowwolf, 2019-12-04:
其实你们看 CI 都自动编译成功了，我自己也没遇到任何问题

darkalexx, 2019-12-04:
光云编译成功不行啊，云编译太慢了。今天都多少人说编译失败了，而且都是这个问题

zhg-em, 2019-12-04:
这个东西确实是有问题的！默认编译，4.9 4.14 都没有问题，就是4.19出现这个问题，已经很长时间了！

star-o0O, 2019-12-04:
> 其实你们看 CI 都自动编译成功了，我自己也没遇到任何问题

大雕，我这选中luci-app-pptp-server就成功，不选中就报错，是这里的原因吗？
其他有出问题的朋友们也看看是不是这里，或者这样能不能暂时解决问题。

qyh214, 2019-12-04:
虚拟机：ubuntu 16.04 LTS 已安装全部需要的软件（根据readme），在之前也一直正常编译。

发生这个issue后，进行过rm 操作，完全重新弄，问题依旧……

配置方面：
- ac58u和acrh17均为CONFIG_LINUX_4_19=y，而网件r7800则是CONFIG_LINUX_4_14=y
- 均取消了pptp-server，ddns（luci）

结果：
- r7800编译成功；ac58u和acrh17全部在mac80211出错，日志与本issue一致。

gsblay, 2019-12-04:
> > 其实你们看 CI 都自动编译成功了，我自己也没遇到任何问题
> 
> 大雕，我这选中luci-app-pptp-server就成功，不选中就报错，是这里的原因吗？
> 其他有出问题的朋友们也看看是不是这里，或者这样能不能暂时解决问题。

同样，选中pptp-server，ddns编译4.19就成功，不选中肯定失败。Gl.iNet B1300.

linjieh, 2019-12-04:
> [image]
> 自己看看 mac80211.ko 放哪里了

找不到这个文件 是什么情况呢

darkalexx, 2019-12-04:
> > 其实你们看 CI 都自动编译成功了，我自己也没遇到任何问题
> 
> 大雕，我这选中luci-app-pptp-server就成功，不选中就报错，是这里的原因吗？
> 其他有出问题的朋友们也看看是不是这里，或者这样能不能暂时解决问题。

成了！居然真的可以，把luci-app-pptp-server勾选就不报错了！

qyh214, 2019-12-04:
> 
> 
> > > 其实你们看 CI 都自动编译成功了，我自己也没遇到任何问题
> > 
> > 
> > 大雕，我这选中luci-app-pptp-server就成功，不选中就报错，是这里的原因吗？
> > 其他有出问题的朋友们也看看是不是这里，或者这样能不能暂时解决问题。
> 
> 成了！居然真的可以，把luci-app-pptp-server勾选就不报错了！

嗯，可以暂时解决问题，但是……自己并没有pptp-server的需求，还是等待大雕的更新了

lyin888, 2019-12-04:
有些大雕默认选择的APP我们都用不到，不取消空间受限导致固件过大打包失败，取消了偶尔出现各种错误。

iamggs, 2019-12-04:
> > > 其实你们看 CI 都自动编译成功了，我自己也没遇到任何问题
> > 
> > 
> > 大雕，我这选中luci-app-pptp-server就成功，不选中就报错，是这里的原因吗？
> > 其他有出问题的朋友们也看看是不是这里，或者这样能不能暂时解决问题。
> 
> 成了！居然真的可以，把luci-app-pptp-server勾选就不报错了！

这么神奇？明天试试。我一向没选pptp的。

zhg-em, 2019-12-04:
先选中luci-app-pptp-server编译成功后去掉就可以了！

keke1023, 2019-12-04:
实测确实在network里vpn选中pptpd就可以过这个error

coolsnowwolf, 2019-12-04:
这个不是默认的么？

darkalexx, 2019-12-04:
默认的好多用不到的，直接反勾选了

iamggs, 2019-12-04:
哭了，勾选pptp果然过了，一周了。。。。。。

aesirteam, 2019-12-04:
> 实测确实在network里vpn选中pptpd就可以过这个error

经过验证，此法在rt-acrh17编译4.19成功 (拉去最新代码，默认未勾选此项)

gsblay, 2019-12-04:
Gl.iNet 的 AR750s-ext 如上操作的确可以编译成功，但用 U-boot 刷入生成的 img 文件或 nor-nand-squashfs-sysupgrade.bin 到路由器，无线都无法启动，刷入 nor-nand-initramfs-kernel.bin 可以启动 2.4 G 无线，5G 无法启动。

iamggs, 2019-12-04:
> qemu-ga 和 mac80211 都是路径问题，只有在 Ubuntu 14 中正确生成

经过测试，目前结果是：无论编译环境是Ubuntu18还是Ubuntu14：
1、只要选中pptp-server就不会报mac80211错误；
2、只要选中qemu-ga就会报错；
3、无论18还是14，只要选中pptp-server且不要选中qemu-ga，编译可以通过。

hyird, 2019-12-04:
> > qemu-ga 和 mac80211 都是路径问题，只有在 Ubuntu 14 中正确生成
> 
> 经过测试，目前结果是：无论编译环境是Ubuntu18还是Ubuntu14：
> 1、只要选中pptp-server就不会报mac80211错误；
> 2、只要选中qemu-ga就会报错；
> 3、无论18还是14，只要选中pptp-server且不要选中qemu-ga，编译可以通过。

真的诶 只要选中了pptp-server 我立马编译就过了

hyird, 2019-12-04:
> 这个不是默认的么？

个人用不着 就反选了

hcym, 2019-12-05:
pptp 还有这种怪事，以后再说吧，已经被mac80211搞怕了，用的1907专门跑v2，感觉还不错

closed 2019-12-05 by iamggs