coolsnowwolf/lede #2278: 新编译的版本samba2不能正常访问

xiangfeigao, 2019-12-04:
新编译的版本samba2不能正常访问，客户端win7，正常挂载都没有问题，大家碰到这个问题吗

xiangfeigao, 2019-12-04:
Wed Dec  4 16:38:19 2019 kern.notice kernel: [  733.781363] scsi 0:0:0:0: Direct-Access     ATA      WDC WD2500BEVT-2 1A02 PQ: 0 ANSI: 6
Wed Dec  4 16:38:19 2019 kern.notice kernel: [  733.791430] sd 0:0:0:0: [sda] 488397168 512-byte logical blocks: (250 GB/233 GiB)
Wed Dec  4 16:38:19 2019 kern.notice kernel: [  733.799370] sd 0:0:0:0: [sda] Write Protect is off
Wed Dec  4 16:38:19 2019 kern.debug kernel: [  733.804180] sd 0:0:0:0: [sda] Mode Sense: 43 00 00 00
Wed Dec  4 16:38:19 2019 kern.notice kernel: [  733.804578] sd 0:0:0:0: [sda] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
Wed Dec  4 16:38:19 2019 kern.info kernel: [  733.823451]  sda: sda1
Wed Dec  4 16:38:19 2019 kern.notice kernel: [  733.828683] sd 0:0:0:0: [sda] Attached SCSI disk
Wed Dec  4 16:38:22 2019 daemon.info hostapd: wlan0: STA 48:98:ca:72:48:73 IEEE 802.11: deauthenticated due to local deauth request
Wed Dec  4 16:38:49 2019 kern.info kernel: [  764.350004] ntfs: volume version 3.1.
Wed Dec  4 16:38:49 2019 daemon.err block: hotplug-call call failed

coolsnowwolf, 2019-12-04:
默认 SMB3 ，需要 Windows 7 以上系统，例如 Windows 10

xiangfeigao, 2019-12-05:
Package samba36-server (3.6.25-15) installed in root is up to date，还是碰到权限不能访问，我看了配置应该没有问题。但新编译的就是不能访问。

closed 2019-12-07 by coolsnowwolf