coolsnowwolf/lede #2282: 更新luci-app-cifsd这个版本后，移动硬盘不自动挂载了

jxph1123, 2019-12-04:
没有选择编译luci-app-cifsd，但这个版本移动硬盘不自动挂载了，应该是luci-app-cifsd的更改影响了自动挂载。我正常的盘挂载是 sda5  ntfs格式。
望修复谢谢
Wed Dec  4 18:27:55 2019 kern.info kernel: [  386.347040] usb 2-1: USB disconnect, device number 2
Wed Dec  4 18:27:56 2019 daemon.err block: hotplug-call call failed
Wed Dec  4 18:27:56 2019 kern.warn kernel: [  386.806981] [EXFAT] trying to unmount...
Wed Dec  4 18:27:56 2019 kern.warn kernel: [  386.810996] [EXFAT] unmounted successfully
Wed Dec  4 18:29:06 2019 kern.info kernel: [  456.736636] usb 2-1: new SuperSpeed USB device number 3 using xhci-mtk
Wed Dec  4 18:29:06 2019 kern.info kernel: [  456.768312] usb-storage 2-1:1.0: USB Mass Storage device detected
Wed Dec  4 18:29:06 2019 kern.info kernel: [  456.775270] scsi host0: usb-storage 2-1:1.0
Wed Dec  4 18:29:09 2019 kern.notice kernel: [  460.072042] scsi 0:0:0:0: Direct-Access     ATA      SAMSUNG HM500JI  01C4 PQ: 0 ANSI: 6
Wed Dec  4 18:29:09 2019 kern.notice kernel: [  460.082319] sd 0:0:0:0: [sda] 976773168 512-byte logical blocks: (500 GB/466 GiB)
Wed Dec  4 18:29:09 2019 kern.notice kernel: [  460.090362] sd 0:0:0:0: [sda] Write Protect is off
Wed Dec  4 18:29:09 2019 kern.debug kernel: [  460.095181] sd 0:0:0:0: [sda] Mode Sense: 37 00 00 00
Wed Dec  4 18:29:09 2019 kern.notice kernel: [  460.095538] sd 0:0:0:0: [sda] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
Wed Dec  4 18:29:09 2019 kern.info kernel: [  460.123862]  sda: sda1 < sda5 >
Wed Dec  4 18:29:09 2019 kern.notice kernel: [  460.129802] sd 0:0:0:0: [sda] Attached SCSI disk
Wed Dec  4 18:29:40 2019 kern.warn kernel: [  491.003626] [EXFAT] trying to mount...
Wed Dec  4 18:29:40 2019 kern.warn kernel: [  491.007968] [EXFAT] mounted successfully
Wed Dec  4 18:30:11 2019 daemon.err block: No "mount.ntfs" utility available
Wed Dec  4 18:30:11 2019 daemon.err block: mounting /dev/sda5 (ntfs) as /mnt/sda5 failed (25) - Not a tty

EnnawYang, 2019-12-04:
自己把ntfs-3g加上

pk49800, 2019-12-04:
我也遇到使用问题，不能访问中文文件夹

coolsnowwolf, 2019-12-04:
没选 ntfs-3g

closed 2019-12-04 by coolsnowwolf