coolsnowwolf/lede #2288: 【教程】会本地编译的情况下，怎么用GitHub Actions云编译？

lietxia, 2019-12-04:
# 为啥写这个
1. 我看到其他2个教程**把【怎么选机型】的部分略过了**，这个很蛋疼，
因为并不是每个人都要编译x86固件，而且每个机型的机型写法都不相同，
你一个个的**告诉他们哪个机型怎么写很麻烦**，**反而是直接make menuconfig产生.config文件反而更简单**。

2. 其他2个教程是让你fork源码，但是你fork的话，不会自动和lean的项目同步。等于说**你要想办法解决冲突+想办法同步上游**

基于以上两点，我是比较推荐用`P3TERX`的actions代码的，简单说就是编译的时候抓取一次lean的项目，这样你一直是最新的。

# 前提
1. 已经会本地编译
2. 有GitHub账号

# 第1步
打开自己编译目录，先和大屌源码同步一下

     git pull
     ./scripts/feeds update -a
     ./scripts/feeds install -a

然后执行 `make menuconfig` 选好后执行

    make defconfig
    ./scripts/diffconfig.sh > seed.config

这样配置的差异部分就写入`seed.config`这个文件了  
在命令行可以输入`cat seed.config`查看这个文件，也可以用别的软件打开  
**反正你先复制一下这个seed.config的文本内容，后面步骤要**
[image]

# 第2步
1. 打开 https://github.com/P3TERX/Actions-OpenWrt  fork这个项目  
[image]

2.进入自己项目Actions页面
[image]

点击绿色的按钮

[image]

2. 之后在自己fork出来的项目里的网页里创建一个新文件(点击Create new file)
[image]

3. 然后把前面提到的seed.config文件的内容粘贴进来
[image]

4. 然后直接点提交
[image]

# 第3步 
其实已经没有第三步了，当你修改`.config`文件的时候，他就会触发自动编译，点击Actions就能看到编译的状况  

要下载固件的话，等他编译完（一般需要2小时）
1. 进项目的Actions页面
2. 点页面左侧的`Build OpenWrt`查看编译详情
3. 页面`Artifacts`有个`OpenWrt firmware`就是你要的固件（ OpenWrt packages是插件包※没啥用）
[image]

详细的教程可以看https://p3terx.com/archives/build-openwrt-with-github-actions.html

KFERMercer, 2019-12-04:
现在已经[可以自动更新上游并推送了](https://github.com/KFERMercer/OpenWrt-CI#12%E6%9C%884%E6%97%A5%E6%9B%B4%E6%96%B0-%E8%87%AA%E5%8A%A8%E6%9B%B4%E6%96%B0%E4%BB%A3%E7%A0%81-%E7%9B%B4%E6%8E%A5%E6%8A%8Amerge-upstreamyml%E6%94%BE%E5%85%A5githubworkflows%E5%8D%B3%E5%8F%AF%E5%90%AF%E5%8A%A8%E8%87%AA%E5%8A%A8%E6%9B%B4%E6%96%B0%E5%A4%A7%E9%9B%95%E4%BB%A3%E7%A0%81). 关于为什么不支持直接读取`.config`, 我在教程里说过了, 但是我承认说的确实不太明白, 在这里我解释一下. 实际上是基于这几点考虑:
-  直接读取.config会造成后期依赖混乱.
- 由于需要直接在首层目录下直接添加一个`.config`, 会随着克隆到本地而取代默认配置.
- 补充上一条, 在根目录下添加任何文实际上都是需要经过慎重考虑的.
- P3的脚本已有此功能.
- 我做这个脚本的初衷是伸手党友好以及自用, P3的脚本确实很方便, 但是不适合开发者.
### 其实我一直在推荐基础薄弱的玩家去使用P3的脚本.

iamggs, 2019-12-05:
两个都给赞。

iamggs, 2019-12-05:
我看P3TERX的教程，自己的编译环境生成`.config`后可以直接上传使用，不需要通过`diffconfig.sh`来找出差异部分上传，两者有何区别呢？

lietxia, 2019-12-05:
> 我看@P3TERX 的[教程](https://p3terx.com/archives/build-openwrt-with-github-actions.html)，自己的编译环境生成`.config`后可以直接上传使用，不需要通过`diffconfig.sh`来找出差异部分上传，两者有何区别呢？

diffconfig出来的比较小，只有差异部分。
.config原本是隐藏文件而且文件很长很大，这么操作一下方便一些。

区别用一个例子解释：
1.  我选了samba插件
make memuconfig出来的完整配置文件:会写着启用samba
而diffconfig比较后，因为和默认选项一样，所以整个一行都没有，也不会写到seed.config里

2. 然后lean更新了一个插件cifsd，用来取代samba和samba4，把samba的默认启用取消了。

如果直接用seed.config直接编译的话，重建.config的时候，samba因为默认值变成不选择了，所以新固件samba会消失。cifsd因为是默认值所以会编译进来

如果用make menuconfig 的.config文件，就会变成samba和cifsd一起编译进来

iamggs, 2019-12-05:
> > 我看@P3TERX 的[教程](https://p3terx.com/archives/build-openwrt-with-github-actions.html)，自己的编译环境生成`.config`后可以直接上传使用，不需要通过`diffconfig.sh`来找出差异部分上传，两者有何区别呢？
> 
> diffconfig出来的比较小，只有差异部分。
> .config原本是隐藏文件而且文件很长很大，这么操作一下方便一些。
> 
> 区别用一个例子解释，（使用旧配置编译，用到新版的时候有区别。如果都是用最新源码产生的就没区别）
> 
> 1. 我选了samba插件
>    make memuconfig出来的完整配置文件:会写着启用samba
>    而diffconfig比较后，因为和默认选项一样，所以整个一行都没有，也不会写到seed.config里
> 2. 然后lean更新了一个插件cifsd，用来取代samba和samba4，把samba的默认启用取消了。
> 
> 如果直接用diffconfig产生的旧的seed.config直接编译的话：
> samba消失，cifsd编进来了（重建.config的时候，samba因为默认值变成不选择了，所以新固件samba会消失。cifsd因为是默认值所以会编译进来）
> 
> 如果用make menucongfig产生的旧的.config文件直接编译：
> 就会变成samba和cifsd一起编译进来

感谢解答！

RealKiro, 2019-12-05:
> 基于以上两点，我是比较推荐用`P3TERX`的actions代码的，简单说就是编译的时候抓取一次lean的项目，这样你一直是最新的。

😯我是萌新，村里刚通网

[Actions-OpenWrt](https://github.com/P3TERX/Actions-OpenWrt)对于上传自定义package是无效的，如果想进阶添加自己的定制可以试试[build-openwrt](https://github.com/project-openwrt/build-openwrt)

P3TERX, 2019-12-05:
> > 基于以上两点，我是比较推荐用`P3TERX`的actions代码的，简单说就是编译的时候抓取一次lean的项目，这样你一直是最新的。
> 
> 我是萌新，村里刚通网
> 
> [Actions-OpenWrt](https://github.com/P3TERX/Actions-OpenWrt)对于上传自定义package是无效的，如果想进阶添加自己专属的定制可以试试[build-openwrt](https://github.com/project-openwrt/build-openwrt)

如果你是直接 push 到仓库，那么确实无效，因为我已经删除了相关代码。但这并不代表不能添加，你可以以拉取代码仓库的方式去添加，相关的使用方法可以去看我的博客。如果使用上有什么问题，欢迎随时提出来，后续我会持续改进我的方案。

uoow, 2019-12-06:
> [OpenWrt-CI](https://github.com/KFERMercer/OpenWrt-CI)现在已经[可以自动更新上游并推送了](https://github.com/KFERMercer/OpenWrt-CI#12%E6%9C%884%E6%97%A5%E6%9B%B4%E6%96%B0-%E8%87%AA%E5%8A%A8%E6%9B%B4%E6%96%B0%E4%BB%A3%E7%A0%81-%E7%9B%B4%E6%8E%A5%E6%8A%8Amerge-upstreamyml%E6%94%BE%E5%85%A5githubworkflows%E5%8D%B3%E5%8F%AF%E5%90%AF%E5%8A%A8%E8%87%AA%E5%8A%A8%E6%9B%B4%E6%96%B0%E5%A4%A7%E9%9B%95%E4%BB%A3%E7%A0%81). 至于为什么不支持直接读取`.config`, 我在教程里说过了, 但是我承认说的确实不太明白, 在这里我解释一下. 实际上是基于这几点考虑:
> 
> * 直接读取.config会造成后期依赖混乱. (这是最重要的一点. 在不断的update配置文件的过程中, 势必会提高混乱度. 不利于配置可控.)
>   
>   * 这个问题很重要, 属于任何自己编译固件的人都需要了解的. 我可能会展开说下, 但是连续爆肝几天我已经累了(待编辑).
> * 由于需要直接在首层目录下直接添加一个`.config`, 会随着克隆到本地而取代默认配置.
> * 补充上一条, 实际上, 在根目录下添加任何文件都是需要经过慎重考虑的.
> * P3的脚本已有此功能.
> * 我做这个脚本的初衷是伸手党友好以及自用, P3的脚本确实很方便, 但是不适合开发者.
> 
> ### 其实我一直在推荐基础薄弱的玩家去使用P3的脚本.

> [OpenWrt-CI](https://github.com/KFERMercer/OpenWrt-CI)现在已经[可以自动更新上游并推送了](https://github.com/KFERMercer/OpenWrt-CI#12%E6%9C%884%E6%97%A5%E6%9B%B4%E6%96%B0-%E8%87%AA%E5%8A%A8%E6%9B%B4%E6%96%B0%E4%BB%A3%E7%A0%81-%E7%9B%B4%E6%8E%A5%E6%8A%8Amerge-upstreamyml%E6%94%BE%E5%85%A5githubworkflows%E5%8D%B3%E5%8F%AF%E5%90%AF%E5%8A%A8%E8%87%AA%E5%8A%A8%E6%9B%B4%E6%96%B0%E5%A4%A7%E9%9B%95%E4%BB%A3%E7%A0%81). 至于为什么不支持直接读取`.config`, 我在教程里说过了, 但是我承认说的确实不太明白, 在这里我解释一下. 实际上是基于这几点考虑:
> 
> * 直接读取.config会造成后期依赖混乱. (这是最重要的一点. 在不断的update配置文件的过程中, 势必会提高混乱度. 不利于配置可控.)
>   
>   * 这个问题很重要, 属于任何自己编译固件的人都需要了解的. 我可能会展开说下, 但是连续爆肝几天我已经累了(待编辑).
> * 由于需要直接在首层目录下直接添加一个`.config`, 会随着克隆到本地而取代默认配置.
> * 补充上一条, 实际上, 在根目录下添加任何文件都是需要经过慎重考虑的.
> * P3的脚本已有此功能.
> * 我做这个脚本的初衷是伸手党友好以及自用, P3的脚本确实很方便, 但是不适合开发者.
> 
> ### 其实我一直在推荐基础薄弱的玩家去使用P3的脚本.

请问那个upstream的脚本会自动与L大的源码同步？是定时的还是什么时候？它自动同步完后会自动编译一次吗？

closed 2019-12-24 by coolsnowwolf

wujunze, 2020-02-10:
不错哦

cfso1234, 2020-02-27:
虽然有点一知半解，但是感觉大神就是大神，考虑问题比我这样的萌新全面多了

rodso, 2020-03-03:
最近几天编译的不能下载了是怎么回事，那个Build OpenWrt不能点进去了。。。

EnnawYang, 2020-03-03:
> 最近几天编译的不能下载了是怎么回事，那个Build OpenWrt不能点进去了。。。

连这个都要到这里问了？

lietxia, 2020-03-04:
> 最近几天编译的不能下载了是怎么回事，那个Build OpenWrt不能点进去了。。。
# 临时解决方法
1. 按f12打开控制台，切换到`console`（或者叫`控制台`），输入

    (function () {
        var a = document.querySelectorAll("a.link-gray-dark.css-truncate.css-truncate-target");
        for (var i = a.length; i--;) { a[i].innerText = "openwrt"; }
    })()

回车

[image]

# 彻底解决方法
检查yaml文件的

    #=================================================
    # https://github.com/P3TERX/Actions-OpenWrt
    # Description: Build OpenWrt using GitHub Actions
    # Lisence: MIT
    # Author: P3TERX
    # Blog: https://p3terx.com
    #=================================================
    
    name: Build OpenWrt

这里的名字是不是删掉了

EnnawYang, 2020-03-04:
> > 最近几天编译的不能下载了是怎么回事，那个Build OpenWrt不能点进去了。。。
> 
> # 临时解决方法
> 按f12打开控制台，切换到`console`（或者叫`控制台`），输入以下代码回车
> 
> ```
> (function () {
>     var a = document.querySelectorAll("a.link-gray-dark.css-truncate.css-truncate-target");
>     for (var i = a.length; i--;) { a[i].innerText = "openwrt"; }
> })()
> ```
> 
> [image]
> 
> # 彻底解决方法
> 检查yaml文件的
> 
> ```
> #=================================================
> # https://github.com/P3TERX/Actions-OpenWrt
> # Description: Build OpenWrt using GitHub Actions
> # Lisence: MIT
> # Author: P3TERX
> # Blog: https://p3terx.com
> #=================================================
> 
> name: Build OpenWrt
> ```
> 
> 这里的名字是不是删掉了

如果以release方式触发的话删除当前release就能显示名字了

ycg31, 2020-03-12:
> diffconfig出来的比较小，只有差异部分。
> .config原本是隐藏文件而且文件很长很大，这么操作一下方便一些。
>
> 区别用一个例子解释，（使用旧配置编译，用到新版的时候有区别。如果都是用最新源码产生的就没区别）
>
> 1. 我选了samba插件
>    make memuconfig出来的完整配置文件:会写着启用samba [...]

如果说我要禁用Lean大默认选中的插件呢  在config里加入 =n吗？ 如果实现 make memuconfig中M的功能   只编译成ipk但不进入固件呢，是=m还是=“！”？

kml0798, 2023-07-16:
为什么新建了.config之后action没反应呢？在此之前我修改了默认的仓库地址到了我fork的仓库，跟这个有关系吗？action里手动开始可以吗？

siren202101, 2024-09-23:
有没有大佬教教GitHub Actions云编译如何把指定的ipk也发布出来？
例如ssp或passwall