coolsnowwolf/lede #2289: 强烈反对Lean搞"OpenWrt CI 在线集成编译环境"!!!

ForgotFun, 2019-12-04:
你把编译做的如此简单，你让我如何卖"跟着佐大学OpenWrt"系列教程。

身旁的儿子问道“爸爸，为啥今天午饭没有鸡腿，前几天不是还有的嘛？”，佐大叹了一口气"怪那lean叔叔，爸爸再想想办法！"，儿子咽了咽口水回道“爸爸，今天就不吃鸡腿了，反正前几天吃过了“。

跟着佐大学OpenWrt报名地址：
http://forgotfun.org/2018/04/openwrt-training-2018.html

mjyhj, 2019-12-04:
支持
并且不但反对，还建议闭源

Promix953, 2019-12-04:
估计RedHat对Linux也是这么认为的

KFERMercer, 2019-12-04:
呜呜呜, 我哭了  建议在楼下commit收款码

dolphinpaopao, 2019-12-04:
没有必要这种语气，各有态度求同存异。

cygmris, 2019-12-04:
作为已经购买佐大教程的一员，觉着这个是没关系的，因为就算用了lean或者按照需求改了lean的repo，还是会买佐大的教程的，因为就非嵌入式出身的我来说，有些知识面得佐大提供。

cygmris, 2019-12-04:
这有点像反容器化了。。

lietxia, 2019-12-04:
应该出一部《跟着佐大学云编译Openwrt》

redchenjs, 2019-12-04:
嗯，广告才是重点

inlyt, 2019-12-04:
[image]

xiangfeigao, 2019-12-05:
这样的人，自私自利。支持lean

M1A6I3L, 2019-12-05:
单就标题讨论，反对默认编译用schedule，真需要用的人会自己打开schedule的，不会的人也没有用schedule的需求，严重浪费资源。默认用push就好，改完配置自动就编译了。

greennyreborn, 2019-12-05:
> 这样的人，自私自利。支持lean

人家是开玩笑吧

closed 2019-12-06 by coolsnowwolf