coolsnowwolf/lede #2294: collectd可否更新一下,

lxwq, 2019-12-05:
我手动更新到5.10.0时编译出错,不知道怎么回事

lxwq, 2019-12-05:
touch /home/zz/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.rrdtool1_installed
echo "rrdtool1" >> /home/zz/lede/staging_dir/target-x86_64_musl/pkginfo/rrdtool1.default.install
make[3]: Leaving directory '/home/zz/lede/feeds/packages/utils/rrdtool1'
time: package/feeds/packages/rrdtool1/compile#0.17#0.04#0.37
make[3]: Entering directory '/home/zz/lede/feeds/packages/utils/collectd'
touch /home/zz/lede/build_dir/target-x86_64_musl/collectd-5.10.0/.prepared_38b8f9ed31082ad2f60f8dbe810a1ac4_6664517399ebbbc92a37c5bb081b5c53_check
. /home/zz/lede/include/shell.sh; bzcat /home/zz/lede/dl/collectd-5.10.0.tar.bz2 | tar -C /home/zz/lede/build_dir/target-x86_64_musl/collectd-5.10.0/.. -xf - 
[ ! -d ./src/ ] || cp -fpR ./src/. /home/zz/lede/build_dir/target-x86_64_musl/collectd-5.10.0

Applying ./patches/001-undefined-AM_PATH_LIBGCRYPT.patch using plaintext: 
patching file fake-am_path_libgcrypt.m4

Applying ./patches/050-backport-modbus-little-endian.patch using plaintext: 
patching file src/modbus.c
Hunk #1 FAILED at 76.
Hunk #2 FAILED at 425.
Hunk #3 FAILED at 434.
Hunk #4 FAILED at 496.
Hunk #5 FAILED at 521.
Hunk #6 FAILED at 535.
Hunk #7 FAILED at 561.
Hunk #8 FAILED at 702.
8 out of 8 hunks FAILED -- saving rejects to file src/modbus.c.rej
patching file src/collectd.conf.pod
Hunk #1 FAILED at 4128.
1 out of 1 hunk FAILED -- saving rejects to file src/collectd.conf.pod.rej
Patch failed!  Please fix ./patches/050-backport-modbus-little-endian.patch!
make[3]: *** [Makefile:342: /home/zz/lede/build_dir/target-x86_64_musl/collectd-5.10.0/.prepared_38b8f9ed31082ad2f60f8dbe810a1ac4_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/zz/lede/feeds/packages/utils/collectd'
time: package/feeds/packages/collectd/compile#0.59#0.12#1.02
make[2]: *** [package/Makefile:109: package/feeds/packages/collectd/compile] Error 2
make[2]: Leaving directory '/home/zz/lede'
make[1]: *** [package/Makefile:103: /home/zz/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zz/lede'
make: *** [/home/zz/lede/include/toplevel.mk:218：world] 错误 2

outemancut, 2019-12-05:
050-backport-modbus-little-endian.patch补丁打不上 错误 目测是上游已经打上了 自己要么改补丁 改个版本号算啥

lxwq, 2019-12-05:
> 050-backport-modbus-little-endian.patch补丁打不上 错误 目测是上游已经打上了 自己要么改补丁 改个版本号算啥

不知道要怎么搞,刚看了一眼OP官方,这个软件已经更新到5.10.0了,移植过来试试看,

closed 2019-12-09 by coolsnowwolf