coolsnowwolf/lede #2308: Trojan无法正常运行

keke1023, 2019-12-06:
填入参数后启动，日志里会看到：
[2019-12-05 16:10:35] [FATAL] exiting. . . 
[2019-12-05 16:10:35] [FATAL] fatal: wrong run_type in config file
2019-12-05 16:10:35 Welcome to trojan 1.13.0 Started!
Welcome to trojan 1.13.0
然后我去/usr/share/shadowsocksr/gentrojanconfig.lua修改run_type为'nat'，运行后
[2019-12-05 17:28:43] [FATAL] exiting. . .
[2019-12-05 17:28:43] [FATAL] fatal: No such node (ssl.alpn)
Welcome to trojan 1.13.0
最后我干脆把整个gentrojanconfig.lua的文件内容替换成eSir软路由对应文件的内容，发现日志倒是没有FATAL但是一直有Warning，运行状态里透明代理是没有运行的，也没法正常使用
实在不清楚到底是什么问题了。。

rookietribe, 2019-12-06:
我把 run_type 为 'client' 之后，没有错误或警告的信息，但是 XXR Plus 显示没有运行

keke1023, 2019-12-06:
@rookietribe 那你的tcp透明代理应该也是没有运行的状态咯？感觉就是这个config和init.d出问题了但是不知道具体是哪里。。

norikohxb, 2019-12-06:
我的也是错误
[2019-12-06 20:18:32] [FATAL] exiting. . . 
[2019-12-06 20:18:32] [FATAL] fatal: wrong run_type in config file
Welcome to trojan 1.13.0
[2019-12-06 20:18:32] [FATAL] exiting. . . 
[2019-12-06 20:18:32] [FATAL] fatal: wrong run_type in config file
2019-12-06 20:18:32 Welcome to trojan 1.13.0 Started!

norikohxb, 2019-12-06:
你是不是也是在op的ssr里面输入trojan的参数，我的trojan架设应该没有问题，显示运行中

keke1023, 2019-12-06:
@norikohxb 手机端可以用的，esir的固件也能用，应该不是服务端问题

norikohxb, 2019-12-06:
那我去下载esir的新固件

chchia, 2019-12-07:
请问楼上两位怎么解决的？

keke1023, 2019-12-07:
@chchia 还没解决呢。。。所以才提issue啊哈哈哈哈

chchia, 2019-12-07:
> @chchia 还没解决呢。。。所以才提issue啊哈哈哈哈

看到你说esir固件能用，我还以为你有解决办法了呢。

norikohxb, 2019-12-07:
是啊 esri的固件里面的ssrplus支持trojan难道不是lean大修改的吗？我相信也是啊，难道是esir自己修改的吗。我现在用上了trojan的openwrt

norikohxb, 2019-12-07:
解决了，是我之前用的op的ssr上面的trojan功能有问题，不知道是编译问题来编译的有问题。

chchia, 2019-12-07:
> 解决了，是我之前用的op的ssr上面的trojan功能有问题，不知道是编译问题来编译的有问题。

能不能共享你的源码？或者帮我编译个Xiaomi Nano的？

norikohxb, 2019-12-07:
> > 解决了，是我之前用的op的ssr上面的trojan功能有问题，不知道是编译问题来编译的有问题。
> 
> 能不能共享你的源码？或者帮我编译个Xiaomi Nano的？

抬举我了，我还没有学会编译呢，都是用别人的op，寄生虫一只

closed 2019-12-08 by coolsnowwolf