coolsnowwolf/lede #2309: manjaro环境下如何编译

birrhof, 2019-12-06:
manjaro环境下如何编译？

hcym, 2019-12-06:
Arch吧？主要就是依赖包名字搞定，不然还是u1804稳妥

birrhof, 2019-12-06:
sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler这一句直接换成sudo pacman -S build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler.好多包没有

redchenjs, 2019-12-06:
用arch的应该对本体系的包有所了解吧，直接替换肯定是不对的呀

KevinMX, 2019-12-06:
请安装[openwrt-devel](https://aur.archlinux.org/packages/openwrt-devel/)。

closed 2019-12-07 by coolsnowwolf